bigtreetech/BIGTREETECH-TouchScreenFirmware #1858: TFT70 Random Short Pauses Causes small blobs

fiveorders, 2021-04-20:
Printing from USB, randomly short pauses causing <1mm small blobs 

TFT70 with v27 firmware, Printing from USB. The issue was not there with V26

Serial + EXP1 + EXP2 connected. FLYF407ZG Controller being used, Pin modification made to prevent speed changes with the knob on Touch mode. Marlin Nightly build bug fix CONFIGURATION_H_VERSION 020008

billyd60, 2021-04-20:
I have the same problem. My theory is it's a serial connection problem. I print in Marlin mode only until they can chase this down and fix it.

fiveorders, 2021-04-20:
If you print in Marlin mode, do you keep the serial cable attached or do you unplug the serial?

fiveorders, 2021-04-20:
It started after I upgraded to v27 on Touchscreen and Marlin Nightly Built bugfix 2.0

billyd60, 2021-04-20:
> If you print in Marlin mode, do you keep the serial cable attached or do you unplug the serial?

I leave it attached. I also have the "Serial mode always on" option selected.

The touch screen mode seems to work fine. I'll be printing for several minutes and then suddenly the printer starts losing it's mind making moves not called for in the gcode, or stopping entirely, and having the extruder spit out a bunch of filament. None of this EVER happens in Marlin mode with the exact same gcode, Marlin build and machine.

billyd60, 2021-04-20:
> It started after I upgraded to v27 on Touchscreen and Marlin Nightly Built bugfix 2.0

I literally bought the touch screen the day they released .27 so I've never had it on any other fw. I am using marlin 2.0.x bugfix from April 8th (or 9th).

billyd60, 2021-04-20:
I also get random gcode errors popping up on the screen from time to time (hours between it happening) and I've not even doing anything.

fiveorders, 2021-04-20:
That's not happening on mine or I am not able to catch it. There were no blobs on previous firmware setup but now I can't tell if it is because of Marlin Nightly Built version or the V27 on TFT70

radek8, 2021-04-20:
Hello
the problems you describe look like a problem in Marlin.
It happened to me several times that the printer behaved strangely after upgrading to the new version of Marli. After another upgrade, for example after a week, the problem was solved.
I'm currently upgrading only when I need to add a new feature or fix a problem. I am currently using the version from 1.2.2021 - c74f972627
https://github.com/MarlinFirmware/Marlin/tree/c74f972627ab8dcf69a637268cf86682959bd18e

Can you please attach your Marlin configuration files?
Configuration.h
Configuration_adv.h
Thank you

radek8, 2021-04-20:
> I also get random gcode errors popping up on the screen from time to time (hours between it happening) and I'm not even doing anything.

Can you specify the errors mentioned more precisely?

billyd60, 2021-04-20:
> > I also get random gcode errors popping up on the screen from time to time (hours between it happening) and I'm not even doing anything.
> 
> Can you specify the errors mentioned more precisely?

Sorry off the top of my head I can't recall the gcode numbers. I will pay attention and write them down. It's pretty rare though. I have sessions with the printer where it never happens. It's always when I am in serial (touchscreen) mode. The marlin mode works flawlessly as does the printer. Which is why I am convinced the problem lies with the serial connection in my case. I am in a conversation with Ultimachine right now to see if they have any ideas (my board is an Archim2).

radek8, 2021-04-20:
> Sorry off the top of my head I can't recall the gcode numbers. I will pay attention and write them down. It's pretty rare though. I have sessions with the printer where it never happens. It's always [...]

Archym 2 with connectors rotated 180 °
I remember :-)

billyd60, 2021-04-20:
> Archym 2 with connectors rotated 180 °
> I remember :-)

lol crazy right? I couldn't believe that worked. Ironically the marlin mode connection is flawless. God has a sense of humor.

fiveorders, 2021-04-20:
> Hello
> the problems you describe look like a problem in Marlin.
> It happened to me several times that the printer behaved strangely after upgrading to the new version of Marli. After another upgrade, [...]

Atached. Thank you

[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6345764/Archive.zip)

fiveorders, 2021-04-20:
> Hello
> the problems you describe look like a problem in Marlin.
> It happened to me several times that the printer behaved strangely after upgrading to the new version of Marli. After another upgrade, [...]

I would definitely go back to another version, I moved to this one due to the LCD Knob problem if you recall from the other issue here.

radek8, 2021-04-20:
Serial printing errors, where Marlin returns an error that he received an unsupported G-code and prints a nonsensical code that is definitely not in the print file, is usually caused by electromagnetic interference.
The serial connection is prone to interference, which can have various origins.
Poorly filtered switching power supply, bed switched PWM signal.
Motor drivers can also emit interference.
I recommend not using the supplied serial cable, but using your own shielded cable.
Place the display away from the heated bed, or at least shield the display from the bed.
Do not run the serial cable around the motor controllers.
Use a quality source.

radek8, 2021-04-20:
> 
> I would definitely go back to another version, I moved to this one due to the LCD Knob problem if you recall from the other issue here.

Support for disabling the knob was added to Marlin Night Release 5 months ago.
So February is fine.

fiveorders, 2021-04-20:
> Serial printing errors, where Marlin returns an error that he received an unsupported G-code and prints a nonsensical code that is definitely not in the print file, is usually caused by [...]

Actually I don't believe that is the case here, because it was perfectly fine with Stable Marlin Release. I will revert back to that one and see if it occurs. If it happens again then it is the problem with V27.

billyd60, 2021-04-20:
> Serial printing errors, where Marlin returns an error that he received an unsupported G-code and prints a nonsensical code that is definitely not in the print file, is usually caused by [...]

I am using a shielded cable. I was the quality source as I built it from scratch. lol. I'm pretty sure the cable is good. But you bring up a point about the stepper drivers, I do have that cable passing within a couple inches of those, hard to avoid doing that, since the cable enters from the top of the case, and that's where the drivers are. I will attempt to reroute the cable and see if that helps.

radek8, 2021-04-20:
I would set the serial port to which the TFT is connected first. I don't know if it's port 1 or 6 in your case

[image]

If the speed is initially set to 115200, it will be less prone to interference.

[image]

Try to change these values
[image]

radek8, 2021-04-20:
@fiveorders 
How much volt source do you have? 12V or 24V

Uncomment!
#define UTF_FILENAME_SUPPORT
#define LONG_FILENAME_HOST_SUPPORT
#define AUTO_REPORT_SD_STATUS
#define SERIAL_FLOAT_PRECISION 4
#define M115_GEOMETRY_REPORT
#define M114_DETAIL  
#define REPORT_FAN_CHANGE 
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define HOST_START_MENU_ITEM

fiveorders, 2021-04-20:
I am on 24v good quality PSU. I have uncommented those, I have set the baud rate to 115200.

What should be the buffers? 32?

radek8, 2021-04-20:
Set!!
 #define CHOPPER_TIMING CHOPPER_DEFAULT_24V

fiveorders, 2021-04-20:
Done! Trying with these modifications tomorrow. 

Just threw out 6 hours of print because I pressed the terminal button in order to M600 but when I press it stuck without switching to the terminal screen.:( I hope all these will be gone with the settings above.

radek8, 2021-04-21:
> > It started after I upgraded to v27 on Touchscreen and Marlin Nightly Built bugfix 2.0
> 
> I literally bought the touch screen the day they released .27 so I've never had it on any other fw. I am using marlin 2.0.x bugfix from April 8th (or 9th).

It's not a problem to downgrade a TFT to an older FW, but it's unnecessary. The new FW TFT is not to blame.
Try these versions of Marlin. I've tried it.

https://github.com/MarlinFirmware/Marlin/tree/c74f972627ab8dcf69a637268cf86682959bd18e

bigtreetech, 2021-04-21:
Hello, Do you have enabled the `Power loss recovery` feature of TFT? Please turn it off in `Settings->Feature->Power loss recovery` menu and print it again to test if there is still such a problem

fiveorders, 2021-04-21:
> Hello, Do you have enabled the `Power loss recovery` feature of TFT? Please turn it off in `Settings->Feature->Power loss recovery` menu and print it again to test if there is still such a problem

I already did that when I was getting frequent pauses, but this is different. Those were very frequent and was stopping for at least 2-3 seconds. these are very short and rarer. Now I am trying with the recommendations above. I will update you as soon as possible.

fiveorders, 2021-04-21:
Blobs are gone. Many thank you all!!

closed 2021-04-21 by fiveorders

radek8, 2021-04-21:
Don't forget to delete the comment for the option I recommended. You will avoid future problems.

fiveorders, 2021-04-21:
> Don't forget to delete the comment for the option I recommended. You will avoid future problems.

I didn't understand this? What should I delete? 🤷‍♂️

radek8, 2021-04-21:
//  -  two slashes in front of parameters in Marlin

# define UTF_FILENAME_SUPPORT
# define LONG_FILENAME_HOST_SUPPORT
# define AUTO_REPORT_SD_STATUS
# define SERIAL_FLOAT_PRECISION 4
# define M115_GEOMETRY_REPORT
# define M114_DETAIL
# define REPORT_FAN_CHANGE
# define HOST_ACTION_COMMANDS
# define HOST_PROMPT_SUPPORT
# define HOST_START_MENU_ITEM

fiveorders, 2021-04-21:
Which action solved my problem do you think? Baudrate? I mean how can we tell if I need these uncommented or not? thanks

radek8, 2021-04-21:
So don't do anything.
But you will still come back.
- advise me the fan speed is not updated
- advise me my hotend position will not be updated
- advise me when printing from an on board card the correct file names are not displayed
- advise me does not pop up me a call to replace the filament.
......

If you read the comments in the config.ini file properly, you would not ask.

fiveorders, 2021-04-21:
Sure, should I underscore these all?

#define EXTENDED_CAPABILITIES_REPORT
#if ENABLED(EXTENDED_CAPABILITIES_REPORT)
  #define M115_GEOMETRY_REPORT
#endif

radek8, 2021-04-21:
Delete // at the beginning of the line for the specified items

fiveorders, 2021-04-21:
I think I misunderstood. I have already enabled these by your recommendation above, I mean I already deleted // on these. So I am not changing anything leaving them enabled. 

#define UTF_FILENAME_SUPPORT
#define LONG_FILENAME_HOST_SUPPORT
#define AUTO_REPORT_SD_STATUS
#define SERIAL_FLOAT_PRECISION 4
#define M115_GEOMETRY_REPORT
#define M114_DETAIL
#define REPORT_FAN_CHANGE
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define HOST_START_MENU_ITEM

radek8, 2021-04-21:
then it's okay

fiveorders, 2021-04-21:
Just disabled this to not to see it on the top of the Marlin Mode menu. Can I keep it disabled?

//#define HOST_START_MENU_ITEM

radek8, 2021-04-21:
this is yes

billyd60, 2021-04-21:
So what should the serial buffers be set to? I didn't see that answer. What about how all the serial options should be setup in Marlin for comms with the tft?

radek8, 2021-04-21:
I read somewhere that Marlin prefers the first serial port (#define serial_port) so the default hardware should communicate on it via the one you want to print, ie TFT.
You can set the port speed to 250000, but if you have problems with serial port errors, set less, ie 115200.
It's a good idea to set your cache for problems like yours.
That is, when short curves are printed at high speed. It may then happen that in the busy state the supply of G-codes will not be fast enough. Caching these states will help overcome without stopping
If you don't have a problem, you don't have to reach for the cache.
I sent the values that I use and I have no problem with that.

fiveorders, 2021-04-21:
Buffers 16 here, didn't touch them... works very good