bigtreetech/BIGTREETECH-TouchScreenFirmware #1860: How to get past both "stuck/looping" touchscreen calibration and resulting "Illegal Flash App"? (BIQU B1 is now unusable.)

DougJoseph, 2021-04-20:
Hello. 

How to get past both "stuck/looping" touchscreen calibration and resulting "Illegal Flash App"? 

(My BIQU B1 is now unusable.)

For my device (TFT) variant installed in my BIQU B1, and for additional info, here is a photo of my "info" page before starting the update effort:

https://imgur.com/a/bG8IxzM

I'm a relatively new BIQU B1 user, and I was directed to a new firmware update for the touchscreen, available here:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

I was not sure which .bin file to use, but I saw "B1" in one of the filenames, so I used that file, which was named "BIGTREE_TFT35_V3.0_B1.27.bin"

I copied the stock "config.ini" and the "BIGTREE_TFT35_V3.0_B1.27.bin" file onto the mini SD card that came with the printer (2GB, formatted as FAT), along with the folder named "TFT35" from the parent folder called "THEME_Unified Menu Material theme."

I placed the SD card into the touchscreen's SD slot, and got a positive response pop-up indicating a validated card had been found. I powered down and restarted the printer.

The images and fonts seemed to load fine, and the firmware update *seemed* to be going alright.

However, I got a screen to calibrate the touchscreen, and each time I would get three red dots (and tap them), but after the third, no forth red dot appeared, but rather a black dot in the middle, with a message to tap a red dot, though there was no new red dot. No matter what I tapped, it would say that the calibration failed, and start over and repeat the same thing. I repeated this quite a few times.

After repeated failures, I turned the printer off, and back on. I got "illegal flash app."

I then tried reformatting the SD card and replacing the files. I got "illegal flash app."

I then tried reformatting the SD card and replacing the files, but this time using a file that had "BIQU" in the name: "BIQU_TFT35_APP1_V2.0.27.bin." I got "illegal flash app."

I then tried reformatting the SD card as FAT32, and replacing the files, but this time using a different file that had "BIQU" in the name: "BIQU_TFT35_V1.2.27.bin." I got "illegal flash app."

I also tried "BIQU_TFT35_V1.0.27.bin" but to no avail.

Somewhere along the way (rather soon into all this), the machine started saying "illegal flash app" from the moment it was turned on, no checking of anything.

All I can get now is "illegal flash app." This happens whether the SD card is in the printer or out of it. !!!!

I sure would welcome any advice on how to fix it. Thank you!!!

oldman4U, 2021-04-20:
Hi.

You did everything right at the beginning.

Format the sd card. 
Put the first firmware, config.ini and TFT folder on it.
Install everything like you did before the calibration screen appeared.

The problem is the description on screen while calibration where it says press the red dot even the fourth is the black one in the middle.

Nothing is broken. Everything will work finally. 👍🏻

DougJoseph, 2021-04-20:
Thanks. Two things in reply:

I tapped / pressed the black dot in the middle repeatedly, and the calibration kept failing.

The printer is currently unusable / broken. All boot ups are showing a blank, black screen with only these words in white, "Illegal Flash App." 

Again, it's definitely broken.

oldman4U, 2021-04-20:
You have to press the dots as they appear. 3 red dots followed by one black dot.

But before you can do so you have to upload a firmware. After that you must remove the sd card like described in the Readme on the front page of this repository.

oldman4U, 2021-04-20:
The foundation of the TFT is the hardware and the bootloader. Both are ok otherwise you would not get the Illegal Flash App warning which indicates that no valid firmware can be found. So your problem is in software and therefore easy to fix. I had this Illegal Flash App maybe 100 times because I forgot to remove the sd card.

BTW. Use the files you used in your first attempt to update the firmware and not files you already used or which are currently on your sd card.

Are you using Windows, macOS or Linux?

DougJoseph, 2021-04-20:
Thank you so much! I am indeed back into a working touchscreen and printer!

Sometimes the simple word that "what you did was correct" gives the added encouragement to try one more time.

Some comments in retrospect: 

I think the initial firmware update happened correctly, all except for the repeatedly failed screen calibration. It seems that by re-booting after that with the firmware files then marked as ".cur" automatically triggered a message of "Illegal flash app" — making me think the prior flash had failed, when apparently, I simply needed to remove the SD card before rebooting? Perhaps?

Finally, by lack of confidence in my thinking the flash had failed, due to the "illegal" message, I then tried other files, which were the wrong files, apparently, which meant I continually got the "Illegal Flash App" message and dispaired I had bricked the TFT when in fact I had not.

Things are working great now.

PS: I had originally used my Mac to add the files to the SD, but then used a virtual machine of Windows 10, to format the SD card as "FAT32" and to copy the files to it.

oldman4U, 2021-04-20:
You would be a big help here. All your assumptions are correct and the error is the wrong message to press a red dot where a black dot has to be pressed and the missing documentation.

First I can not fix. Second I can. So I will add this information to the Readme and hope it will help other users.

I asked because I also use macOS and doing so you have to format the sd card after using it for the firmware update. Deleting the files is not enough, guess it has to with the way the macOS handles fat formatted devices.

Please help the community and close the ticket once you do not need it anymore.

Thank you and happy printing

DougJoseph, 2021-04-20:
OK, thank you again! Closing it as it's completely solved, and quite happily!

closed 2021-04-20 by DougJoseph

oldman4U, 2021-04-21:
Changed the README based on your input: https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware

Happy printing

oldman

DougJoseph, 2021-04-21:
> Changed the README based on your input: https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware
> 
> Happy printing
> 
> oldman

Very cool! Thank you!

Luizm80, 2022-01-30:
Hi, Oldman...

I'm a beginner at 3DPRINTER and I need to learn a lot about it.

I had this issue before but I did something and I fixed that, but now I got the upgrade for my Biqu B1 Bd Printer looks "BIQU-B1-Two-in-one-out dual-color printing upgrade kit" so following the steps to install the extruder I,m not sure if I installed correctly on the mainboard but I'm keeping tried the process.
The directions said to download the firmware here:

https://github.com/bigtreetech/BIQU-B1/tree/master/Firmware/2021-04-28/Motherboard-210428/Motherboard-SKR%202-210428

I did and transferred it to the Sd card and input inside the Sd slot card close the LCD and I got the "Illegal flash Card".

I started the process to revive my printer by doing the download here:

https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware.git

and follow the steps to install the files but I keep having "Illegal flashcard"

Can someone help me to fix these problems because first of all, I want to use my printer and finish and finish the dual-color installation and get back to using the printer.

Best Regards

Luiz

oldman4U, 2022-01-30:
Hi Lutz.

In this repository we help in case of a problem with the firmware. This is not a user group where we can help with setup issues. For this the group's mentioned when starting a new issue are good for.

meier2009, 2022-11-17:
hi 
im new in this section!
i search a while on the web for my issue, but everythings seems not to point to my problem,

i tried to update my E3 RFF and TFT 24 1.1 but after first try, the display only shows ILLEGAL Flash App.

AFTER THIS TRY even if there is a card or not in either slot on board or the display. i seems "hardbricked"....

also not working to format different cards and put the firmware.bin oder the TFTxx.bin onto it.

doesn´t make any difference.

if i power the printer on, Illegal flash app. 
when reset, try put card in or something else

also it doesn´t looks like it try´s to update with any file

s