bigtreetech/BIGTREETECH-TouchScreenFirmware #1865: is TFT screen compatible with marlin bugfix

sinole1987, 2021-04-22:
I have been trying to setup marlin bugfix with my skr 1.4 T and TMC 2209, I got it to work with marlin mode, but when i went to TFT mode. some crazy stuff was happening, like nozzle crashing to the bed when I homed any axis, no axis moved with command and ETC. Have anyone been able to do it? is it even yet compatible with TFT.

oldman4U, 2021-04-22:
I am a bit confused. You are posting in this repository for some weeks now and now this question!???

Let's ask the other way around. Is there a single user here using the last official Marlin 2.0.7.2 version which is several months old?

I use Marlin bugfix version from April 11th with SKR E3 DIP/2208 and TFT35 E3 v3 with the master firmware from the beginning of this week successfully.

If it does not work for you I would recommend to use the official Marlin version or does this also not work?

sinole1987, 2021-04-22:
> I am a bit confused. You are posting in this repository for some weeks now and now this question!???
>
> Let's ask the other way around. Is there a single user here using the last official Marlin [...]

I have marlin 2.0.7.2, it is working, but some features are bugged on it. and it is not the best when it comes to using UART mode Drivers. a lot of noise and vibrations. I am just trying to upgrade to bugfix. but I don't know  why is it acting so strange with my TFT mode.

billyd60, 2021-04-22:
I haven't been able to get the serial side of the TFT to work with marlin bug fix to date either. Marlin mode works flawlessly. Very strange behavior with random nozzle moves and skipped gcodes. I am told it is interference with my serial connection but I am using a shield cable. I have given up and will wait for someone else to figure it out. In the meantime I print in marlin mode.

sinole1987, 2021-04-22:
what is your board and driver combination?
at least you could get the printer to home in TFT mode.

billyd60, 2021-04-22:
> what is your board and driver combination?
> at least you could get the printer to home in TFT mode.

I am using an Archim2 main board with the TFT 70 v3 on .27
It works in touch mode (serial connection) for periods of time, but eventually something goes wrong after several minutes of printing and it starts skipping codes or making up it's own moves out of the blue. Printing from serial is the only problem, all other touchscreen operations via serial work fine, such as ABL or configuring settings or any other non-printing function. Even terminal works great, sending commands to the printer via serial. When printing from the tft in marlin emulation (flat ribbon connection) it's flawless in every respect, including printing. This isolates the problem to the serial connection without a doubt, since I am using the same SD card, same gcode, and card reader in either config, and the same printer and marlin firmware (and settings) of course.

So it's either the serial configuration in Marlin, a problem with serial connections on the Archim2 in general, the cable, or a problem with the the tft hardware or firmware. I am still working the problem but I don't have a lot of enthusiasm at the moment. Tired of it.

oldman4U, 2021-04-26:
@sinole1987 

What is the status of this ticket?

sinole1987, 2021-04-26:
> @sinole1987
> 
> What is the status of this ticket?

I really don't know man. I gave up, ill wait for next revision marlin. we can close the ticket I guess.

sinole1987, 2021-05-01:
hey I just had an idea, I download my marlin from marlin github and set it for SKR. do you think I should download it from SKR Github? that firmware must me improved for the board and BTT products. Am I right?

sinole1987, 2021-05-01:
forget bugfix marlin 2.0.8 came out.

oldman4U, 2021-05-01:
I will install it in about 8h

sinole1987, 2021-05-01:
its too much work. lol 
I don't know anything about the new interpolation stuff on TMC drivers. need to do whole bunch of research. I just got my hybrid threshold right.

oldman4U, 2021-05-01:
This ticket is about the compatibility of the BTT TFT firmware with Marlin bugfix and I had no problem with it and now I finished the first print using v2.0.8 which is more or less the same like bugfix.

I must add that I use stealthchop but not hybrid threshold so in case you have a problem like described above it is because of it.

Please help the community and close the ticket once you do not need it anymore.

sinole1987, 2021-05-02:
TMC 2209 had a issue with stealthchop and 0.9 degree stepper after 100mm/s movement speeds your print makes excessive vibrations and make crazy noise. so I had to activate hybrid threshold. maybe the issue is fixed now.

closed 2021-05-02 by sinole1987

oldman4U, 2021-05-02:
In this case you would see this issue independent of using the TFT or not.

sinole1987, 2021-05-02:
it is independent. its a known issue, with tmc 2209s. there is also issue of motor whine  in full spread cycle mode. 0.9 degree motors are niche  and hard to setup, but the print quality is worth trying them specially in a bed slinger like my printer. If you wana learn more about the issue see these links. the youtube one shows it very nice. https://github.com/MarlinFirmware/Marlin/issues/18286. 
https://www.youtube.com/watch?v=A7GWkTTyzW0
hopefully they could update TMC software and fix those issues.

oldman4U, 2021-05-02:
My printer works nice as it is. But thank you.

So you can use Marlin 2.0.8 with v27 TFT firmware?

sinole1987, 2021-05-02:
Im still dealing with dropouts. I just received ferrite beads and new cable that has thicker power delivery line . I looked at the connection I used wrong side on motherboard side. the shield wasnt going to the ground it was on TX. So I sent all of the distortion to the TFT. 🤣 
as soon as i get some stable prints and I know I have solid machine I will update the firmware.