bigtreetech/BIGTREETECH-TouchScreenFirmware #1867: TFT35 status screen does not show part fan speed with E3-RRF board using RRF

Wombat37, 2021-04-22:
### Description

When a print is initiated from an SD card or remotely via DWC or Cura plugin, the display only shows 0% F0 fan speed. If the g code print method sets or changes this fan speed, this is not reflected on the TFT35 display. 

### Steps to reproduce
Print something with an active part cooler fan setting.
Note that the zero fan speed displayed is always 0%.

**Expected behavior**
I would expect the F0 % reading on the status screen to display the current fan speed 

**Actual behavior**
Part fan speed display always shows 0% unless manually changed on the TFT. The fan appears to be controlled properly even though the status is not being updated.

### Hardware Variant
E3-TFT35-3.0
E3-RRF-1.1

### TFT Firmware Version & Main Board Firmware details
E3-TFT35-V3.0.27
RRF 3.2.2

### Additional Information

radek8, 2021-04-23:
V Marlin je situace taková. Na rychlost venrilátoru se nedá zeptat, proto byla v Marli provedena úprava aby Marlin hlásil každou změnu rychlosrti ventilátoru. Proto se TFT aktivněneptá na rychlost ventilátoru, ale jen přijímá hlášení od Marlin.
Předpokládám že v RRF bude situace podobná a bude potřeba úprava na straně RRF. V případě že by RRF umožňoval dotazování, byla by tu možnost opravit FW TFT.

Wombat37, 2021-04-25:
Thanks for the clarification radek8.

I understand that some things just aren't possible.

However, it's not clear why this we have such a feature that doesn't do anything :)

radek8, 2021-04-25:
What function do you mean?
In Marlin, the fan speed display works well.
Refer to the RRF documentation for how rrf can report the fan speed via the serial interface.
If it is in the RRF setting (so it is with Marlin) then set it.
If there is a command that triggers the required response, write what the syntax of the command is.
Then we can contact Bigtreetech about its implementation.

I repeat that the FW for TFT is tuned for Marlin and the implementation for RRF and other alternative systems has started recently. But it is nothing systematic. Usually, a user like you will hear that something is not working for him, or he will control the programming and adjust the function himself so that it works both in Marlin and in his system.

Bigtreetech would like to sell its displays to RRF users as well, but it doesn't do much. 
The programmer community on this project is mostly Marlin. 
It's hard to program something you don't have to test.

Wombat37, 2021-04-25:
Hi radek8, I'm sorry I don't mean to criticize - I understand and appreciate what you are saying. I'm sorry that BTT don't seem to support these developments as much as I thought they would be and leave the work to others. 

From my perspective, I saw the RRF and the TFT as parts of the same product that I was trying to use. To say the fault is in one of these parts and not the other may be true but it doesn't help anything. I'm learning the hard way the politics of this situation.

Should I be posting this in the RRF repository?

I would be more than willing to test any of these fixes - my E3-RRF/E3-TFT35/Ender 3 Pro system seems to be working very well under RepRap - once I understand the various work-arounds. I will also endeavor to learn more about the system to make my own fixes as you kindly suggest.

Thank you for all your efforts.

radek8, 2021-04-25:
Find out how the RRF reports a change in fan speed to the serial port

Wombat37, 2021-04-27:
Hi Radek8, I'm sorry but I have no idea on how to answer your question.

I've posted a message on the TeamGloomy RRF site to see if anyone there can help.

Thanks you for your help.

Wombat37, 2021-04-27:
P.S. Is this the same issue as #1834?

bigtreetech, 2021-04-27:
> Hi Radek8, I'm sorry but I have no idea on how to answer your question.
> 
> I've posted a message on the TeamGloomy RRF site to see if anyone there can help.
> 
> Thanks you for your help.

Hello, TFT35 communicates with RRF through the standard gcode protocol. At present, I haven't found a way to report or poll the fan speed on RRF, So this problem needs to be solved by modifying the logic of RRF part (or TFT communicates with RRF through JSON format similar the PanelDue, which needs a lot of modification).
For example, if the RRF reports the speed of the fan after receiving the `M106 Px` gcode,  the TFT can parse and display it normally.

> P.S. Is this the same issue as #1834?

Yeah, It is caused by the same principle that the motherboard does not report the fan speed to the TFT through serial when the speed changes,

jaysuk, 2021-05-19:
M409 can be used to query the fan speed directly from the object model

bigtreetech, 2021-06-02:
Hello, This should be fixed in #1962   Please test

bttguy, 2021-06-07:
@bigtreetech #1962 resovles this issue, I started a print, and the fan speed updated during it, also the values set by the slicer, and also my manual overwrite.

PR #1962 (2021-06-15): improve RRF
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

### Benefits

### Related Issues
fix #1856 M303 Auto tune PID for RRF
fix #1867 Auto query status to get fan speed
fix #1909 Set relative/absolute mode after startup.
closed 2021-06-15 by bigtreetech