bigtreetech/BIGTREETECH-TouchScreenFirmware #1870: [Q] Can I use the power loss recovery

nobaj, 2021-04-23:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

oldman4U, 2021-04-23:
Does the PLR of the TFT not work!?

nobaj, 2021-04-23:
omg the tft has a plr?

oldman4U, 2021-04-23:
There is even a setting for this, where you can turn it off. Default on, so it is very likely activated already.

Please close the ticket once you do not need it anymore.

Thank you

oldman4U, 2021-04-24:
Are you still looking for it, or do you have a problem with it?

oldman4U, 2021-04-26:
@Msq001 This ticket is unrelated to the firmware and the user does not reply.

Please check and close the ticket. Thank you

closed 2021-04-27 by bigtreetech