bigtreetech/BIGTREETECH-TouchScreenFirmware #1874: Hanging at the end of the firmware update

BortSerg, 2021-04-24:
### Description

At the time of the firmware, the process stops at the time of updating the "keywods". The display shows: "Updating cinfiguration ... Total keywords found: 115". And that's all. Then nothing happens, no matter how much I wait.
 I originally thought the problem was changing the parameters in Configuration.h, but nothing changed when the firmware was compiled with the original (unmodified file).

### Steps to reproduce

1.  In the platformio.ini file change the display type to BIGTREE_TFT35_V2_0
2.  Compiling firmware in VS Code (platformio)
3.  Display firmware.

**Expected behavior**
I expected the firmware to be completed as with other previous versions

**Actual behavior**
Nothing happens. How long did not wait, the information on the display does not change. After reboot, just a black screen, there is not even a display of the BTT logo

### Hardware Variant
BTT TFT35 V2.0
BTT SKR 1.4
### TFT Firmware Version & Main Board Firmware details
TFT - BIQU_TFT35_APP1_V2.0.27
Marlin - 2.0.7.2

### Additional Information
[Configuration + foto.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6369246/Configuration.%2B.foto.zip)

GregSKR, 2021-04-24:
Yes, I had the same problem.

You have to do a 'Simple Reset':
Create a blank textfile named "reset" (with the ".txt" extension), and place it in the root folder of your SD card.
Insert the SD card into the TFT's SD card reader and power cycle your printer or reset your TFT.

After that, your TFT will start as it should, and you have to 'flash' again your config.ini file.

I think it should be solved with PR #1866

oldman4U, 2021-04-24:
Readme reading helps also.

oldman4U, 2021-04-24:
Or you have to do this:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Bootloaders/TFT35_V2%20Bootloader%20fix

BortSerg, 2021-04-25:
@GregSKR , thanks. Your advice helped.

@oldman4U, I read about errors, but it also said that the error code will be shown on the display, but there was no code. Just stopping the firmware process.

The question can be hidden. Updated the firmware, everything works fine.

closed 2021-04-25 by BortSerg

oldman4U, 2021-04-25:
Happy printing

👍🏻