bigtreetech/BIGTREETECH-TouchScreenFirmware #1881: [BUG] Screen Freeze and high-pitched beep

pauluchin, 2021-04-26:
### Description

While printing, it suddenly starts to emit a high-pitched beep, the screen stops responding and then the printing stops.
I emphasize that I have PLR deactivated

### Steps to reproduce

1. Print from Octoprint or from any media from the screen
2. wait for it to happen

### Hardware Variant
BIGTREETECH TFT35-E3 V3.0
TFT Firmware 27
Main Board SKR-mini-E3 V2 Firmw. Marlin 2.7.2

### Additional Information
*Zip with  Configuration.h and Configuration_adv.h
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6373671/Marlin.zip)

[Video](https://youtu.be/EoACc3Omd_s)

oldman4U, 2021-04-26:
And you believe or even have some evidence that the firmware of the TFT and nothing else causes this because....?

oldman4U, 2021-04-26:
And what should the video show? It shows normal behaviour and nothing what you describe...

pauluchin, 2021-04-27:
set the sound to Max and you can hear the beeps and you will see that it does not change mode to marlin

oldman4U, 2021-04-27:
Because you are printing you can not switch the mode. The time you are not printing you can press the knob like this to switch the mode. 

Because you press the knob you hear the sound. Release the knob to get rid of the sound. 

Please help the community and close the ticket once you do not need it anymore. Thank you

pauluchin, 2021-04-27:
After that I show in the video the whistle was continuous and the printing stopped and the screen crashed

oldman4U, 2021-04-27:
The Video starts at 14:02 print time and is 18sec long. At the end of the movie the print time jumps from 14:19 to 14:20.

When you release the know the beep stops.

Maybe a better movie shows the issue, but so far everything I have seen and heard shows a normal behaviour.

And you showed no evidence that the problem is the TFT firmware?

pauluchin, 2021-04-27:
I will close while collecting info to upload

closed 2021-04-27 by pauluchin