bigtreetech/BIGTREETECH-TouchScreenFirmware #1883: TFT70 BTT stealtchop X and Y not showed in TFT mode

tittopower, 2021-04-26:
Issue occurred updating to latest firmware TFT70:
After updating TFT 70 to latest firmware in the menu->options->printer->parameters->TMC stealtchop  for X and Y is showed value 0.000 even if stealtchop is enabled in marlin (2.0.8). Is possible set to 1.00 but after exit 0.000 is showed. In marlin mode the stealtchop is correcly showed as enabled. Same issue not occurred updating with the same version of firmware a TFT35V3.

oldman4U, 2021-04-26:
Which TFT firmware are you using, the one from 4 days ago or the official version v27?

radek8, 2021-04-26:
Do you write that you only have a problem on the TFT 70 display?
Do you also write that the problem is only on the X and Y axis? Is the Z and E axis OK?

guruathwal, 2021-04-27:
This is a major drawback of displaying only the settings reported by Marlin firmware in M503. Even if stealthchop is enabled in Marlin, it will not report stealthchop settings if it turned off.

radek8, 2021-04-27:
> This is a major drawback of displaying only the settings reported by Marlin firmware in M503. Even if stealthchop is enabled in Marlin, it will not report stealthchop settings if it turned off.

I treated this by setting it in the parseACK.c file
if the TMC drivers are in Uart mode and respond to the M503 command with parameter M906, zero is set when the Chop stealtch is switched off.
@guruathwal,  what you write would happen to him if the TMC controllers were not in Uart Mode, which I do not assume.

[image]

radek8, 2021-04-27:
@tittopower , You can enter your printer's answer to M503 here
Thank you

guruathwal, 2021-04-27:
@radek8 I did not miss it.  It is not necessary that the stealthchop feature is enabled in Marlin for a TMC driver. With your suggestion, stealthchop will be visible for all of the steppers even if stealthchop is disabled in Marlin firmware. which will confuse the user in thinking that the stealthchop is available for all steppers. 

[image]

I used the hybrid threshold to set the stealthchop parameter visibility separately for each stepper that has hybrid threshold enabled because the hybrid threshold is dependant on stealthchop to be enabled in Marlin firmware.

radek8, 2021-04-27:
It is well.
But if I enable Stable chop in marlin and disable Hybrid Threshold, will Stealth chop be unavailable?
Is it better to have the TFT option available even though it is forbidden in Marlin than to have it unavailable even though it is allowed in Marlin, just because I do not have a Hybrid Threshold enabled?
Of course, it makes sense to enable both parameters in Marlin, but a lot of people don't, because HT is disabled in Marli by default.

I agree with your logic, but it would be good to point this out in the config.ini comment, as a recommended configuration.

tittopower, 2021-04-27:
Precisations about this issue: 
Firmware used is the BIGTREE_TFT70_V3.0.26.x dated 13/03/2021, not the latest probably as now
Used the same version on TFT35V3
The TMC settings are in SPI mode in the printer with TFT70 and UART mode in the printer with TFT35V3, the mainboard is the same (BTT 1.4Turbo) for the 2 systems with the same version of marlin. 
Before updating, with  the original firmware of the TFT70 that come when I buy it all was fine but I do not know the original version and date.
An image from M503 is here, pick the page for TMC setting:
[image]
Hybrid threshold is enabled and fixed to 111 mm/sec.

tittopower, 2021-04-27:
[image]

radek8, 2021-04-27:
I can already see what the problem is
You have 2 stepper motors Z
Line M569 S1 I1 Z overwrites the X and Y axes to zero ....
The detection of parameter M569 does not take into account the second stepper motor on the Z axis

radek8, 2021-04-27:
I'd fix it, but @guruathwal is already working on an update, where the printer's entire parameters have been reworked into multi-step motors on the axis.
I will not spoil his great work :-)

guruathwal, 2021-04-27:
I believe this problem is already fixed in #1884. @tittopower can you test #1884 to see if it solves the problem?

tittopower, 2021-04-27:
Thanks guruathwal, tell me if the this is the right way to proceed:  

1. correct line 51 of  file: TFT/src/User/API/HomeOffsetControl.c   
from  "z_offset_value = getParameter(P_HOME_OFFSET, Z_STEPPER);"     to
          "z_offset_value = getParameter(P_HOME_OFFSET, AXIS_INDEX_Z);"
2. compile a new firmware with VScode and upload it
3. Leave an empty line before "return z_offset_value;"
Is this correct ? I use Notepad++ is OK ?
Thanks !

guruathwal, 2021-04-27:
You don't need to edit any file. Just compile it.

radek8, 2021-04-27:
for  @tittopower 
[BIGTREE_TFT70_V3.0.27.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6386465/BIGTREE_TFT70_V3.0.27.zip)

radek8, 2021-04-27:
@tittopower ,
You have quite high current values on stepper motors. So you have to have big engines and a big printer :-)

tittopower, 2021-04-27:
@radek8 : I have this printer, but images are not updated to the actual electronic : 
https://www.tittopower.it/page69.html

tittopower, 2021-04-27:
@radek8 
The firmware you send me work great, thank you to solve this for me !
[image]

Before try your firmware I compiled mine with no success, so if iou can please tell me from what version of TFT firmware you started to compile your....starting to upload your I had a scare ! It started in inverted screen....then happens the calibration....then update language and finally  started correcly .... in my past upgrades never happened this kind of sequence (first files and then calibration). Thanks a lot !

tittopower, 2021-04-27:
The electronic  now:
[image]

radek8, 2021-04-27:
It is best to use GitHub Desktop.
https://docs.github.com/en/desktop
You clone the bigtreetech repository.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
Use VS Code to compile.

radek8, 2021-04-27:
@guruathwal .
I just tested your PR 1884.
Stealthchoip doesn't work well.
The initial state is read from the M503 correctly. 
When changed from 0 to 1, the menu must be exited and returned to be updated.
After changing from 1 to 0, the change is sent to Marlin, but does not appear on the display even after leaving the menu.
The problem will be in interfaceCmd.c

radek8, 2021-04-27:
this is wrong
 uint8_t k = (cmd_seen('S')) ? cmd_value() : 0;
The constant k must have a value as the parameter S.
Conditions if it is contained = 1 otherwise = 0 is wrong

radek8, 2021-04-27:
[image]

kisslorand, 2021-04-27:
@radek8 
The ones you labeled "OK" and "WRONG" have absolutely the same the same result. Athwal just used ternary operator instead of defining `k` and than using the `if` conditional expression.

radek8, 2021-04-27:
Sorry my mistake.
Why InterfaceCMD.c cannot capture M569
while others are okay?

radek8, 2021-04-27:
Bingo!
I found it
This is command M569
Case 596 is specified in interfaceCMD.c

[image]

PR #1884 (2021-05-06): Clean-up parameter settings and display only axis/stepper supported by the machine
- Clean-up parameter settings.
- Display only axis/stepper supported by the machine.
- Fix some bug in parameter parsing in parseACK.c and interfaceCmd.c
- Workaround to display stealthchop settings if not reported by Marlin firmware when it is turned off.
- Separate persistent info code from parameter settings.

Build report:

Device | Old RAM Usage | New RAM usage | RAM Change | Old Flash Size | New Flash Size | Flash Size Reduced
-- | -- | -- | -- | -- | -- | --
TFT35 V1.1 | 17544 | 17596 | 52 | 148188 | 1476 [... cut]
closed 2021-05-06 by bigtreetech