bigtreetech/BIGTREETECH-TouchScreenFirmware #1886: [BUG] Errore continuo in stampa: Printer Halted TFT35 E3 V3

SteULM, 2021-04-27:
### Description

### Steps to reproduce

1. [Riscaldamento]
2. [Avvio stampa]
3. [Dopo poco si blocca riportando l'errore]

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

kisslorand, 2021-04-27:
Sajnálom de nem értem mi a gond. Próbáld angolul írni, több az esély, hogy választ kapjál.

bigtreetech, 2021-04-28:
You mean that the firmware version v.27 will have a "Printer Halted" error during printing? And you can print it normally with the firmware of v.26?

Please click "OK" in the "Printer Halted" error pop-up window, and then click the title bar to enter the `Notifications` menu and send me the detailed information displayed in the `Notifications` menu

SteULM, 2021-04-28:
Yes, during the first printing stages, the error appears. With version 26, however, no problem. Attached is the video where you can see the crazy temperatures in version 27 and stable in version 26.

https://user-images.githubusercontent.com/83312403/116388193-d79e6f00-a81b-11eb-9ccb-42292425b7ba.mp4

[image]
[image]

https://user-images.githubusercontent.com/83312403/116388255-ebe26c00-a81b-11eb-9169-51ffc0b22f87.mp4

oldman4U, 2021-04-28:
What kind of thermistor are you using and how many?

One for extruder and one for bed?

SteULM, 2021-04-28:
Yes, 100k (Creality CR10) and 100k for the bed.
I confirm that it is with the firmware vers. 26 that by reassembling the RepRapDiscount Full Graphic Smart Controller everything works correctly without any type of error, the problems are only with the firmware vers. 27 installed correctly following the guide.

SteULM, 2021-04-28:
Thanks for the help

SteULM, 2021-04-28:
With the vers. 27 the extruder cannot reach the set temperature while with vers. 26 reaches it.

oldman4U, 2021-04-29:
Please provide the Notification like requested by Bigtreetech 22h ago.

SteULM, 2021-04-29:
What is reported in the notification menu is displayed in the photos I attached to previous posts, however the strange thing is that the problem only occurs with version 27 of the firmware.
The errors reported are:
Thermal Runaway, System stopped! Heater_ID: 0
Printer Halted. kill() called!

oldman4U, 2021-04-29:
I do not believe that you provided the required information.

The error is shown in case the heater does not heat properly or the thermistor has a problem. I had this once because of a broken cable. But I do not have this problem using v27 and it seems other users do not have it too. 

Hope BTT can help you.

SteULM, 2021-04-29:
Thanks for the help, I can only say that I have double checked all the wiring and there are no problems, as I said before, the problem occurs only with the v27 and I have no problem even if I put the graphic display I used before. I hope they can help me because the tft is excellent and well done and I would like to continue using it with version 27 because I find it very complete compared to the previous one.

bigtreetech, 2021-04-30:
It is very strange that according to the video you provide. the reason for Printer Halted is that the temperature of Marlin motherboard can not be stabilized. In fact, TFT only receives and displays the temperature. Marlin is responsible for PID temperature control. In principle, no matter which version of TFT firmware is likely to cause this problem. Unless the TFT modifies the PID parameter somewhere?

The only difference between your two videos is the firmware version of TFT (One is V.x.x26, the other is Vx.x27)  and everything else is exactly the same (Including the hardware environment, and Marlin firmware is exactly the same)?

SteULM, 2021-04-30:
Yes hardware itself. In fact, trying to replace the SKR 1.4 board with one with the same Marlin the problem disappears. In this case the TFT has no problems. Thanks for the support and I congratulate you on a great product.

oldman4U, 2021-04-30:
This leaves the question left, how the older TFT firmware could "repair" the broken SKR.

Good to hear you found the problem, please close the ticket once you do not need it anymore.

Happy printing

SteULM, 2021-04-30:
No after more time but it also crashed with the old firmware. Thanks anyway to everyone for the support.

closed 2021-04-30 by SteULM