bigtreetech/BIGTREETECH-TouchScreenFirmware #1890: BIGTREE_TFT_V3.0.27.bin The Extrude (Load/Unload) screen is not working correctly. The the bottom 4 buttons are from the heat screen

gaahner, 2021-04-29:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

gaahner, 2021-04-29:
[image]

oldman4U, 2021-04-29:
And because they are from the heat screen they can not be useful somewhere else?

What if you have more than one nozzle or you can only (Un)Load at a minimum temperature. And how would you leave the screen without a back button?

Does UnLoad work correctly?

kisslorand, 2021-04-29:
@gaahner 
What is the functional issue you're reporting here? Yeah, those icons and functions can be seen in another menu. So?

bigtreetech, 2021-04-30:
Hello, This menu calls `M701/M702` to inform Marlin to go to `Load` or `Unload` and required to enable `FILAMENT_LOAD_UNLOAD_GCODES` in Marlin.
Maybe the menu you want to use is `"Menu->Movement->Extrude"` ?

gaahner, 2021-04-30:
the function of the buttons do not work like the old extrude screen. The load and unload buttons give an unknown command M701/M701. the heat button should be distance to extrude length not heat. The heat button now takes you out of the extrude screen. The cool down button should be a speed setting. Now it turns off the nozzle heat.

oldman4U, 2021-04-30:
And you get the unknown command because you do not follow what is written in the ReadMe on the front page.

oldman4U, 2021-05-02:
@gaahner Hove you found the Extrude menu in the meantime?

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-05-02 by gaahner