bigtreetech/BIGTREETECH-TouchScreenFirmware #1891: [FR] Show remaining time

MLiv79, 2021-04-29:
Would be possible to add the remaining time during a print?

oldman4U, 2021-04-29:
Please see PR #1410.

kisslorand, 2021-04-29:
Yes, it's possible if you know the total time needed for the print job.
That would be possible by analyzing the gCode before the print job. Not an impossible task but it would be overkill for the TFTs MCU with little benefit.
It would take some time to analyze it and than people will start tickets that it takes too long for a print to start...

oldman4U, 2021-04-29:
Or it is part of the gCode. But this only works as long as the speed is not changed.

MLiv79, 2021-05-01:
> Yes, it's possible if you know the total time needed for the print job.
> That would be possible by analyzing the gCode before the print job. Not an impossible task but it would be overkill for the [...]

Marlin standard do this without any issue. Progress bar, Elapsed Time and remaining time. These are written by the slicer (prusa for example) in the gcode. Using touch mode no remaining time with the same gcode that with Marlin mode show all the data written above, included remaining time.

oldman4U, 2021-05-01:
Have you tried the PR I mentioned?

kisslorand, 2021-05-01:
Where should it be displayed? I mean exactly where in the print menu? There is not really any extra space left.

oldman4U, 2021-05-01:
Using the PR I mentioned above the remaining time is shown INSTEAD of the elapsed time. The final implementation should show the elapsed and remaining time alternating like Flow and Speed is shown today.

oldman4U, 2021-05-01:
@MLiv79 Please test it and let us know.

MLiv79, 2021-05-01:
> Using the PR I mentioned above the remaining time is shown INSTEAD of the elapsed time. The final implementation should show the elapsed and remaining time alternating like Flow and Speed is shown today.

Exactly in this way

kisslorand, 2021-05-01:
How would you know which is which?

oldman4U, 2021-05-01:
Using different Icons and maybe even Text. Like Speed and Flow.

MLiv79 and, have you tested the PR in the meantime?

MLiv79, 2021-05-01:
> Using different Icons and maybe even Text. Like Speed and Flow.
> 
> MLiv79 and, have you tested the PR in the meantime?

I'm printing a thing sliced with IM now so I've not looked at it again but the first time I've printed i used prusa and on prusa/superslicer I've m73 enabled (even in cura now is possible) and see only elapsed time. But I've to look at it again. If would be possible in next release also a number at the end of the progress bar would be useful instead of the bar alone. Thanks @oldman4U

oldman4U, 2021-05-01:
You have to use #1410, otherwise you will not see the remaining time.

MLiv79, 2021-05-02:
> You have to use #1410, otherwise you will not see the remaining time.

Could you elaborate a little bit more? Because I've m73 p r enabled and i have it in the gcode but on tft i still see the elapsed time

oldman4U, 2021-05-02:
Which firmware of the TFT ate you using?

MLiv79, 2021-05-02:
> Which firmware of the TFT ate you using?

Latest .27 compiled by myself (but didn't work even with the ones already compiled included)

oldman4U, 2021-05-02:
But I asked you to use #1410 and not latest v27...

V27 does not #1410 does show remaining time.

MLiv79, 2021-05-02:
> But I asked you to use #1410 and not latest v27...
> 
> V27 does not #1410 does show remaining time.

I compiled by myself master.zip release, is not included in there?

oldman4U, 2021-05-02:
No it is not.

MLiv79, 2021-05-02:
> No it is not.

in which zip is included? Developer?

oldman4U, 2021-05-02:
When you go to Pull Request you will find 1410 and there you can download the firmware like normal.

MLiv79, 2021-05-02:
> When you go to Pull Request you will find 1410 and there you can download the firmware like normal.

This should include all last commit included yours, right?
[image]

oldman4U, 2021-05-02:
It is based on an older version of the firmware, but you can see if it does what you want: https://github.com/belese/BIGTREETECH-TouchScreenFirmware.git

sarvenn, 2021-05-03:
> It is based on an older version of the firmware, but you can see if it does what you want: https://github.com/belese/BIGTREETECH-TouchScreenFirmware.git

The one above or this? https://github.com/belese/BIGTREETECH-TouchScreenFirmware/tree/M77Support

oldman4U, 2021-05-03:
Should be the same

MLiv79, 2021-05-20:
> Any news?

It worked for me when i tested it but it is not included in the master

oldman4U, 2021-05-20:
Ok.

@sarvenn?

oldman4U, 2021-05-22:
There is a new PR available related to the requested function. Please test and let us know. Thank you

https://github.com/digant73/BIGTREETECH-TouchScreenFirmware.git

sarvenn, 2021-05-24:
> Ok.
> 
> @sarvenn?

Hi @oldman4U, I am away from home for a couple of weeks due to my on-going projects at my work. I will try that pr asap. But we may merge it now and follow the bug reports as well. It's a matter of choice and I respect both ways.

Hitamku, 2021-05-25:
> There is a new PR available related to the requested function. Please test and let us know. Thank you
> 
> https://github.com/digant73/BIGTREETECH-TouchScreenFirmware.git

Hai, remaining time not showing. I use Biqu B1. What did i miss?

oldman4U, 2021-05-25:
Make sure your slicer adds the needed information to the gCode.

Hitamku, 2021-05-25:
> Make sure your slicer adds the needed information to the gCode.

I've tried activating m73 in marlin, m117 in config, display progess on lcd in Cura. But the remaining time still doesn't appear

oldman4U, 2021-05-25:
Attach the gCode file.

Hitamku, 2021-05-27:
I did print test and also did test using gCode test

; FLAVOR: Marlin
M117 Time Left 0h01m40s
G4 P10000
M73 P10
M117 Time Left 0h01m30s
G4 P30000
M73 P40
M117 Time Left 0h01m00s
G4 P30000
M73 P70
M117 Time Left 0h00m30s
G4 P30000
M73 P100
M117 Time Left -1h59m59s

the result is the same. Does it only work in Mks?