bigtreetech/BIGTREETECH-TouchScreenFirmware #1893: [Q] ps_on documentation

RotorVulura, 2021-05-01:
Hi,
I am whining around looking for soft start (aka power on) my atx psu from tft.

Connecting 5v stb and gnd from power supply to tft, display starts. No problem.
Connecting ps_on from power supply to ps_on pin located on tft does not do the magic.

Or, better, sending m81 trough the tft terminal, power supply comes alive. With m80 it shut down.
It seems inverted.

Changing 1 to 0 in config.ini ps_on_active_high does not sort any effect.

Maybe ps_on on tft works with btt ups or stuffs like that. 

Does anyone have any clue ? My friend Google did not help much. I found just big solid state relay cutting the main ac power. I would like to use my atx ps_on (off) features.

My tft is MZ version. It comes with e3-35 firmware version. Tried out with all 35 series firmware (vanilla, e3 and b1). Everyone works, but ps_on. Same behavior, m81 power on, m80 power off.

Thanks for reading this,
Have a nice weekend :)
RV

RotorVulura, 2021-05-02:
Compiled Vxx27 with inverted m80-m81 command in extend.c
M80 is now !infosettings.ps_active_high and m81 is infosettings.ps_active_high

So, m80 console command starts my atx psu and m81 stop it like expected.

Also, 5v stb connected to tft provide power via rs232 to the skr 1.4t. So, when the fiscal switch of the atx psu is on, skr comes alive with tft and I can send gcode. With M80, skr comes totally up with fan and stepper powered.

guruathwal, 2021-05-21:
`ps_on_active_high` in config.ini is applicable only if the power supply is directly connected to the TFT's PS_ON pin.
You are able to control the power supply with M80/M81 commands which mean you have connected your power supply to the Mainboard. You need to check the power supply configuration in your Marlin firmware.
If you have any issue relating to M80/M81, then you can open an issue in Marlin firmware's repo here:  https://github.com/MarlinFirmware/Marlin/issues.

RotorVulura, 2021-05-21:
@guruathwal hi, thanks for kind reply. Tested directly from tft ps_on to atx psu. No Marlin in between. The "issue" is that if you set in tft config.ini 0 or 1 (in the part of the code related to auto shutdown), nothing change. I expect that it can invert the logic of tft ps_on (from low to high and/or viceversa), but it does not.

guruathwal, 2021-05-21:
Ok, thanks for the quick reply. I found the problem and will push a fix for it.

oldman4U, 2021-05-21:
Hi @RotorVulura.

Could you please test Gurmeet's PR to see if it solves the problem.

Thank you

rightawy, 2021-05-22:
I had a similar issue earlier, so I connected the power supply to the mainboard instead. Now with #1940 It is fixed when connecting the power supply to the TFT for me.

PR #1940 (2021-05-22): Fix power supply bug
- Fix Power supply invert logic not working. (#1893)
- Remove unused setting variable.
- Remove the Babystepping icon from mesh validation menu.
- Misc code clean-up to reduce program memory.

Fixes: #1893
closed 2021-05-22 by bigtreetech