bigtreetech/BIGTREETECH-TouchScreenFirmware #1895: [BUG] after loading new firmware on tft 35 the led green is with wite color!and the others with the white color ((( if i switch to the old version everything is good)

mitofun, 2021-05-02:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-05-02:
The exact type of TFT and the exact firmware version are a secret?

radek8, 2021-05-02:
The whole description of the problem is one big secret.

oldman4U, 2021-05-02:
Maybe it is a Samsung TFT35...?

;-)

oldman4U, 2021-05-02:
Just tested the firmware from April 28th on my TFT35 E3 v3 and the colours are working fine.

mitofun, 2021-05-03:
> Just tested the firmware from April 28th on my TFT35 E3 v3 and the colours are working fine.

https://youtu.be/i1zXh8DkLsI

mitofun, 2021-05-03:
> The whole description of the problem is one big secret.

https://youtu.be/i1zXh8DkLsI

oldman4U, 2021-05-03:
So this is a TFT with neopixel connector and you have neopixel or RGB leds connected there?

mitofun, 2021-05-03:
> So this is a TFT with neopixel connector and you have neopixel or RGB leds connected there?

neopixel - also if i send m150 u200 or other color command works good) also if i install the previous version of tft firmware - everything works good! but the new one does not works properly*(

oldman4U, 2021-05-03:
And you have seen the new neopixel function in confi.ini and setup it right?

mitofun, 2021-05-03:
Need to check it ;) > And you have seen the new neopixel function in confi.ini and setup it right?

mitofun, 2021-05-03:
> And you have seen the new neopixel function in confi.ini and setup it right?

the neopixels connects to the board) also if i print with gcomand in cura works good- only from tft does not works properly)

mitofun, 2021-05-03:
> neopixel

the neopixel function only for the terminal from tft) i have got connection to the bard ) also if i send command m150 u200 works good! also if i switch of from the tft the led and try to send command again nothing works! but once i press green from the tft the rgbw is green and white) and if want it light green go to commands and send m150 u200 works again good) so the problem might be in the congif ) the previous version works good)

mitofun, 2021-05-03:
> And you have seen the new neopixel function in confi.ini and setup it right?

3 May 2021 https://youtu.be/8agzcoLhu-E

digant73, 2021-05-03:
@mitofun it seems you have and RGBW. from the LED custom menu move to second page and set the white component to 0. Previous TFT fw was working because the white component was never sent to Marlin (only RGB components were sent)

mitofun, 2021-05-03:
Ok thank will try it;)))> @mitofun it seems you have and RGBW. from the LED custom menu move to second page and set the white component (W) to 0. Previous TFT fw was working because the white component was never sent to Marlin (only RGB components were sent)

mitofun, 2021-05-03:
Yep did it on custome and set the white to 0 works - but still do not sorted why I press the green from the tft and it is green and white ;)

digant73, 2021-05-03:
yes I know that. For standard colors, the withe componet is always set in this moment. The LED Color menu is general and it was preferred to provide also W component

PR #1888 (2021-05-06): added action:resumed and action:resume + code cleanup
**IMPROVEMENTS:**
 * added "action:resumed" and "action:resume" on parseAck fixing a missing dialog for the M600 not properly working reported in #1829. Now, resuming a print, the "Resume" button is properly switched to "Pause" button even in case REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER or ADVANCED_PAUSE_FEATURE is not enabled in Marlin
* Code cleanup. Reviewed .h and .c files just to include only the needed dependency files (in particular avoiding to include general files in the .h files)
* added ENABLE_MBL_START_Z in Configuration.h [... cut]
closed 2021-05-06 by bigtreetech