bigtreetech/BIGTREETECH-TouchScreenFirmware #1896: [FR] Child lock

szepnorbee, 2021-05-02:
A child lock capability would be very useful. I print a lot of small things for my older child and he often looks at the machine while printing. Sometimes he will wittingly or unwittingly presses a few buttons, causing the print to fail a few times.

It would be nice to have a timer that locks the display after a short time during printing, which can be unlocked in some simple way at any time. (special, invisible / visible pattern or long press on some lock button, etc.)

I would find this a simple but occasionally useful feature. 

Since not everyone is likely to find it useful, I imagine it as an optional option before bin compile.

kisslorand, 2021-05-02:
They can always switch the power off....

Even thou it's an interesting idea, it's unlikely to be implemented due to it being a small niche.
I am sure you can come up with smart ideas how to child-proof your printer.

sinole1987, 2021-05-03:
ye I would keep the whole thing out of reach BTW. nozzle on your printer is 200+ degrees, you don't need to be worried about screen. it like putting child like on hydraulic press or jigsaw machine.

oldman4U, 2021-05-03:
I have the problem the other way around. Maybe you can develop an access rights management which allows my son to access the printer only during certain times and in case he finished his homework

;-))

sinole1987, 2021-05-03:
I  think marlin has a password option in the firmware configuration.

sinole1987, 2021-05-03:
//#define PASSWORD_FEATURE
#if ENABLED(PASSWORD_FEATURE)
  #define PASSWORD_LENGTH 4                 // (#) Number of digits (1-9). 3 or 4 is recommended
  #define PASSWORD_ON_STARTUP
  #define PASSWORD_UNLOCK_GCODE             // Unlock with the M511 P<password> command. Disable to prevent brute-force attack.
  #define PASSWORD_CHANGE_GCODE             // Change the password with M512 P<old> S<new>.
  //#define PASSWORD_ON_SD_PRINT_MENU       // This does not prevent gcodes from running
  //#define PASSWORD_AFTER_SD_PRINT_END
  //#define PASSWORD_AFTER_SD_PRINT_ABORT
  //#include "Configuration_Secure.h"       // External file with PASSWORD_DEFAULT_VALUE
#endif

does this work with TFT? I don't know.

guruathwal, 2021-05-05:
How do you stop your child from touching the Heatbed while it is hot?

szepnorbee, 2021-05-05:
Ok guys. Let's make it clear that the child is not stupid at all, but rather curious. In my case, he is 3 years old and knows very well that the bed is hot. (although with PLA, the maximum 50 degrees doesn't do much damage...) And the nozzle (Biqu B1) is almost inaccessible. And a stove display can be locked even though the top is hundreds of degrees. From here on I'm not interested in the subject. Thanks for the constructive comments... The topic can be closed.

sinole1987, 2021-05-05:
didn't mean to offend you, I just thought3D printer is a tool and dangerous. Did you try the password thing to see wheatear you can lock the screen with a G-code? If it  works All you need  is to add M510 in your costume G_codes.  and every Time you press the key printer should lock.

sarvenn, 2021-05-05:
You may build an enclosure from IKEA Lack table and have a small lock (padlock) on the enclosure doors. This may also improve your print quality especially for hard to print materials such as ABS.

digant73, 2021-05-10:
> Ok guys. Let's make it clear that the child is not stupid at all, but rather curious. In my case, he is 3 years old and knows very well that the bed is hot. (although with PLA, the maximum 50 [...]

Only you can close the ticket

closed 2021-05-10 by szepnorbee