bigtreetech/BIGTREETECH-TouchScreenFirmware #1897: The display freezes during printing.

AntoszHUN, 2021-05-03:
Hello!

The display crashes randomly during printing. In this case, there is a white dot in the upper left corner. But I noticed that a yellow dot usually appears in the upper right corner.

Has the problem of occasional freezing been solved? For me, it only crashes occasionally in Touch mode. Marlin mode is flawless. Therefore, I think there may be a problem around the TFT firmware.

Made by: AntoszHUN

oldman4U, 2021-05-03:
Can you do a test and remove the device you use to print from during a print to see f this leads to the same issue.

Any hint that the TFT firmware causes this?

AntoszHUN, 2021-05-03:
> Can you do a test and remove the device you use to print from during a print to see f this leads to the same issue.
> 
> Any hint that the TFT firmware causes this?

Yes. If you print from the TFT70 in Marlin mode, the error will not occur.

oldman4U, 2021-05-03:
And when you disconnect the EXP1 cable Marlin still works?

This only shows that Marlin mode and the related hardware is not affected but is zero evidence that there is a problem with the TFT firmware.

Please do the test and let me know because on my printer it is the only way to force this issue by simulating a problem with the storage I print from.

oldman4U, 2021-05-03:
Since when do you see this issue? What has changed?

oldman4U, 2021-05-03:
Sorry. You wrote the display freezes. But the print itself does not freeze but prints correctly until the end?

AntoszHUN, 2021-05-03:
No change. Printing also stops. You can see it at the link below. I posted to the BTT group. https://m.facebook.com/groups/bigtreetech/permalink/1198986037223695/?notif_t=video_processed&notif_id=1619278737714280&ref=m_notif

guruathwal, 2021-05-03:
@AntoszHUN I have also printed several hours on TFT70 and did not have any freezing issue. Please check the serial cable connection on the mainboard. It may look good but that dupont connector does not fit well on a JST connector. Sometimes it may get loose which may cause a loss in connection when the machine vibrates while printing.

oldman4U, 2021-05-03:
What means no change. You are active here since a year or two and always had this issue?

I am not on Facebook.

AntoszHUN, 2021-05-03:
During the video, several people wrote that freezing usually occurred in them as well. There was no firmware update 1 month ago. Maybe an update would improve it. I'll see you when I get home from work.

AntoszHUN, 2021-05-03:
> @AntoszHUN I have also printed several hours on TFT70 and did not have any freezing issue. Please check the serial cable connection on the mainboard. It may look good but that dupont connector does [...]

I used to replace the dupont connector with a JST. :D

oldman4U, 2021-05-03:
Maybe you can answer my questions from above.

AntoszHUN, 2021-05-04:
The questions are meaningless. The memory card flawlessly with other cards also crashes in touch mode. But since the firmware is 1 month old, I will update. I close this mistake because I see that no one understands what is wrong. Bagatel asks questions. I'm not stupid. I know what to check in such cases. I close this mistake. If it reappears after a firmware update, I will open a new one.

closed 2021-05-04 by AntoszHUN

oldman4U, 2021-05-04:
How can a smart person like you even ask others for help!?

AntoszHUN, 2021-05-04:
> How can a smart person like you even ask others for help!?

I'm not smart. We are here to help. That's what I thought. I usually write here as a last resort when I’ve tried everything.

oldman4U, 2021-05-04:
This is the BTT TFT firmware repository which is related to issues and feature requests of the firmware.

oldman4U, 2021-05-04:
So it is only that my questions are not worth to be answered.

Maybe you can solve this problem like you solved the problem you had with the SFS.

PEZ18, 2023-01-31:
What does the Yellow and White Dot in the upper corners indicate? My printer has been freezing randomly out of nowhere... When I preview the gcode, it all looks fine, so I dont know what is causing it. Today I noticed the upper right corner has a blinking yellow dot during a print, but I cant find any information on what this blinking yellow dot even means. It blinks for a couple seconds, and then  turns off, then blinks again after about 5-10 seconds.

PEZ18, 2023-02-10:
> […](#)

Interesting... So are you saying the voltage to the LCD is too low, or do you mean the overall voltage to the main board is too low? I noticed it mainly only does this in the beginning of the prints, but it sometimes will happen in other sections. Also, do you know where I can find some documentation about this? maybe a menu diagram that explains other things like the white dot on the left side of the screen?