bigtreetech/BIGTREETECH-TouchScreenFirmware #1900: [BUG] Printer stuck on LCD12864 / Touch Mode

zackypick, 2021-05-05:
### Description

The printer boots up with two buttons: "LCD12864 Mode" and "BIGTREETECH Touch Mode", but when clicking "BIGTREETECH Touch Mode" nothing happens.

### Steps to reproduce

1. Turn on Printer
2. Try to click Touch Mode

**Expected behavior**
Printer to show main home printing screen

**Actual behavior**
The following screen is diaplayed

[image]

Then when clicking on the center left of the screen (black area) the following screen appears

[image]

But then everything is stuck.

### Additional Information
This happens very often and is really hard to get out of this situation. It seems that magically I only sometimes am able to get back to the home print screen and proceed from there.

bigtreetech, 2021-05-08:
Hello， Please try the firmware after https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1906

digant73, 2021-05-08:
hmm the default settings for MKS (and the HW changes) were provided by me. In case HW modification to support Marlin mode is missing there is no problem to switch to Touch mode (you should simply get the black screen we see in Marlin mode). Just retested now also with LCD enabled/disabled in Marlin fw with the same zackypick's HW.

kisslorand, 2021-05-08:
Same here, no problem getting in and out of Marlin mode on an MKS TFT. Even thou Marlin mode is not active on my system (same black screen as digant73 mentioned) it is not a problem switching back to touch mode.

guruathwal, 2021-05-08:
Marlin mode should be disabled for hardware that does not support it out of the box. It does not matter if the problem is coming from marlin mode or not. It may or may not be a hardware issue. There are many times when an issue exited but not all are able to replicate it. Activating an option not supported by the default hardware will waste the time of both the users and developers in finding the cause of the actual issue.

digant73, 2021-05-08:
@zackypick when you boot up the print, the last selected menu (Marlin mode or Touch mode) should be started. Based on what you reported, it seems you always see the select mode menu (the menu with the two icons). Did I undestand correctly? if so, I suspect your TFT can need a recalibration and/ot the frame of your TFT is pressing on the TFT and that's could be the reason of the displayed selecy mode menu

oldman4U, 2021-05-08:
The user does not respond to his own ticket since 3 days. Seems Marlin is disabled now by default. Maybe this ticket should be close then.

oldman4U, 2021-05-11:
@Msq001

User does not respond anymore. Please check and close. Thank you