bigtreetech/BIGTREETECH-TouchScreenFirmware #1901: Re-order file list with newest files at top

Wombat37, 2021-05-05:
**Is your feature request related to a problem? Please describe.**
My BBT-E3-RRF motherboard SD card must kept in position when using RepRap firmware. 

Files to print are loaded onto this card and, over a period of time, the list of files will occupy multiple pages on the TTF print file selection page. The newest file (i.e. the most likely one to print will be at the bottom of this list and then several pages may have to be scrolled to get to it. See the image below:

[image]

 
**Describe the solution you'd like**
I would like to see the order of the listed files to be reversed, so that the newest file would be at the top of the list and be immediately accessible. This is how it worked on the original Ender 3 board.
 
**Describe alternatives you've considered**
One solution would be for me to delete older files and keep the list short - but what's the point of providing a list like this if you can't keep much in it? 

**Additional context**
Not a serious problem but tedious and inconvenient.

kisslorand, 2021-05-05:
As far as I know the list is given to the TFT by the mainboard. So, it's not a TFT issue.

Wombat37, 2021-05-05:
Thanks for the clarification.

I seem to be making a habit of posting to the wrong site. If I post in the mainboard site, they tell me I should be posting here.

This seems to be a strange demarcation considering both products are made by the same company.

RotorVulura, 2021-05-05:
Hi,
If you are using tft sd slot there is not any solution at the moment (as I can see, but I could be wrong).SD write works in blocks. The last file added will goes at the very first free blocks (usually at bottom of the list). But, If you delete some old file (from the top of the list and/or in the middle), new files may be goes to fill free blocks creating a new fantastic "random" list. End of the story.
If you print from mainboard sd slot, you can tell Marlin in configuration.adv.h for sorting files in alphabetical order or the newest at the top of the list.
The best solution, for now, is keeping files subdivided in folders. Also, there is a very nice btt tft plugin for cura slicer that shows a preview of the .stl directly on the screen. Browsing per image instead name is more nice and less boring:)
Cheers

oldman4U, 2021-05-06:
It is not the hardware but the software which provides this functionality. In both cases it is not BTT. BTT provides "only" the hardware.

kisslorand, 2021-05-06:
@RotorVulura 
> If you print from mainboard sd slot, you can tell Marlin in configuration.adv.h for sorting files in alphabetical order or the newest at the top of the list.

He's using RRF.

RotorVulura, 2021-05-06:
@Wombat37 @kisslorand @oldman4U 
Sorry, my fault. Need to sleep at night instead posting :)

Wombat37, 2021-05-06:
Thank you all for your inputs.

A few further points...

The RepRap DWC lists files on the onboard SD card with the newest files first (as I'm proposing). However the M20 command does list the files with the oldest first.

RotorVulura's suggestions are good but using the STL preview on the TFT seems to have some issues for me. The files are now displayed nicely with icons of the part to be printed but after selection, the following error "M118 Missing Parameter "S"" appears. However, the print does continue. 

I also note that the STL preview doesn't work for files on the motherboard SD card but I still get the M118 error if I print these modified files from there.

This error is repeatable with different files I'll write this up as another issue.

Wombat37, 2021-05-06:
Looked into the STL_Preview feature, looks like there's some Marlin code used for the M118 statement which wouldn't work with RepRap. I assume that this is nothing to do with the TFT firmware and so I've posted the issue on the developer's site:

https://github.com/bigtreetech/Bigtree3DPluginSuit/issues/11

oldman4U, 2021-05-09:
@RotorVulura could you please test PR #1908 which offers sorting function through config.ini.

Thank you

radek8, 2021-05-09:
> Looked into the STL_Preview feature, looks like there's some Marlin code used for the M118 statement which wouldn't work with RepRap. I assume that this is nothing to do with the TFT firmware and so I've posted the issue on the developer's site:
> 
> [bigtreetech/Bigtree3DPluginSuit#11](https://github.com/bigtreetech/Bigtree3DPluginSuit/issues/11)

Yes this code M118 has something in common d TFT :-)
Unfortunately, he probably doesn't work with RRF
Explained here:
bigtreetech/Bigtree3DPluginSuit#11

RotorVulura, 2021-05-09:
> @RotorVulura could you please test PR #1908 which offers sorting function through config.ini.
> 
> Thank you 

Yep, tomorrow afternoon. Sorry for that. This action will delete my changes made for  #1893. Did you read about that ? It is just a problem of mine or it is a bug ?
Tomorrow I'll be back in touch with sorting.
Cheers

oldman4U, 2021-05-10:
Hi. Thank you for your feedback. Not sure about the hardcoded part. I have to check the PR tomorrow.

It is enough here. Thank you again.

Happy printing

RotorVulura, 2021-05-10:
FeatureSettings,c
[image]

oldman4U, 2021-05-17:
Hi RotorVulura.

You know that you can change the text here, but like many other text elements in the settings area, this can not be changed using the language files.

Wombat37. The TFT now allows to sort the file when printing from TFT storage by Name and Date and it allows to invert the selection. Any progress related to the RRFirmware?

RotorVulura, 2021-05-17:
@oldman4U hi, thanks for your kind reply. Yes, I know. You aks for test. I tested ^_^"
Many thanks for your time and support.
Cheers 
RV

guruathwal, 2021-05-18:
@Wombat37 please close this issue.

Wombat37, 2021-05-18:
Hi, 

Thank you to everyone who worked on this.

I've been asked to close this issue - however I've installed the new BIN and Config_RFF.INI files and I still have the following concerns on my RepRap system:

- The file list on the SD card on the BTT E3 RFF motherboard is still unchanged with the oldest file listed first when displayed on the TFT35-E3 - I asked if it was possible for this list to be reordered
- On the SD card plugged into the TFT35-E3, the file list is now in order of newest first (good - halfway there)
- The Name and Date sort options (mentioned above) are not available to me for either the onboard SD card or the TFT35-E3 SD card, so I don't know what's gone wrong here. For RepRap, I'm obviously running in Touch Screen mode.

I don't think we're quite ready to close this issue yet - or do you want me to close this particular issue and open more issues to cover the stuff still not working?

I do appreciate these efforts.

oldman4U, 2021-05-19:
This ticket is about the ability to change the way files are sorted on the on board sd card. This has to be done by the firmware running on the mainboard like it is done by Marlin today.

Make sure to upload the current master firmware of your TFT which gives you the ability to change the sorting of TFT related storage.

You have been asked to close the ticket because it is unrelated to the firmware of the TFT. And this repository is about the firmware of the TFT only and not about other hardware or software or a user group.

Thank you and happy printing

Wombat37, 2021-05-20:
My apologies. I didn't realize that the TFT firmware had nothing to do with the way a list of main-board SD files is presented and selected from the TFT screen. 

It's interesting that RepRap on the DWC screen does show them ordered with the newest first.

Thanks for the clarification.

closed 2021-05-20 by Wombat37

oldman4U, 2021-05-20:
Sorting works for TFT storage now?

Wombat37, 2021-05-20:
Not for me - although the list was now showing the newest first, the arrows to allow reordering weren't visible to me.

oldman4U, 2021-05-20:
Where do you expect the arrows?

Wombat37, 2021-05-20:
Sorry - I was just going on the description given above by RotorVulura. I must have misunderstood.

radek8, 2021-05-20:
[image]

oldman4U, 2021-05-20:
It is a new feature. 

Happy printing

Wombat37, 2021-05-20:
Now I see the change - I was looking in the wrong place! My bad!

Very nice.

Would love to see this applied to the on-board SD file list - but that's off limits I understand.

Thanks for this.

oldman4U, 2021-05-20:
On board is controlled by the mainboards firmware. Marlin does it.