bigtreetech/BIGTREETECH-TouchScreenFirmware #1904: Config dot ini file corrupted

BlueBlue58, 2021-05-06:
I have downloaded the files to update my BTT-TFT35-E3-V3.0, saved the correct files to an SD card then tried updating it.
All fonts OK.
All BMPs ok. Config file would not load.
I tried editing and resaving the file - did not work.
I tried copying the new infor into an old config file and saving it - did not work.
Conclusion is the configu files is corrupted when it was last updated and did not save correctly.
I now have a monitor that I cannot use and a machine sitting idle because I cannot get on it due to not being able to use the monitor.
[image]

oldman4U, 2021-05-06:
Please try this firmware and let me know: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/0f464dc0e95d12f60a49a8a66fba9045e84b6274.zip

This is the version which works on my TFT's.

radek8, 2021-05-06:
I just updated my display with the latest version of FW and everything works as it should.
Download the new config.ini file, save it to a tab, and restart the TFT.
If that doesn't help, save the blank reset.txt file on the card and restart the TFT. Then reload the configuration.

BlueBlue58, 2021-05-06:
I have updated my TFT firmware many times before so I do know what to do but the config file was corrupted.
I have now brought it into Visual Studio Code and re-saved it as Config.ini and it now works thank you.
The file in the download folder is corrupted and that will cause problems for people who don't know how to or don't have Visual Studio Code to correct it.

radek8, 2021-05-06:
The config.ini files appear to be fine.  The problem will be in the program in which you are editing the file.

oldman4U, 2021-05-06:
The downloaded config.ini is fine.
[image]

Please verify again and close this ticket afterwards.

Thank you

radek8, 2021-05-06:
@BlueBlue58 do you have the  option cleared in Windows "hide file extensions of known types"?

Cheo7811, 2021-05-06:
@oldman4U I think something is going on, I got the same error and after trying a couple times with a new config.ini this happens:

[image]

Then tried an empty reset.txt and got the calibration screen (sort of) and back to the previous one:

[image]

My display is a TFT70, working just fine before the update.

BlueBlue58, 2021-05-06:
radex

> @BlueBlue58 do you have the option cleared in Windows "Hide extensions for known file types"?

No. I got the same error when I downloaded the zip file on my Mac and also on my Windows Laptop so both different systems but gave the same error. It was only cleared when I opened the file in Visual Studio Code and re-saved the file as an .ini file.

bigtreetech, 2021-05-07:
Hello, Please check if the file of the corrupted config.ini has 3 bytes `0xEF 0XBB 0xBF` at the beginning.  Is it the problem of UTF-8 BOM?

oldman4U, 2021-05-07:
If it is not downloading as config.ini- what else is it downloading? Send a picture.

guruathwal, 2021-05-07:
@Cheo7811 There is an issue after #1888 where the config update failed for TFT70. The issue is solved in #1903.

Cheo7811, 2021-05-07:
> @Cheo7811 There is an issue after #1888 where the config update failed for TFT70. The issue is solved in #1903.

@guruathwal if a fix was already merged them it shouldn't happen, I am using the latest available files, so it means it has not been merged as of yet?

guruathwal, 2021-05-07:
@Cheo7811 the fix is in PR #1903. It is not merged yet.

oldman4U, 2021-05-07:
But you can download it from Gurmeets repo which is in the PR.

Fixes and new functions should be tested before merged.

BlueBlue58, 2021-05-07:
> If it is not downloading as config.ini- what else is it downloading? Send a picture.

[image]

Cheo7811, 2021-05-07:
> But you can download it from Gurmeets repo which is in the PR.
> 
> Fixes and new functions should be tested before merged.

Nope, I just downloaded his repo and tried flashing again with his precompiled firmware.bin, TFT70 directory (THEME_The Round Miracle Menu Material theme) and config.ini file, I get the same scrambled display.

I even compiled his sources for the TFT70 and got the same result.

Any ideas?

BlueBlue58, 2021-05-07:
> But you can download it from Gurmeets repo which is in the PR.
> 
> Fixes and new functions should be tested before merged.

Have uploaded a picture of what is being downloaded. The fix is simple if someone who has authority can be bothered to just re-save the corrupt file as a dot ini file and exchange it for the existing one. Such a big issue for a simple fix! I only reported it because it took me a while to figure out what was wrong and found the answer but others may not be able to do so.

Cheo7811, 2021-05-07:
> Have uploaded a picture of what is being downloaded. The fix is simple if someone who has authority can be bothered to just re-save the corrupt file as a dot ini file and exchange it for the [...]

BlueBlue58, I don't think the file is corrupted, you need to go to View, Options and unselect Hide extensions for known types.

[image]

[image]

guruathwal, 2021-05-07:
@BlueBlue58  there is nothing corruped. All the files have correct names in the screenshot you provided. You have set a default app to open the .ini file, because of this the file extension is not visible. You need to copy the file as it is without renaming it.

guruathwal, 2021-05-07:
@Cheo7811 you will have to flash all fonts and icons also.

BlueBlue58, 2021-05-07:
> @BlueBlue58 there is nothing corruped. All the files have correct names in the screenshot you provided. You have set a default app to open the .ini file, because of this the file extension is not visible. You need to copy the file as it is without renaming it.

And when I did that it locked out my screen!!!! That was the problem. You guys do not make things easy for others, a simple fix and you go round the houses trying to blame one another for gods sake. I wash my hands of it. If you knacker other people's screens because no one can be bothered to do a simple task then so be it.

Cheo7811, 2021-05-07:
> @Cheo7811 you will have to flash all fonts and icons also.

I did, they  are inside the TFT70 directory and as I mentioned I copied them from your forked repo together with your BIGTREE_TFT70_V3.0.27.bin and config.ini

Still the same scrambled display as in my initial picture.

BlueBlue58, 2021-05-07:
> @Cheo7811 you will have to flash all fonts and icons also.

The fonts and images were fine. It was the config file that was corrupt. Which part are you guys not listening too or reading?

I wash my hands of this nonsense. You are too bright for your own good sometimes.

BlueBlue58, 2021-05-07:
> BlueBlue58, I don't think the file is corrupted, you need to go to View, Options and unselect Hide extensions for known types.
>
> [image]
>
> [image]

Don't be insulting!

closed 2021-05-07 by BlueBlue58

Cheo7811, 2021-05-07:
@BlueBlue58 , the only one showing a bad actitud is you, I also have a printer I can't use because of the update, I don't think the file is corrupted but the code broke somewhere and that is what needs to be figured out.

Perhaps the developers can provide an older snapshot of the firmware from a few days back so that you can get your printer back in working order while they figure out what has happened in the code.

guruathwal, 2021-05-07:
@BlueBlue58 I addressed @Cheo7811 in my comment about the fonts and icons. You need to learn some manners first. So many people are trying to help you but you are taking everything as an insult and then you are blaming others of trash talk.

@Cheo7811 with my PR you will have to build the firmware. Do not use the precompiled files. Did it displayed fonts and icons update process?

Cheo7811, 2021-05-07:
@guruathwal , yes, as I mentioned in a previous post, I also compiled your PR and loaded the resulting firmware, I will try one more time with your firmware and the current TFT70 folder and config.ini file.

I have also downloaded the sources of V 027 to compile and try those to recover the printer if all esle fails.

guruathwal, 2021-05-07:
@Cheo7811  while you do so, watch for font, icon and config update process and report if the update process for each of them was displayed.

radek8, 2021-05-07:
@Cheo7811 Did you compile the right branch?
When I get home, I can generate the correct BIN file with a fix to try.

radek8, 2021-05-07:
@BlueBlue58 Your problem is between the keyboard and the chair.

oldman4U, 2021-05-07:
Radek. We try to be nice here🤗

oldman4U, 2021-05-07:
[image]

Time for a update😁

Cheo7811, 2021-05-07:
@BlueBlue58 I don't know if this PR will work for you on the TFT35 but you can give it a try, I have compiled the PR for your display so you can update using it whit the TFT35 folder for your theme and config.ini both from the current repo

[BIGTREE_TFT35_V3.0.27.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6440647/BIGTREE_TFT35_V3.0.27.bin.zip)

BlueBlue58, 2021-05-07:
> [image]
> 
> Time for a update😁

I was updating to v.3.0.27 when the error occured. I now have my printer back and working due to re-saving the config file from Visual Studio Code programme.

BlueBlue58, 2021-05-07:
> @BlueBlue58 I addressed @Cheo7811 in my comment about the fonts and icons. You need to learn some manners first. So many people are trying to help you but you are taking everything as an insult and [...]

Manners? I have plenty thanks. I raised the issue because other people may not be able to use the download and solve the problem on their own. I won't be reporting issues again if you do not listen to what I said originally. I was not being rude, just frustrated at not being listened to because you seem to know best as always. Did I swear, did I insult anyone? No. So stop this blame game because other people are sometimes right. Enough said.

guruathwal, 2021-05-07:
@BlueBlue58 it's ok. It is not your fault. Maybe the people replying to this thread are too arrogant to understand you.