bigtreetech/BIGTREETECH-TouchScreenFirmware #1909: Erroneous control of extrusion distance on TFT movement screen with RepRap

Wombat37, 2021-05-09:
### Description

I use a BMG extruder on my Ender 3 Pro. I find it very difficult to push new filament through the extruder and down the Bowden tube until it reaches the nozzle. Consequently I insert the filament slightly into the extruder and use the extruder step motor under touch panel control to push the filament down to the nozzle. The extrusion distance needed was about 450mm. This worked well on the original Ender 3 board and operating firmware (v1.1.4).

With my newly installed BTT-E3-RRF v1.1 motherboard running RepRap 3.2.2 and BBT TFT35 v3.0 display running firmware 3.027, I have a big problem in extruding a set length of filament. The first time I tried this I extruded several meters of filament instead of the 450mm expected.

Further attempts to use the extrude movement screen have revealed more erratic behavior - it pushes filament forwards (loading) when I've tried to pull it back (unloading). The distances extruded are huge - normally meters when I'm expecting just a few mm.

Further investigations revealed just what was happening...

[image]

In the figure above, we see that the extruder position is currently at 220mm. The change increment is set to 1mm. I would expect that if I press the Load button (or rotate the rotary encoder one click clockwise), the filament would be extruded 1-mm forward and the updated position would become 221mm. If I pressed the Unload button I would expect the filament to be pulled back by 1mm and the updated position would become 219mm. 

This is not what happens...
The length of filament extruded is not the expected change but rather the new total position. In other words, pressing the Load button pushes the filament forward by 221mm and pressing the Unload button pushes the filament, again forward, by 219mm.

If, for instance, I use the button or the rotary encoder to try and move the filament by 10mm (10 presses or clicks), the filament would move forward by 221+222+223+224+225+226+227+228+229+230 = 1155mm or a >100x error! A small quick twist of the rotary encoder can result in many meters of filament being dumped and no way of stopping it other than power-down or reset.

I see no easy way of reversing the direction of movement or getting any other semblance of control over the filament position using this screen. 

Note that the esteps had been correctly calibrated and the G0-3 commands correctly deliver the right length of filament and at the right speed.

**Expected behavior**
I love the look and feel of this extrusion screen and, if it worked, it would be truly great!

I would expect the extruder to advance or retract the filament by the set change increment each time the Load or Unload button is pressed or for each click when the rotary encoder is turned. 

This is what the original Ender 3 screen did.

 

### Hardware Variant

a. BTT-E3-RRF v1.1
b. BTT-E3-TFT35 v3.0

a. RepRap 3.2.2
b. 3.027

bttguy, 2021-05-09:
Hi!
As you can see in #1791 load/unkoad menu is not yet optimised for RRF. I would also be very thankful for a sollution to this as well.

Wombat37, 2021-05-09:
OK - sorry - I did a search but missed this one.

bigtreetech, 2021-06-02:
Hello, This is probably the default extrusion mode is relative after startup of your RRF config.g, But TFT default mode is absolute mode. So in #1962, the gcode of setting mode was added after startup . I have tested that it can work normally. Please confirm! Thank you for your time!

jaysuk, 2021-06-02:
The default extrusion mode for RRF is relative

bttguy, 2021-06-07:
@bigtreetech PR #1962 resolves the movement screen bug, based on my testing. It extruded the correct amount after heating. Thank you!

PR #1962 (2021-06-15): improve RRF
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

### Benefits

### Related Issues
fix #1856 M303 Auto tune PID for RRF
fix #1867 Auto query status to get fan speed
fix #1909 Set relative/absolute mode after startup.
closed 2021-06-15 by bigtreetech