bigtreetech/BIGTREETECH-TouchScreenFirmware #1911: [BUG] First layer height reported on screen incorrectly sometimes

billyd60, 2021-05-12:
### Description

The first layer of my prints are often reported as .4mm Z height, regardless of actual layer height. For example I usually use .2 or .1mm slices. However during the first layer of a print the touch screen often shows .4mm in Z height instead (Even though the nozzle is at the correct .2mm or .1mm height and printing successfully). Importantly, the subsequent layers are reported correctly on the screen. This behavior is isolated to the first layer only and sometimes the Z height is correctly reported in the beginning, which is odd.

### Steps to reproduce

1. Run a print via serial connection from the touchscreen. I suspect the issue is caused by a startup script I use the at the beginning of all my prints, it is automatically added to the gcode print file by my slicer and is to prime the nozzle for printing:

G28 ; home all axes
G28 X Y ; Home X and Y back to zero after safe home to bed center for bltouch
G92 E0 ; zero extruder
G1 Z0.4 ; position nozzle (This is the line I think is tricking the screen)
G1 E10 F300 ; prime nozzle
G1 X5 Y5 Z0.15 F1200 ; slow wipe
G1 X10 Y10 Z0.5 F1200 ; lift
G92 E0 ; zero extruder

2. Observe reported first layer height on the screen. It should read a value equal to the first slice thickness. However it often reports .4mm regardless of the actual slice thickness used. I believe it is picking this up from my startup script accidentally. (shown in step 1). It is not always picking up the Z height before the main print begins.

3. Observe subsequent layer heights are reported correctly

**Expected behavior**
The screen should show the actual z height at the beginning of the print.

**Actual behavior**
I suspect sometimes the screen instead reads the Z position of the nozzle when it primes in my startup script. It needs to read the actual Z once the main print starts. Occasionally the screen does actually get the Z height of the first layer correct, I don't know why it is not consistent in behavior.

### Hardware Variant
BTT TFT70 V3
### TFT Firmware Version & Main Board Firmware details
,27 firmware. Latest Marlin bugfix 2.0.x

### Additional Information

kisslorand, 2021-05-12:
What version of TFT FW do you use? The layer height display was recently reworked (build from 6.V.2021). Please check if the issue is present with the latest FW build.

billyd60, 2021-05-12:
> The layer height display was recently reworked (build from 6.V.2021). Please check if the issue is present with the latest FW build.

Ok is the latest build the Master branch? Just want to make sure.

kisslorand, 2021-05-12:
Yes.

billyd60, 2021-05-12:
Thank you, the latest build fixes this issue.

billyd60, 2021-05-17:
Ok I have noticed that while this issue is improved it is not completely fixed. I am using prusaslicer and the way that slicer uses supports it will occasionally print support at between regular layer intervals. For example it might print supports at a layer height of 1.76 when printing at .2mm regular intervals.

So what is happening is the screen will pick up that odd layer height, and continue to report it even after the printer has advanced to the next regular interval (1.8 in our example). As before, it corrects it on the next subsequent regular layer interval.

This is a very minor issue, but I thought I should reopen this as the bug is not quite squashed yet.

oldman4U, 2021-05-17:
Maybe attaching a simple example gCode could help.

kisslorand, 2021-05-17:
Working on it...

billyd60, 2021-05-19:
> Working on it...
> 
> Please check PR #1927 and give feedback.

This has fixed the issue. Thank you!

PR #1927 (2021-05-19): Layer display   more robust
### Requirements

BTT TFT or MKS TFT
### Description

It's a more robust approach to layer height display update. It updates micro layer change only if before that change there was a bigger layer change defined by `LAYER_DELTA`. For consecutive micro layer changes (vase mode) it updates the layer height only at each `LAYER_DELTA` change.

### Related Issues

Fixes #1911
closed 2021-05-19 by billyd60