bigtreetech/BIGTREETECH-TouchScreenFirmware #1912: TFT35 E3 V3 firmware does not work

RocksaltX, 2021-05-12:
Send firmware to TFT, screen is locked up.

### Steps to reproduce

1. Copy BIGTREE_TFT35_V3.0_E3.27.x.bin to flash card, copy TFT folder, copy config.ini
2. boot
3. Frown

**Expected behavior**
Work,,,
**Actual behavior**
Screen froze, can see BTT logo in the status area, touch does not work, holding button in for many seconds does not bring up marlin mode.

### Hardware Variant

TFT35 E3 V3

BIGTREE_TFT35_V3.0_E3.27.x.bin, Marlin 2.0 Bug may 2011

### Additional Information
Reverting back to a .zip I downloaded on 4/4/21 works properly.
I did reset the TFT in the menu and tried it again with same issue.

kisslorand, 2021-05-12:
Did you try to reset the TFT touch calibration with "reset.txt" file placed in the root of your SD card?

RocksaltX, 2021-05-12:
yes, I did. That just resets the touch calibration. Even when pressing the rotary dial to get to marlin mode it does not work. It's locked up with this .bin file. As I stated, if I revert back to a saved version it's working.

oldman4U, 2021-05-12:
Is this master or v27 release version? Self compiled or pre compiled?

oldman4U, 2021-05-13:
All your previous postings are about TFT35 v2. Have you changed the TFT hardware to a new TFT35 E3 V3?

oldman4U, 2021-05-13:
Just updated 2 TFT35 E3 v3 with the precompiled version from today without any issue.

Please help the community and close the ticket once you do not need it anymore. Thank you

RocksaltX, 2021-05-13:
Well shoot-fire. Sorry to waste your time. I downloaded the version today and went at it again. Same result. I made a mistake and did not copy the TFT folder, at least it seems that way. I loaded the new firmware and config.ini and had the same problem. I then did it again with the TFT folder and it worked. I am 99% sure I loaded the TFT folder when I opened this request or when I downloaded the TFT files on 5/11/21.
Again, I apologize.

closed 2021-05-13 by RocksaltX

Flipsand, 2021-08-11:
Maybe you got that ZIP (Link) from April because i am having actual the same issue ... and i do use config, BIN and TFT35 folder but i have no old ZIP to get it working again.

Flipsand, 2021-08-11:
Maybe you got that ZIP (Link) from April because i am having actual the same issue ... and i do use config, BIN and TFT35 folder but i have no old ZIP to get it working again. Would be great getting it to work again. THX in advance.