bigtreetech/BIGTREETECH-TouchScreenFirmware #1918: [BUG] (short description)

wigw, 2021-05-13:
Hi, I ask you to reverse the icons. The bottom line is that when you reverse the Z axis in the menu, only the microstep icon is not reversed.

### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-05-13:
Hi.

Reverting the axis in the menu will change the direction of the motors which drive your printer, but it will NOT change the Icons in any way.

Please send pictures or a movie which shows the reported issue.

wigw, 2021-05-14:
[image]
[image]

wigw, 2021-05-14:
Here are my settings in config.ini
#### Inverted Axes For Move Menu And Manual Leveling Menu
# Invert axis button function in Move menu & Manual Leveling menu in order it matches the actual axis movement.
#
# NOTE: The Y-axis of different printer (move hotbed or move nozzle) move in different directions.
#       So Y-axis leveling invert can't follow up invert_axis[Y_AXIS].
#       We separate a single variable "LY"(Leveling Y-axis) to deal with the y-axis leveling movement direction.
#
#   Format: [X<option> Y<option> Z<option> LY<option>]
#   Options: [enable: 1, disable: 0]
invert_axis:X0 Y0 Z1 LY0

wigw, 2021-05-14:
I'm talking about the Z axis

digant73, 2021-05-14:
with the exception of the icons reverted, was the z offset value changed properly?

wigw, 2021-05-14:
All movements plus reverse work in the correct direction, icons in the correct direction, only the microstep icon has not been reversed or the direction of microstep movement. I made a screenshot of a microstep.

oldman4U, 2021-05-14:
So you mean 0.01 moves into another direction than 0.1?

wigw, 2021-05-14:
I did not understand you.
I only had a babystep on BL Touch not reversing.
See photo above

digant73, 2021-05-14:
> All movements plus reverse work in the correct direction, icons in the correct direction, only the microstep icon has not been reversed or the direction of microstep movement. I made a screenshot of a microstep.

do you have the same problem also with the Mesh Tuner?

guruathwal, 2021-05-14:
In probe offset menu the icons depict nozzle movement instead of bed movement. to change these icons you can turn off FRIENDLY_Z_OFFSET_LANGUAGE in configuration.h and recompile the firmware.

digant73, 2021-05-14:
It is needed to handle Z invert as it was done for XY in Move menu. Yes the icons must remains the same. Just swaping the labels and a small change in the code to handle the inverted logic. Pressing on left icon (lowering the nozzle), an increase of z offset is produced (instead of a decrease) etc... The same must be done for babystep. For Mesh Tuner, MBL it is enough just to swap the labels (the code already handles the inverted logic). I will fix that in the WE

oldman4U, 2021-05-14:
Maybe we should use Closer and Further instead of Up and Down.

guruathwal, 2021-05-14:
> It is needed to handle Z invert as it was done for XY in Move menu. Yes the icons must remains the same. Just swaping the labels and a small change in the code to handle the inverted logic. Pressing [...]

Lowering the Nozzle and Moving the bed Up, both means the same thing. Inverting of Z axis is a little bit different from inverting X and Y axis.  For printers with bed moving up/down, a down arrow means moving the bed down which is increasing the Z height. A Nozzle Icon with down arrow created confusion.

digant73, 2021-05-14:
yes basically it is the meaning of the nozzle icon. For printers with moving bed up/down it will be better to use the "+" and "-" icons. Otherwise they have to see the nozzle icons as moving the bed close/far as also reported by oldman.
In any case the inverted z axis has to be handled. Just testing now

digant73, 2021-05-14:
@wigw I just checked the logic in all the menus providing Z movement.
With the exception of the Mesh Tuner (a bug fixed in my current PR), in case of inverted Z axis (in config.ini) you get the expected behavior in all menus.
The problem you highlighted is simply on the display of nozzle icons while, in your case, bed icons should be more appropriate. Unfortunately as reported also by guruathwal, no bed icons are available. However, if are are able to compile the code by yourself you can simply turn off FRIENDLY_Z_OFFSET_LANGUAGE in configuration.h and recompile the firmware. All the menus providing Z movements will then display the icons "-" and "+" with labels "decrease" and "increase" respectively. That will give you a better representation of the logic

oldman4U, 2021-05-14:
Or to make your own bmp's and replace those coming with the firmware.

wigw, 2021-05-15:
Thanks.
I should probably close the problem?

oldman4U, 2021-05-15:
In case you have a solution you can close the ticket. Once closed we can still use it to discuss a better solution.

wigw, 2021-05-15:
unfortunately there is no solution.

closed 2021-05-15 by wigw

oldman4U, 2021-05-15:
Why not following what Gurmeet recommended or make your own buttons? Two fast solutions for your problem.

wigw, 2021-05-15:
It's just that I'm already used to this arrangement of icons.

oldman4U, 2021-05-15:
Down and Up is also wrong in this case. How did you change it?

Changing the buttons is easily done on Windows.

PR #1928 (2021-05-19): friendly z offset icons/labels for inverted Z axis
If FRIENDLY_Z_OFFSET_LANGUAGE is enabled (in Configuration.h) and inverted Z axis is configured (in config.ini), icons "arrow up"/"arrow down" and labels "up/"down" are used in menus Babystep, MBL, Mesh Tuner and Z offset instead of icons "nozzle down"/"nozzle up" and labels "down"/"up".

fixes #1918
fixes #1926

**PR STATE:** ready to merge