bigtreetech/BIGTREETECH-TouchScreenFirmware #1923: TFT35 E3 V3.0 load filament problem

Grzesinski, 2021-05-17:
### Description

TFT35 E3 V3.0 graphical menu load filament problem

After the filament is fully loaded using the Load / Unload graphical TFT menu, it displays the message "Print is Paused for User Continue?". The message is looped and returns when the key [OK] is pressed. But after several [OK] presses, further filament loading begins. Then same message returns. There is no way to end this process in graphical mode.

At the same time, a message displayed in the Marlin window
Resume Options: Purge More, Continue.
After choosing "Continue" printer resume work.

Unloading is working properly.

### Steps to reproduce

1. Menu / Load Unload 
2. Heat hotend
3. Load filament

**Expected behavior**
Message with two options to choose (eq. Purge More, Continue) or stop loading after [OK] is pressed.

**Actual behavior**
Looped message described above.

### Hardware Variant
BIGTREETECH SKR MINI E3 V2.0 Marlin 2.0.7.2 (2020.12.10 firmware-bltouch.bin)
TFT35 E3 V3.0 (2021.05.17 BIGTREE_TFT35_V3.0_E3.27.x.bin)

### Additional Information
[image]
[image]

digant73, 2021-05-17:
update Marlin to last 2.0.8.1 released on 2021-05-15 and try again please. Lots of bugs were on Marlin 2.0.7.2

oldman4U, 2021-05-17:
It works also with older Marlin as soon as you have all features activated in Marlin listed in the comfig.ini or ReadMe.

oldman4U, 2021-05-17:
Btw. It is required also for new Marlin versions.

Grzesinski, 2021-05-18:
In
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/Marlin-2.0.7.2-SKR-mini-E3-V2.0/Marlin/Configuration_adv.h
options are set.

I didn't compile my own versions of Marlin, I used only the newest versions available on github bigtreetech. If I remember correctly, the problem was also in earlier versions of the software. I did not notice any other problems with the cooperation between TFT and SKRmini. In my opinion, the cooperation of both modules (with the factory software) should run smoothly.

oldman4U, 2021-05-18:
On my tiny screen of my phone I can see that almost nothing is enabled which is mention in the documentation available in this repository. If you can not compile your own firmware you should ask someone able to do so in the repository of the mainboard or in a user group related to it. But I would strongly recommend to learn it to compile firmware and there are many movies and documentation's available showing this.

Please help the community and close this ticket.

Thank you

guruathwal, 2021-05-19:
> In
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/Marlin-2.0.7.2-SKR-mini-E3-V2.0/Marlin/Configuration_adv.h
> options are set.
>
> I didn't compile my own versions of [...]

@Grzesinski `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT` are not enabled in this file. you need to enable them and compile the firmware yourself.

Grzesinski, 2021-05-19:
OK. I will compile and check it.
Can anyone confirm that the TFT / LOAD function is working properly for him?

kisslorand, 2021-05-19:
I confirm it is working OK on my machine (latest TFT FW, latest Marlin bugfix FW).

Grzesinski, 2021-05-20:
Thanks for support.

closed 2021-05-20 by Grzesinski