bigtreetech/BIGTREETECH-TouchScreenFirmware #1931: [BUG] (short description)

discip, 2021-05-19:
### Description

black screen after updating MKS_TFT28_V4_0 firmware

### Steps to reproduce

1. compile latest firmware
2. flash it

**Expected behavior**

working at least as good as before flashing

**Actual behavior**

last screen after flashing freezes and stays, until printer is switched off and after reboot the screen stays black after showing loading

### Hardware Variant

MKS TFT 28 V4.0
### TFT Firmware Version & Main Board Firmware details

latest Marlin bugfix
which is working with the TFT firmware from May 3. from this branch

### Additional Information

I left **`Configuration.h`** as is.
<details>
  <summary>
    <b>not working setup</b>
  </summary>

<details>
  <summary>
    <b>config.ini</b>
  </summary>

```ini
#--------------------------------------------------------------------
#
# DEFAULT CONFIG FILE FOR BigTreeTech TFT CONTROLLERS
#
#--------------------------------------------------------------------

#--------------------------------------------------------------------
# Supported TFT Variants
#--------------------------------------------------------------------
#
# BIGTREE_TFT35_V1_0    / BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2
# BIGTREE_TFT35_V2_0    / BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0
# BIGTREE_TFT35_B1_V3_0 / BIGTREE_TFT43_V3_0 / BIGTREE_TFT50_V3_0
# BIGTREE_TFT70_V3_0    / BIGTREE_TFT28_V1_0 / BIGTREE_TFT28_V3_0
# BIGTREE_TFT24_V1_1    / MKS_TFT32_V1_3     / MKS_TFT32_V1_4
# MKS_TFT28_V3_0        / MKS_TFT28_V4_0
#
# Firmware source: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

#--------------------------------------------------------------------
# Supported Marlin Firmware Versions
#--------------------------------------------------------------------
#
# Minimum Marlin firmware version: 2.0.8.1
# Distribution date: 2021-05-15
# Firmware source: https://github.com/MarlinFirmware/Marlin/releases
#
# In order the TFT firmware is able to provide all of its functionalities/features,
# ensure that the following options are enabled in Marlin firmware.
#
# General options:
[... 1741 lines cut ...]
#   Value range: [min: 3, max: 75 characters]
start_gcode:G28 XY R20\n

#### End Gcode
# This gcode will runs after a print is completed if `end_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
end_gcode:M104 S0\nM140 S0\nM107\nM18\n

#### Cancel Gcode
# This gcode will runs when a print is canceled if `cancel_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
cancel_gcode:G91\nG0 F6000 Z10\nG90\nG0 F10000 X0 Y270\nG1 Z0\n
```

</details>
<details>
  <summary>
    <b>platformio.ini</b>
  </summary>

same as 'not working setup'

</details>

</details>

kisslorand, 2021-05-19:
Happened to me also today. A `reset.txt' on the SD Card solved the problem on my side.

Sonicboom247, 2021-05-19:
happened to me as well but my screen is white, i call it the WSOD..lol. with last release and with this one as well using MKSTFT28V3, last time i had to flash my stock Artillery sidewinder firmware to get the screen back.  Odd though since the MKS directory where the icons and fonts are was renamed to MKS.CUR as well as the bin and config files all appended .cur

kisslorand, 2021-05-19:
What envelope did you use to compile the TFT FW? Or did you use a precompiled one?

Sonicboom247, 2021-05-19:
> What environment did you use to compile the TFT FW? Or did you use a precompiled one?
> 
> For Artillery you must use "MKS_TFT28_V4_0".

WOW you fixed mine!! a million thanks! so odd my TFT says V3 on it when flashing so i must have assumed it was a V3.. who knew..

kisslorand, 2021-05-19:
OK. Now don't forget to reload your "config.ini".

discip, 2021-05-19:
@kisslorand
Good evening Sir,
thank you very much! 👍
The **`reset.txt`** worked for me as well!

closed 2021-05-19 by discip