bigtreetech/BIGTREETECH-TouchScreenFirmware #1934: [BUG] continuous error: Found Outdated Data

Sonicboom247, 2021-05-20:
### Description

not sure if documentation is off or i am, I have copied the THEME_The Round Miracle Menu Material theme, MKSTFT28.bin for V4, and config.ini renamed font to fonts and bmp to icons and placed them in the MKS dir and even though the MKS directory gets a .CUR extension so i know it is being read, i continue to get invalid keywords and outdated Data.

### Steps to reproduce

1. follow directions above, not sure if specific to THEME_The Round Miracle Menu Material theme
2. [Second Step]
3. [and so on...]

**Expected behavior**
successful flash

**Actual behavior**
invalid keyword errors for numerous settings it seems

### Hardware Variant

MKSTFT28 V4 Marlin 2.0.8.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Sonicboom247, 2021-05-20:
ok so oddly enough, i switched it back to font and bmp and it is now happy...

kisslorand, 2021-05-20:
**MKSTFT28.bin** , **config.ini** and **language.ini** must go to the root of your SD Card.

oldman4U, 2021-05-20:
Please help the community and close the ticket once you do not need it anymore. Thank you

Sonicboom247, 2021-05-20:
> **MKSTFT28.bin** , **config.ini** and **language.ini** must go to the root of your SD Card.

Thank you @kisslorand , did that about 20 times and it was not until i renamed the two folders back to bmp and font that is updated.  might be only affecting the Round Miracle theme.

closed 2021-05-20 by Sonicboom247