bigtreetech/BIGTREETECH-TouchScreenFirmware #1944: [FR] (disable touch screen on marlin mode)

devildant, 2021-05-23:
**Is your feature request related to a problem? Please describe.**
Currently the area at the top right in marlin mode, is clickable and activates the marlin menu.
It also sometimes happens that when the screen is in standby (brightness at 0) the click on the screen (outside the area at the top right) triggers a click.

the dimmer mode being a simple visual action, making the screen clickable in marlin mode can be quite painful especially on a tft35 E3 V3 where there is the rotary button, the touch of the screen in marlin mode is not therefore not useful

**Describe the solution you'd like**

add an option to completely disable the touch screen in marlin mode (for screen with rotary button)

devildant, 2021-05-23:
just an option to block the menu action of the marlin mode from the screen will be sufficient :)

devildant, 2021-05-23:
https://user-images.githubusercontent.com/8700106/119268425-3674bf80-bbf3-11eb-96ac-a0097e0dab86.mp4

to make it simple add an option to remove the click box on the screen (not reduce it remove it)
(see video)

radek8, 2021-05-23:
This allows you to control Marlin mode if you do not have an encoder. Swipe up or down to simulate the rotation of the encoder. No sw adjustment required, just don't click on the right edge :-)

devildant, 2021-05-24:
haha, I was waiting for this answer: p

I never use this function because I have the rotary knob.
from time to time I click in the wrong place (to get out of sleep mode), or I get a false positive (it's rare but it happens),  and it bothers me
I currently disabled this by commenting directly in MarlinMode.c
```
    #if LCD_ENCODER_SUPPORT
      sendEncoder(LCD_ReadTouch());
      
      //disable zone touch area on marlin mode
      // if (LCD_BtnTouch(LCD_BUTTON_INTERVALS))
      //   sendEncoder(1);
    #endif
```
I did the feature request if it was of interest to other people and to do it cleaner than my dirty solution

radek8, 2021-05-24:
Your solution is good and will probably work for you

devildant, 2021-05-24:
so i will close my feature request, at worst if someone needs it my changes are there

closed 2021-05-24 by devildant