bigtreetech/BIGTREETECH-TouchScreenFirmware #1945: [FR] (Add an option to block the screen on a marlin or graphic mode, remove the switch mode)

devildant, 2021-05-23:
**Describe the solution you'd like**
add an options for completely disable the switch mode, to stay in the default mode and no longer be able to change mode

oldman4U, 2021-05-23:
How would you leave this mode in touchscreen mode and Marlin mode?

radek8, 2021-05-23:
This is a big advantage of this display and you want to cancel it?

devildant, 2021-05-24:
@radek8 I bought a TFT35 E3 V3, for two things, changing the text color and the sleep function.

I never use the touchscreen (it's not even connected to my cm because, it doesn't support it anyway)
in this kind of case the switch is useless

@oldman4U the goal is never to leave marlin mode (for motherboards that don't support graphic mode)

I have for the moment made the modification dirty, by commenting the loopCheckMode in MarlinMode.c
    ```
#if LCD_ENCODER_SUPPORT
      sendEncoder(LCD_ReadTouch());
      
       if (LCD_BtnTouch(LCD_BUTTON_INTERVALS))
         sendEncoder(1);
    #endif
    // disable swith menu 
    // loopCheckMode();
```

oldman4U, 2021-05-24:
You found a good solution for you. All others can simply avoid to press the encoder for more than one second. This helps to avoid switching the mode. I had the problem the other way around and also found a solution.

[image]

radek8, 2021-05-24:
> @radek8 I bought a TFT35 E3 V3, for two things, changing the text color and the idle function.
> 
> I never use the touchscreen (it's not even connected to my cm because, it doesn't support it anyway)
> in this kind of case the switch is useless
> 
> @oldman4U the goal is never to leave marlin mode (for motherboards that don't support graphic mode)
> 
> I have for the moment made the modification dirty, by commenting the loopCheckMode in MarlinMode.c
> 
> ```
>    #if LCD_ENCODER_SUPPORT
>       sendEncoder(LCD_ReadTouch());
> 
>        if (LCD_BtnTouch(LCD_BUTTON_INTERVALS))
>          sendEncoder(1);
>     #endif
>     // disable swith menu 
>     // loopCheckMode();
> ```

Your solution is good.
I doubt that there will be anyone else who bought a touch screen and would like to disable touch mode ...

radek8, 2021-05-24:
> You found a good solution for you. All others can simply avoid to press the encoder for more than one second. This helps to avoid switching the mode. I had the problem the other way around and also found a solution.
> 
> [image]

I've always wondered why you disassembled the encoder :-)

oldman4U, 2021-05-24:
Removed the reset the same time. ;-)

devildant, 2021-05-24:
@oldman4U I also removed the reset button, too dangerous

so i will close my feature request, at worst if someone needs it my changes are there

closed 2021-05-24 by devildant

oldman4U, 2021-07-31:
Update. Seems digant73 implemented it in his last PR.

Happy printing