bigtreetech/BIGTREETECH-TouchScreenFirmware #1946: [BUG] (short description)

Zoli-user, 2021-05-23:
Bigtreetech E3 RRF V1.1 and TFT35 Ender 3 V2 issue
### Description

### Steps to reproduce

1. BTT-E3 RRF V1.1 have only the 2.0.7.x firmware and the TFT35 have the 2.0.8.x firmware.
2. Install the latest firmware version for the motherboard and the TFT35 but the printer is not working. 
3. Install the 2.0.7 firmware version for the motherboard and install the 2.0.8 the tft35 but the printer is not working.
4. Install the RepRap firmware the motherboard and the TFT35 and the printer is not working
5. Use the original Ender 3 V2 display and the new BTT-E3 RRF V1.1 and the display have a blank screen and beep continuously.
6. I try to install the older 2.0.7 firmware for the TFT35 and the actual 2.0.7 for the Motherboard the error is same. The printer not working normally. BLTouch is off, the Z axis zero point out of range.

**Expected behavior**

I would like to see the new Marlin 2.0.8.x firmware by the BTT-E3 RRF V1.1. The original BTT-E3 RRF V1.1 motherboard and the BTT TFT35 is a same product, but the software is different. Why?

**Actual behavior**

The printer is on and every system is ready. The TFT communicated the motherboard. After power on I have a several error notification. Bed level error, BLTouch not working, the Z axis zero point is out of the printer range.

Inside the TFT35 config.ini file:
#--------------------------------------------------------------------
# Supported Marlin Firmware Versions
#--------------------------------------------------------------------
#
# Minimum Marlin firmware version: 2.0.8.1
# Distribution date: 2021-05-15

Super. :-( the BTT-E3 RRF Github page have only the 2.0.7.x motherboard firmware.... ????????????????

What works: 
1. Communication between the TFT and the motherboard
2. Moving the axis step by step
3. Touch screen
4. Every menu item

What doesn't work:
1. BLTouch
2. Axis reference run
3. Bed leveling
4. Print :-)
5. Wi-FI connection
 

### Hardware Variant

BTT TFT35 E3 V3.0 ZZB000378
BTT-E3 RRF V1.1 ZZB000521
### TFT Firmware Version & Main Board Firmware details

BTT TFT35 E3 V3.0 ZZB000378 Use the 2.0.8.x. Download this site: BIGTREETECH-TouchScreenFirmware
BTT-E3 RRF V1.1 ZZB000521 Use the 2.0.7.x Download this site: bigtreetech / BTT-E3-RRF

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2021-05-24:
Hello, what is the specific phenomenon of not working?  “No printer attached!” on TFT ?

oldman4U, 2021-05-24:
And please verify the version number of the TFT firmware. 2.0.8.1 is Marlin and not TFT.

Zoli-user, 2021-05-24:
Hello. First of all, I am grateful you are answering.
No, after power on the system the connection is establised. 
Step by step:
1. Power on :-)
2. The display established the connection for the printer
3. Apparently everything is ready.

But the printer can’t run the reference “Z-homing” for some reason the printer zero point is up to 60mm. I press the home run icon the printer go to the X axis end switch and I have an error notification. 
Another things after power on the printer I have some notification. Ca… bed level error…..etc.
I go inside the menu and try to turn on the BLTouch…. But I can’t.
I try to move manualy the axis and its work. I using the small steps and its moving.

First of all, please tell me which firmware should I try to the TFT and the Motherboard. Tonight I try to install again the display and the motherboard and send some picture fot You the issues. 
I’m only the simple user. I can’t handle the VisualStudio. Just find the firmware.bin files copy the SD card and install.
I know the display have more than one file I see Your user manual. And I see the Marlin have a 2.0.8 firmware but only the source code. I try to create the bin file but I have a many error. This method is doesn’t work for me. 

A solution might be to send me what to install on the machine
and I can test and give feedback on it. My Ender 3 V2 is a good printer and I see the Bigtreetech tools are also quality products. I'm sure I'm ruining something. I’m just an avid amateur who loves these things.

Interesting thing I change back the original hardware and the printer is working fine…

radek8, 2021-05-24:
You knit the Marlin RRF and TFT together.
they are 3 different FWs.

Learn to control VS. Custom compilation of Marlin is a must.

jaysuk, 2021-05-24:
or use RRF where no compilation is required
Thanks,

Jay

radek8, 2021-05-24:
the new Marlin is here.
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

jaysuk, 2021-05-24:
and the latest RRF is here
https://github.com/gloomyandy/RepRapFirmware/releases/tag/v3.3-RC2%2B1_1
Thanks,

Jay

radek8, 2021-05-24:
unfortunately TFT is not yet fully compatible with RRF

radek8, 2021-05-24:
FW for TFT is here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

jaysuk, 2021-05-24:
it's mostly there and can handle the majority of the jobs needed when
you're stood at your machine and for everything else there are macros.
Thanks,

Jay

oldman4U, 2021-05-24:
@bttguy

Could you please help 

Thank you

Zoli-user, 2021-05-24:
[image]

When I installed the firmware and start the system this is a first notification what I have…

radek8, 2021-05-24:
> [image]
> 
> When I installed the firmware and start the system this is a first notification what I have…

Install the latest Marlin with bug fixes.
Uncomment option:
AUTO_REPORT_POSITION (in Configuration_adv.h)

radek8, 2021-05-24:
> [image]
> 
> When I installed the firmware and start the system this is a first notification what I have…

@guruathwal, if I'm not mistaken, this error should not occur.
if Marlin does not support the M154 command, the TFT should not use it.

guruathwal, 2021-05-24:
The M154 error is already fixed after #1940

radek8, 2021-05-24:
@guruathwal, Thank you.
@Zoli-user , Then you don't have the latest FW on TFT

bttguy, 2021-05-24:
I can also help you in case you would like to try RRF, I am not that familiar with Marlin since I switched. As Jay said, most of the essential features work, and the rest is handeled by RRF and its web interface from you PC/phone :)

oldman4U, 2021-05-24:
@Zoli-user seems to use rrf and I believe you can expect some new rrf related stuff soon. So maybe you can help to fix the 3 outstanding rrf tickets by testing possible fixes.

bttguy, 2021-05-24:
I would be glad to test new features/bugfixes, and also can quide new users in setting up their screen. I checked the RRF related tickets, but they are known limitations, more like FR-s than bugs.
Sadly I cant code either, and the most I can do is that I can send resources to the developers to see what and how the gcode is different in Marlin and RRF.
Until that I also included these "bugs" to the RRF enhancement ticket I made. I want to add that the support is already awesome for RRF users compared what we had before, I dont complain here, I try to do my best to help :)

Zoli-user, 2021-05-25:
Hello,

Today 8:00pm I start the new installation for the TFT and the Motherboard by RRF latest firmware.
If You have time please follow me and "help". I show You by pictures the process.

Let us see what firmware ready for install:
TFT35 E3 V3 display
1. Bigtree_TFT35_V3.0_E3.27.x.bin
2. config_rrf.ini
3. TFT35 folder

Motherboard BTT-E3 RRF v1.1
1. firmware-stm32f4-esp8266wifi-3.3rc2+1_1.bin
2. firmware sub folder inside this file: DuetWiFiServer-stm32f4.bin

bttguy, 2021-05-25:
@Zoli-user How is it going? Just a quick question, did you renamed the firmware.bins and the config.ini + did you included the required folder structures with the sys files and the example configs from the online configurator? + you need a board.txt with some lines.

Zoli-user, 2021-05-25:
Of course. Every file name renamed the correct words. And every subfolder with correct file are ready. Meanwhile I try to build a 2.0.8.1 Marlin alternate version. If the RRF is not working I try to install the Marlin if I know and success the firmware build...

jaysuk, 2021-05-25:
why not join our discord channel? we'll support you with any RRF issues
https://discord.gg/uS97Qs7

Thanks,

Jay

Zoli-user, 2021-05-25:
I try to create my own firmware. Not working :-(

[image]
[image]

Zoli-user, 2021-05-25:
Hello. Now I start the BTT E3 RRF V1.1 motherboard install
[image]

Zoli-user, 2021-05-25:
Now the new MB is installed and next the TFT
[image]

Zoli-user, 2021-05-25:
[image]

Zoli-user, 2021-05-25:
[image]

Zoli-user, 2021-05-25:
[image]

Zoli-user, 2021-05-25:
Now first off all the TFT install. Follow by ruiraptor YouTube video
[image]

Zoli-user, 2021-05-25:
Now after the MB firmware installed I have this screen
[image]

Zoli-user, 2021-05-25:
After this process, restart the printer and I have a connection the TFT and the MB

[image]

bttguy, 2021-05-25:
Did you connected to your WiFi yet? You should see your IP for DWC(web interface)?

Zoli-user, 2021-05-25:
At this moment the printer axis are not moving. The next step the wi-fi settings. Use an USB cable and run the Pronterface software.

[image]

Zoli-user, 2021-05-25:
The printer is online.

[image]

Zoli-user, 2021-05-25:
After connecting type M115. I see the printer online and see the version number.

[image]

Zoli-user, 2021-05-25:
Next type M997 S1. The system update the modul.

[image]
[image]

Zoli-user, 2021-05-25:
Type a correct command and the printer have an IP.

[image]

jaysuk, 2021-05-25:
I'm not quite sure what you're trying to document here

Zoli-user, 2021-05-25:
@jaysuk You can find here to the wifi settings code. See 14:32
https://www.youtube.com/watch?v=2ytFdP2_Myo&t=5s

jaysuk, 2021-05-25:
But again, what are you trying to do? Or what are you trying to show? Or
what are you trying to document?
We already have the whole process documented here
https://teamgloomy.github.io/ that at least 2.5k users have accessed and
followed

Zoli-user, 2021-05-25:
Finally type IP the browser and You see the printer menu.
After entering the SSD and pass I had to wait 5 min for the IP to appear.

[image]

Zoli-user, 2021-05-25:
And last how to use the software. User manual is here:
https://duet3d.dozuki.com/Wiki/Duet_Web_Control_v2_and_v3_%28DWC%29_Manual

jaysuk, 2021-05-25:
Ok, this thread obviously is nothing to do with the TFT firmware and should
be closed

Zoli-user, 2021-05-25:
And finally this is the first error. 

[image]

Zoli-user, 2021-05-25:
Touch this icons and nothing happened. The BLTouch not “moving”

[image]

Zoli-user, 2021-05-25:
In the website G29 error Z probe 0 not found.
[image]

Zoli-user, 2021-05-25:
For some reason it considers this height as zero point. 

[image]

Zoli-user, 2021-05-25:
How to setup a real Z null point and the BLTouch? This is a question today. :-)

Zoli-user, 2021-05-25:
And what is the PLA Filament config.g code.

[image]

Zoli-user, 2021-05-25:
My settings is correct every file I use find the Bigtreetech page and I follow the instructions. Copy all Ender 3 V2 config file to the right place.

Zoli-user, 2021-05-25:
And I use only the latest release.

Zoli-user, 2021-05-25:
This is the config.g file inside the sys folder:

; Axis Limits
M208 X0 Y0 Z0 S1                               ; set axis minima
M208 X230 Y230 Z250 S0                         ; set axis maxima

; Endstops
M574 X1 S1 P"xstop"                            ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"ystop"                            ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S1 P"zstop"                            ; configure active-high endstop for low end on Z via pin zstop
;M591 D0 P7 C"e0stop" S1  R50:130 L7.0 E10     ; BTT Smart Filament Sensor
;M591 D0 P1 S1                                 ; Filament runout Sensor

; Z-Probe  Bltouch
;M950 S0 C"servo0"                              ; create servo pin 0 for BLTouch
;M558 P9 C"^probe" H5 F120 T6000                ; set Z probe type to bltouch and the dive height + speeds
;G31 P500 X-40 Y-10 Z2.95                      ; set Z probe trigger value, offset and trigger height
;M557 X10:180 Y10:210 S40                       ; define mesh grid

Zoli-user, 2021-05-25:
I found something. Maybe this is not good.
The BLTouch code is grey.

[image]

jaysuk, 2021-05-25:
Those settings are incorrect as you have commented out the settings.
This is the last time I will say this. This is the incorrect place to be
raising your issues as they are not related to the TFT firmware. You need
to seek support elsewhere. I have given you the link to the STM RRF support
channel on discord. If you are unwilling to use that then I suggest either
the duet forum or one of the RRF Facebook groups.

Again, can an admin close this issue as it is unrelated to the TFT firmware

Zoli-user, 2021-05-25:
Ok

oldman4U, 2021-05-25:
@Msq001

Please close this ticket as it is unrelated to the TFT firmware.

Thank you

oldman4U, 2021-05-25:
Thank you @jaysuk

closed 2021-05-26 by Zoli-user