bigtreetech/BIGTREETECH-TouchScreenFirmware #1954: [Q] documentation error

radek8, 2021-05-29:
@oldman4U 
You have an error in the table
SKT V1.3 cannot be connected via the AUX1 connector on the EXP3 display!

[image]

radek8, 2021-05-29:
The same applies to SKR V1.1

[image]

radek8, 2021-05-29:
The AUX1 connector is for serial connection to the MKS display

oldman4U, 2021-05-29:
Hi radek8

Thank you. This would be correct then, right? Because CR-10 is also wrong.

[image]

radek8, 2021-05-30:
Yes, that's right, but why are the lines duplicated?

oldman4U, 2021-05-30:
🤪

closed 2021-05-30 by radek8

oldman4U, 2021-06-02:
Hi radek8.

Thomas White replied this: Not dupes, those boards have connectors that can use either setup just not both.

I do not have such a board to verify so I will finally trust him.

radek8, 2021-06-11:
Oldman4U, you returned that error to the spreadsheet.
I looked at the layout of the PINs.
Yes, these boards can be connected via one 10-pin connector with the CR10_STOCKDISPLAY driver.
But then it must be connected via the EXP1 connector, but not via the AUX1 connector. The AUX1 connector on the SKR board is not compatible with the EXP3 connector on the TFT. If someone could connect it (EXP3 has 10 pins, but AUX1 only 8 pins) it could have fatal consequences.

If the CR10_STOCKDISPLAY driver is selected, then it would be possible to connect EXP1 in the SKR, with the EXP3 connector in  the TFT.
The same would apply to SKR V1.4 and SKR V1.4Turbo