bigtreetech/BIGTREETECH-TouchScreenFirmware #1961: [BUG]After preload filament print don't work

mytique38, 2021-05-31:
After pre-load filament - print don't start 

### Steps to reproduce Issue

1. Power on Printer
2. Prepare new filament 
3. Launch in the menu "load filament, heat 180 ° and filament go to the buse and extrude (work's well) 
4. after loading the filament when I want to print I launch the printing on a SD card, nothing happens, to solve the problem I have to turn off / on the printer and launch the printing and print works well 

One fix exist for this issue ? 

### Hardware Variant

BIGTREETECH TFT35 E3 V3.0
SKR 1.4 Turno Marlin 2.0.7.3 - Firmware TFT BIGTREE_TFT35_V3.0_E3.27.x.bin

oldman4U, 2021-05-31:
Firmware source: https://github.com/MarlinFirmware/Marlin/releases

Minimum Marlin firmware version: 2.0.8.1

Distribution date: 2021-05-15

mytique38, 2021-06-02:
> Any news?

Hi oldman4U, 

For solve my issue, need to upgrade Marlin to 2.0.7.X to 2.0.8.0 ? it's mandatory ? 
no need to change the firmware of the screen? 

the Marlin upgrade is long and will take me several days thanks for your help

oldman4U, 2021-06-02:
Thank you for your answer.

First it would be important to answer your own ticket in case someone replies.

Please read the https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md and follow the guidelines there. If your problem is solved after that, please close this ticket. If it still exists, describe the important steps needed again, because I do not understand what "2. Prepare new filament" really means and if it is important to reproduce the reported issue.

Thank you

oldman4U, 2021-06-15:
See attached movie which shows that printing from SD card works fine after loading filament.

[Print after Load Filament.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6656792/Print.after.Load.Filament.mov.zip)

Please close this ticket as the reported issue is most likely not related to the TFT firmware.

Thank you

oldman4U, 2021-06-17:
@Msq001

Please check and close this ticket. 

Thank you

closed 2021-06-18 by bigtreetech