bigtreetech/BIGTREETECH-TouchScreenFirmware #1963: [FR] (feature request title)

MarkusThur, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
This request is related to #1952.

There definitely at least with the BTT 70" is a encoding issue with the language headers and inis in combination with the provided fonts.

language header files and .ini files are or should be encoded UTF-8

according to the readme.md in the fonts file contain character representations  encoded ASCII
or UTF-16. 

> The "byte_ascii.fon" is the bitmap fonts of ASCII, size: 12*24
> The "word_unicode.fon" is the bitmap fonts of UTF-16, size: 24*24
> Scan direction: form UP to DOWN, from LEFT to RIGHT

Sadly I can't check the fon files, as I don't know the proper editor for them.

For the first 127 characters of the Unicode this is fine, for the higher characters this in some way must and does fail.

Sure is correct UTF-8 encoded 2 characters representation of &#xB0;&#x43; does work, one character representation of &#x2103; does fail in different environments.

**Describe the solution you'd like**
- Providing a proper editor for the fonts.
- marking the needed encoding for language_xx.ini files, language_xx.h files, and fonts clearly (preferably UTF-8 as there the BE / LE problem does not exist) 
- checking the fonts, if they have a representation for U+2103 &#x2103; or idicating that that character can or cannot be used

**Describe alternatives you've considered**
- there is no alternate to checking if some characters needed with Temperatures are representet in the font files
- alternatively UTF-16 encoding could be used, if BE / LE encoding is announced and maintained throughout the project clearly
- implementing a autoconversion in the `buildroot/scripts/auto_gen_language_pack.py` to the correct encoding would be great but in my eyes not needed

**Additional context**
Thank you for reading and considering this

guruathwal, 2021-06-03:
@MarkusThur The character `℃` is being properly encoded in both language_xx.ini and language_xx.h file for all languages. 
The UTF-8 encoding is already implemented in `buildroot/scripts/auto_gen_language_pack.py`. 
I inspected all the files and found that all unicode characters are being encoded correctly (see image below). If there was an issue with the firmware then it will be with all the TFT variants because the API is the same for all the variants. the only difference is the screen resolution.
It is not clear what kind of problem you are having with just the `℃` character and why? Did you modify any part of the firmware? You need to share the file which has improper encoding and share a photo of the display with the issue.

[image]

MarkusThur, 2021-06-03:
Some files contained the UTF-16 / Windows 1252 encoding of it `21 03` at some point, somewhere at the Vx.x.27 tag. This happens as the files do not contain a encoding indicator and the editor "guesses" the encoding wrong.

Lets have a markdown file at the relevant positions to remind on the right encoding and everything is fine.
If the auto_gen_language_pack.py would automatically take care of it, would be really great.

closed 2021-08-14 by stale[bot]