bigtreetech/BIGTREETECH-TouchScreenFirmware #1974: [Q] Can we now officially remove the M109 / M190 redirections?

Tannoo, 2021-06-05:
I have seen that the M109 / M190 have been redirected to M104 / M140 respectively.

The original issue of not waiting for that process to finish seems to have been corrected.

I, personally, have just modified the redirection back to itself for now. Testing it seems to be running just fine and temperature LED transitions work just fine.

Can we revert this redirection? Or is there another fix angle being worked on still?

oldman4U, 2021-06-07:
As far as I know BTT would like to solve this problem using another approach. We will see soon.

Tannoo, 2021-06-10:
I believe is was fixed with PR #402 but the "re-direction" was not removed.

discip, 2021-07-27:
I'll second the official removal of this redirection, since there is no way to use *custom code* including **`M109 / M190`** at the moment.

regards

Tannoo, 2021-08-18:
I have noticed that the transitions work just fine as long as the printer doesn't sit idle for too long.
After sitting for some unknown time, another print can be started just fine, but the LED transition does not work.
I can, however, set the LEDs on whatever color by the TFT and they respond. They will remain that color for the entire print and beyond. I have to turn them off or change the color myself.

I have found that if:

   1. I start a print and the LEDs do not respond. I cancel the print while the bed is heating, then once it stops, the transitions and LED "automatic" controls work just fine when the print is restarted.
   2. If I reset the printer via the TFT E-Stop button first, then print is started, the transitions and LED "automatic" controls work just fine.
   3. After some time (Amount of time is random so far due to life, but greater than an hour.) Used TFT to set the LEDs to a color, then set them off. I start a print and the LEDs do not respond.
   4. After some time (same as #3), I set the LEDs to WHITE for printing. I then start a print and the transitions does not work and the print continues without a hitch. Once the print is finished, the LEDs remain WHITE and do not turn GREEN. I use the TFT to set the LEDs off.

So far, I was off on a trip out of town over the last week. I left the printer on. When I got back today, I started a print and the LEDs did not respond as predicted. I did the same as #1, and they were working just fine after that.

I have not found anything other than serial issues when searching for the reasoning of these "bypasses". There is still an issue and I am trying to describe them the best as I can as I try to get this figured out.

discip, 2021-09-07:
@digant73
## Thank you Sir! 😃👍

closed 2021-09-09 by bigtreetech