bigtreetech/BIGTREETECH-TouchScreenFirmware #1975: [Q] (Your Question?)

Smartich0ke, 2021-06-06:
**Your question or details you need. Please describe.**
When I update the display firmware, the touchscreen does not work. The screen part works fine, but tapping any of the buttons do not work. I have posted an unlisted video on YouTube [here](https://youtu.be/rxhTE7eFC8M) demonstrating the issue

**specify the device variant related to your question.**
The printer is an Artillery Sidewinder X1.

My screen is an MKS TFT28 clone that is original to the printer.

Also note that, when I put the original firmware on the screen, it works without issues.

**Additional context**
I hope I am doing this right, but just to confirm, the way I flash the screen firmware goes as follows (from within the 'Copy to SD card' directory):

- Download and extract the firmware repository.

- Drag the config.ini file onto the root directory of the SD card.

- Drag the MKS_TFT28_V4.0.27.x.bin file onto the SD card. (I have also tried v3.0.27.x but returns white screen).

- Drag the TFT28 folder from inside the Hybrid Red theme directory.

- Rename the .bin file to MKSTFT28.bin

- Rename the TFT28 folder to MKS

- Eject SD card, plug into printer and restart.

oldman4U, 2021-06-06:
Try the reset.txt and a different sd card like described in the trouble shooting section of the readme.

oldman4U, 2021-06-10:
Msq001

This user does not respond. Please close this ticket

Thank you