bigtreetech/BIGTREETECH-TouchScreenFirmware #1976: AUTO BED LEVELING

IronXRN, 2021-06-06:
Hi! 
The problem occurred when AUTO BED LEVELING on the new firmware BIG TREE_TXT 35_V3. 0_E 3.27. x. bin. The HOME offset and the BLTOUCH offset are set (the nozzle and BLTOUCH work exactly). When manually adjusting the table, the nozzle stands exactly at the corners and center, taking into account the offset. But when you enable automatic table alignment, BLTOUCH goes beyond the table and does not take into account the specified offsets, the process returns an error. Tell me, is this a firmware bug or have I not configured something?

My equipment: 
Ender 3 PRO
BIGTREETECH SKR mini E3 v1.2  
Marlin CONFIGURATION_H_VERSION 02000801
BIG TREE_TXT 35_V3. 0_E

 Thank you!

radek8, 2021-06-06:
You will have a misconfigured Marlin.
Points for Autolevel (M29) are defined in Malin not on TFT

IronXRN, 2021-06-06:
> You will have a misconfigured Marlin.
> Points for Autolevel (M29) are defined in Malin not on TFT

You probably meant the G 29... But the question is that the manual settings are also specified in Marlin, but the printer works them perfectly (taking into account the shifts and indents set to TTF) . The problem only occurs when AUTO-leveling is started.

radek8, 2021-06-06:
Sorry it was supposed to be G29 :-)

Can you send Marlin and TFT configuration files?

IronXRN, 2021-06-06:
> Sorry it was supposed to be G29 :-)
> 
> Can you send Marlin and TFT configuration files?

IronXRN, 2021-06-06:
I don't know how to unload an already configured configuration from TTF,.. I can also attach a TTF photo.

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6604977/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6604978/Configuration_adv.txt)

[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6604984/config.txt)

Tannoo, 2021-06-07:
I would start with Marlin.

In `configuration.h`, `#define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }` is the probe offset (X, Y, Z).
This works for all probes.
It looks as if these are still default values in Marlin. This only means that if you do an EEPROM factory reset, this is what it will 'default' to.

UBL does need anything more aside from this:
`#define MESH_INSET 20              // Set Mesh bounds as an inset region of the bed`

I am not sure how the TFT writes the offsets. But, if that is working, save to EEPROM and also change your configuration.h file to match for the next time you re-flash, it will already have been the latest numbers.

There is also this:
https://github.com/MarlinFirmware/Marlin/issues/20278
They did say it was a bug and to flash with the latest `bugfix-2.0.x` Marlin branch. (That was over 6 months ago).

Update your Marlin firmware to the latest `bugfix-2.0.x` and see if your issue persists.

IronXRN, 2021-06-08:
> I would start with Marlin.
> 
> In `configuration.h`, `#define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }` is the probe offset (X, Y, Z).
> This works for all probes.
> It looks as if these are still default values in Marlin. This only means that if you do an EEPROM factory reset, this is what it will 'default' to.
> 
> UBL doesn't need anything more aside from this (as far as the probe boundries are concerned):
> `#define MESH_INSET 20 // Set Mesh bounds as an inset region of the bed`
> 
> I am not sure how the TFT writes the offsets. But, if that is working (View the current offsets with M503), save to EEPROM and also change your configuration.h file to match for the next time you re-flash, it will already have been the latest numbers.
> 
> There is also this:
> [MarlinFirmware/Marlin#20278](https://github.com/MarlinFirmware/Marlin/issues/20278)
> They did say it was a bug and to flash with the latest `bugfix-2.0.x` Marlin branch. (That was over 6 months ago).
> 
> Update your Marlin firmware to the latest `bugfix-2.0.x` and see if your issue persists.
> Remember to reload factory defaults after flashing another version. (M502 then M500)

I updated the firmware to bugfix-2.0. x. but it didn't help. #define PIZZLE_TO_PROBE_OFFSET solves the problem of BLTOUCH displacement relative to the nozzle, but how can Marlin solve the problem of nozzle displacement?

Tannoo, 2021-06-08:
It should be `NOZZLE_TO_PROBE_OFFSET`

Or do you mean something about maybe an extrusion multiplier?

IronXRN, 2021-06-08:
In the photo above, there are two values. If I understand correctly, the Probe Offset changes-NOZZLE_TO_PROBE_OFFSET, and what does the Home offset setting change?

radek8, 2021-06-08:
As Tattoo wrote.
You have a badly defined NOZZLE_TO_PROBE_OFFSET
Home Offset adjusts the distance of the limit switches from the edge of the bed

Tannoo, 2021-06-08:
> As Tattoo wrote.

*Tannoo

Tannoo, 2021-06-08:
`NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }`
Or
`NOZZLE_TO_PROBE_OFFSET {  X,  Y,  Z}`
Has 3 values you need to set.

Yes... Home Offset does just what that says. Offset your nozzle from the actual 0,0 (Home) position.

Tannoo, 2021-06-08:
Attach a picture of your nozzle / BLTouch assembly. And I can try and help you get some more worable setting for you.

radek8, 2021-06-08:
If you have Ender3 without HW modifications, it is not necessary to change the Home Offset.

Tannoo, 2021-06-08:
So, that "should" mean the X offset would be 39 not -39.

radek8, 2021-06-08:
You are using UBL and with Probe Offset not all positions for the probe are available. Shouldn't you define an available area for the probe?

[image]

Tannoo, 2021-06-08:
Usually, you don't have to. I use UBL exclusively.

Just the inset (edge of mesh inset from edge of defined bed size).

IronXRN, 2021-06-08:
> Если у вас есть Ender3 без изменений HW, не нужно менять смещение Home.

Tannoo, 2021-06-08:
No picture?

Tannoo, 2021-06-08:
You really should start a new issue on Marlin's github:
https://github.com/MarlinFirmware/Marlin

IronXRN, 2021-06-08:
[image]
[image]

> No picture?
> 
> It cannot just "know" center without knowing the home limit positions, or at the VERY least, putting into a position manually and telling it that is starting "Home" point.

[image]

Tannoo, 2021-06-08:
Thank you for the pictures. So, the probe is to the right of the nozzle by ~40mm? That means you need X = 40mm.
Looks like the probe is ~10mm behind the nozzle? So Y = -10.
Now, when probe actives, how far is the nozzle above the bed? That would be Z = -??.

`NOZZLE_TO_PROBE_OFFSET { 40, -10, -?? }`

Tannoo, 2021-06-08:
Now, being that your probe is so far away from the nozzle, your bed may have quite a few points that do not get probed. (Depending on the mesh size you choose)

Tannoo, 2021-06-08:
Do you have end limits and at which ends?

Tannoo, 2021-06-08:
You can manually set the home positions in `configuration.h`.
```
MANUAL_X_HOME_POS 0
MANUAL_Y_HOME_POS 0
MANUAL_Z_HOME_POS 0
```
I might set the `MANUAL_Y_HOME_POS` to 10 or 20. See if that gets the home offset you want.

IronXRN, 2021-06-08:
> 
1. Installed Marlin 2.0.8.2 
2. Set the values of NOZZLE_TO_PROBE_OFFSET {40, -10, 0 }. I attach a photo of the position of the sensor and the nozzle at the command g28, and the second G0 X125 Y125 (center). As you can see, both the nozzle and the sensor are shifted.

IronXRN, 2021-06-08:
[image]
[image]
[image]
![Uploading IMG_20210608_163752[1].jpg…]()

Tannoo, 2021-06-08:
Did you initiate a factory reset of the EEPROM?

IronXRN, 2021-06-08:
> Did you initiate a factory reset of the EEPROM?

Yes

Tannoo, 2021-06-08:
Can you please attach the new config file?

radek8, 2021-06-08:
Tannoo, you wrote it the other way around.
if the probe is to the left of the nozzle, the X-axis must be negative
pokuid is the probe behind the nozzle muxi hitting the Y axis positive.

IronXRN, 2021-06-08:
> Can you please attach the new config file?

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6617202/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6617203/Configuration_adv.txt)

radek8, 2021-06-08:
IronXRN  don't describe the numbers written by Tanno, it's an estimate, you have to measure the exact value.

IronXRN, 2021-06-08:
> Tannoo, you wrote it the other way around.
> if the probe is to the left of the nozzle, the X-axis must be negative
> if the probe is behind the nozzle, the Y axis must be positive.

Ок.. I'll try to change the values now

IronXRN, 2021-06-08:
left 47, back 10 
put # define NOZZLE_TO_PROBE_OFFSET {-47, 10, 0 }

radek8, 2021-06-08:
Probe offset -47 does not allow you to probe the right edge. That would hit the extruder into the frame. you must set the probing limits.

[image]

IronXRN, 2021-06-08:
#define NOZZLE_TO_PROBE_OFFSET { -42, -10, 0 } - BLTOUCH в центре -ОК. 
G0 X115 Y115 смещение сопла от центра по Y= 28, и именно с этой проблемой не могу справиться. 
[image]

radek8, 2021-06-08:
When you enter
G0 X115 Y0
Where will the nozzle be?

IronXRN, 2021-06-08:
> When you enter
> G0 X115 Y0
> Where will the nozzle be?

![Uploading IMG_20210608_180218[1].jpg…]()

radek8, 2021-06-08:
Again and better :-)

radek8, 2021-06-08:
Yes, this is your problem

radek8, 2021-06-08:
you have 3 possible solutions

IronXRN, 2021-06-08:
> Probe offset -47 does not allow you to probe the right edge. That would hit the extruder into the frame. you must set the probing limits.
> 
> [image]

Your advice has significantly improved the Auto-level result. The process took place without leaving the table. (move it slightly in Y direction). I set the values : 
 #define MESH_MIN_X MESH_INSET
  #define MESH_MIN_Y MESH_INSET - 9
  #define MESH_MAX_X X_BED_SIZE - 42
  #define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)

#define NOZZLE_TO_PROBE_OFFSET { -42, -10, 0}

radek8, 2021-06-08:
If you move the bed by hand as far forward as possible. Where will the nozzle be?

IronXRN, 2021-06-08:
> If you move the bed by hand as far forward as possible. Where will the nozzle be?

In the last photo, Y0 .. already to the end

radek8, 2021-06-08:
This is the beginning. I want a second stop on the Y axis

radek8, 2021-06-08:
1. You move the end stop, by cc25mm
As I know the dimensions of Ender3 are tight, you can shorten the size of the bed by about 20 mm in the Y axis.
  2. Edit Marlin
  #define Y_MIN_POS -25
Same problem as in point 1.

3. move the entire rail along which the bed travels 25mm to the front.
This will solve the Y-axis problem and not reduce the bed.

Do not set the Home Offset!

IronXRN, 2021-06-08:
> This is the beginning. I want the maximum stop on the Y axis
> Pull the bed towards you as far as it will go
> Where is the nozzle to the bed

![Uploading IMG_20210608_183108[1].jpg…]()

radek8, 2021-06-08:
I've been waiting for this.
I recommend the 3rd solution above

radek8, 2021-06-08:
This is due to the different geometry of the extruder.

IronXRN, 2021-06-08:
> I've been waiting for this.
I'm sorry, I had to step back. 
Thank you for your advice, I will think about what to sacrifice the table or time. ;)

IronXRN, 2021-06-08:
**I will describe the result in detail, maybe it will also help someone, especially when replacing the standard extruder!** 

Thanks again to Tannoy and radek 8 for their help. 
Moved the mechanical Y-axis by 20 ml. Slightly adjusted the Home offset. Set the values of #define NOZZLE_TO_PROBE_OFFSET taking into account the Home offset. Now G28 is the sensor in the center, G0 X117 Y117 (235*235) is the nozzle in the center. Auto-leveling goes without problems on 10х10. I printed G26, I attach a photo (not quite the right printing material, and it's time to change the tape), the main thing is all the points within the table and almost in the center!

[image]

IronXRN, 2021-06-08:
The only question... I want to transfer the settings to Marlin, so that they do not get lost when resetting. Tell me what value regulates the Home offset in the Marlin firmware?

radek8, 2021-06-08:
Do not use Home Offset.
use a negative value in
#define X_MIN_POS
#define Y_MIN_POS

IronXRN, 2021-06-08:
> Do not use Home Offset.
> use a negative value in
> #define X_MIN_POS
> #define Y_MIN_POS

- Home Offset - why is this bad? 
- I have already tried this, but I have X=5, Y=-2 (returns an error)

radek8, 2021-06-08:
Why X = 5
A positive shift does not make sense.

radek8, 2021-06-08:
I also have an Ender3 with a small conversion and my own extruder. That's why I recommend you what I've tried myself.

IronXRN, 2021-06-08:
> I also have an Ender3 with a small conversion and my own extruder. That's why I recommend you what I've tried myself.

Once again, thank you very much!

Tannoo, 2021-06-08:
Thank you @radek8, I was getting tired... had to get some sleep.