bigtreetech/BIGTREETECH-TouchScreenFirmware #1979: [FR] M117 support for RRF

bttguy, 2021-06-08:
Just checked how's M117 working with RRF, and it needs a little optimization. It would be a nice feature for RRF users too.
Te current behavior after I enabled M117 at config.ini:
M117 sent from TFT terminal -> notification pop up on TFT and on DWC
M117 sent from DWC console -> notification pop up only on DWC
I would like to implement the feature, when M117 sent from DWC it would also pop up on the TFT as a notification. Maybe it could also help in macros where a massege should pop up/user action is required.

jaysuk, 2021-06-08:
does M118 behave the same way?
https://duet3d.dozuki.com/Wiki/Gcode#Section_M118_Send_Message_to_Specific_Target

bttguy, 2021-06-08:
@jaysuk Thats what my second thought was, after work I will check it again with M118. What P parameter should I use, 0/2?

jaysuk, 2021-06-08:
might as well try both and see what happens.
hopefully both will work, if not as long as one of them works we can at
least note that as a current restriction

guruathwal, 2021-06-09:
M117 is a one way message. It will only be displayed on the LCD or DWC So this will be useless if it did not pass through the TFT. With M118, the message should be relayed to all the connected hosts as an echo message if the message does not contain any of the keywords which are suppressed by the TFT. You should check to which ports/targets the M118 message is relayed without the P parameter.

digant73, 2021-06-10:
On Marlin, M117 seems forwarded to the TFT as a notification. E.g. the "M117 Time Left xxx" is received as "action:notification Time Left xxx". I Don't know if a specific setting has to be enabled in RRF to provide this kind of feature.

jaysuk, 2021-06-10:
As far as I understand it, M117 does it automatically if a TFT is
configured.
The next step would be to see what is sent.

bttguy, 2021-06-11:
> M117 is a one way message. It will only be displayed on the LCD or DWC So this will be useless if it did not pass through the TFT. With M118, the message should be relayed to all the connected hosts [...]

It seems like M118 cant be executed without a parameter.
[image]
I tested with all the possible values, but none ofe them was work as I wanted.

Wombat37, 2021-06-12:
I was hoping to use M118 P3 to send messages to trigger an external WiFi server. 

Is this implemented and, if so, how would it be used?

Sorry if this is off topic.

Wombat37, 2021-06-15:
This would be wonderful if it's possible. It would be great for providing status from custom macros. Let me know if I can help with the testing

Wombat37, 2021-06-17:
Can we also include M291 in these deliberations?

dameq1, 2021-07-09:
You can print notification on TFT by command: 

M118 P2 S"//action:notification Text_here"

Wombat37, 2021-07-10:
Yes, that does work!

M118 in print files, macros and console messages formatted this way and applied on the DWC or mainboard do appear on the TFT!

This doesn't help for M117 (not really a problem) or M291 (would be nice to have).

bttguy, 2021-08-17:
@pfn Do you have and idea/solution to let these msg-s displayed on the TFT? Did you tried it/would like to help solve this problem or figure out a workaround? :)

pfn, 2021-08-17:
I need to know what the wire protocol is for M117, it's not obvious to me at the moment, so I'm not going to look into it.

I'm looking into doing interactive M291 for the TFT next. I have a plan on how to approach and resolve it already. That means interactive macros will work (at least for OK/Cancel -- no plan to support jog buttons yet)

pfn, 2021-08-18:
M117 looks like it won't be supportable given the current configuration to use with the TFT.

We typically use `M575 P1 S3 B57600` to connect the TFT to RRF. `S3` is "raw mode with checksums" per https://duet3d.dozuki.com/Wiki/Gcode#Section_M575_Set_serial_comms_parameters.

RRF has the following code for handling M117:
```
void RepRap::SetMessage(const char *msg) noexcept
{
    message.copy(msg);
#if SUPPORT_12864_LCD
    ++messageSequence;
#endif
    StateUpdated();

    if (platform->IsAuxEnabled(0) && !platform->IsAuxRaw(0))
    {
        platform->SendPanelDueMessage(0, msg);
    }
    platform->Message(MessageType::LogInfo, msg);
}
```

Basically, the M117 payload is only sent if the port is not in "raw mode"

I don't know what the scope of getting the TFT to work in PanelDue mode would be.

bttguy, 2021-09-10:
@pfn Any changes that had been implemented recently could make this FR possible, or its still not possible without a major rework?

pfn, 2021-09-10:
This does not seem possible without switching to PanelDue mode; and that requires a large amount of rework

pfn, 2021-09-13:
Looks like the changes in #2171 will enable this to happen pretty easily

bttguy, 2021-09-13:
@pfn wonderful news to a monday!!

bttguy, 2021-09-17:
@pfn #2171 is merged, if you have time this feature would be appreciated

xPew, 2021-11-09:
When I tested M73 support it was a bit confusing wether it was working or not. Not sure if it is a TFT or RRF or user related bug, but anyway. SuperSlicer now has M117 support for remaining time, which might be useful, at least as an option. I would be also interested in this feature. @bigtreetech maybe you could check on this one too? :)

bttguy, 2021-11-29:
@pfn Do you have a sollution to this FR? Sadly nobody came up with an idea, and you mentioned it will be easy after #2171 is merged. :) RRF mode need you!

pfn, 2021-11-29:
It is pretty easy, but my understanding is that M575 S0/S1 isn't working well lately, and I don't have time to investigate it at the moment. The next time I open up my printer, I'll extend a breakout cable for rx/tx/gnd so that I can attach a serial monitor to see what's going on. TFT work is on the back burner for me currently.

closed 2022-02-05 by stale[bot]