bigtreetech/BIGTREETECH-TouchScreenFirmware #1981: [FR] Files in icon mode in RRF

bttguy, 2021-06-09:
With the new soon to be merged thumbnail support, I would like to ask some rework/rethinking of the file menu for in RRF.
Currently only list mode is supported, it can't be turned off/switched to icon mode. 
We discussed earlier this, but maybe this new thumbnail PR could help with speed. Even there would be some latency to load thumbnails, I and possibly many others would use thumbnails.

Wombat37, 2021-06-09:
Icon mode already works for me under RRF - but only for Cura generated print files on the TFT SD card.

It would be great if there was also support for files on the mainboard SD card (as well as the ability to order files). I understand that this is a RRF development issue not TFT.

guruathwal, 2021-06-09:
The reason for files in icon mode is not available for onboard SD is because there is no way to load thumbnails from the onboard SD and also even if it gets added in future, it will be dead slow. So the List mode was made permanent for OnboardSD to make it easy to read the file names.

bttguy, 2021-06-11:
> The reason for files in icon mode is not available for onboard SD is because there is no way to load thumbnails from the onboard SD and also even if it gets added in future, it will be dead slow. So the List mode was made permanent for OnboardSD to make it easy to read the file names.

@guruathwal how slow is dead slow? In case we would like to use an USB drive, it could be useful, can it be enabled to test the new thumbnail feature?

guruathwal, 2021-06-12:
@bttguy As slow as file transfer through serial.

bttguy, 2021-08-17:
Is there an option to enable the icon mode with blank icons, and let the user choose whether to use icon or list mode? I prefer the grid view, regardless the names are shorter. I would be thankful for an option even without the thumbnails at first :)

bttguy, 2021-08-23:
@pfn Could you check on this one wether is it easy enough to quickly add it to your recent PR bundle? :)

pfn, 2021-08-23:
> @pfn Could you check on this one wether is it easy enough to quickly add it to your recent PR bundle? :)
> (The grid view without the thumbnails only)

@bttguy I celebrate giving users choice. In this instance, though, I agree with @guruathwal. It's a low value improvement to, I wouldn't be giving this a very high priority. If it's trivial, I might look into it if I'm fiddling around that area of code.

bttguy, 2021-08-23:
@pfn Yeah, agree and understand both of you. It would be just a nice touch if it is easy, just for the customisation option, but I also happier with bigger/better improvements/features. The nice feature would be a workaroud to be able to display the thumbnails :D

PR #2150 (2021-09-07): RRF: enable A-Z, Z-A, oldest and newest sort in print menu
### Requirements

No new requirements

### Description

* Adds `stream_handler` to requestCommandInfo to support >5K payloads
* Adjusts scope of parser/handler in RRFParseACK
* Unlocks icon/grid mode print job listing, fixes #1981
* Resets macro menu cur_page to 0 upon entry (fixes state between print
  and macro menus)
* Code in place to support `M20 S3` once the kinks are worked out

### Benefits

Improves RRF support

### Related Issues

Addresses #1981 request for print menu icon mode

### Testing

Verified working on my MKSTFT28

 [... cut]
commit 89139653 (2021-09-07): RRF: enable A-Z, Z-A, oldest and newest sort in print menu (#2150)

* RRF: enable A-Z, Z-A, oldest and newest sort in print menu

* Adds `stream_handler` to requestCommandInfo to support >5K payloads
* Adds M20 S3 parsing support to RRFM20Parser
  * Unsolved: The parser is tested to work, however using M20 S3 from the
    streaming request variant causes the firmware to crash. Testing
    using the usual `cmd_rev_buf` mechanism works fine
* Adjusts scope of parser/handler in RRFParseACK
* Unlocks icon/grid mode print job listing, fixes #1981
* Resets macro menu cur_page to 0 upon entry (fixes  [... cut]
closed 2021-09-07 by bigtreetech