bigtreetech/BIGTREETECH-TouchScreenFirmware #1982: [BUG] Home Offset Y value does not update properly

sarvenn, 2021-06-09:
Hi,
In TFT mode, I enter to machine setting screen and update my home offset values as X: 5 , Y:-5. Then save the settings to eeprom, then I go to terminal in TFT mode and send M503 to read back the eeprom values.

I noticed two issues.
1)In home offset screen, I enter my values above. X value shows correctly but Y value always turns back to 0. But, in the backround it is OK. I did save to eeprom command and do a homing. I saw X:5 Y:-5 Z:0 on main screen.
2) While I was sure that my Y value is save correctly but showed as 0 in homeoffset screen, I tried to readback eeprom value sending m503 from tft terminal screen.
It responded as "M206 X5 Y". That's all. I'm sure that it was correct on the older terminal view layout. May be it was also correct after new terminal layout. But I'm sure that it was broken. I updated my TFT and Marlin bugfix2 firmwares last Sunday. My Marlin configs are also inline with the instructions for TFT.
I wish I could attach my screen photos as well. I will be able to take photos 12 hours later.

I think in the meantime our developers would find the bug in the code.

Printer: Ender 3 Pro
Board: SKR Mini E3 V2
Screen: TFT 35 E3 V3
Marlin FW: Bugfix 2 (last updated on 6th of June)
TFT FW: Last updated on 6th of June

radek8, 2021-06-09:
I couldn't simulate your problem.

You write that you have saved the new value at Home offset (M206) and the printer responds to it.
But the answer to question M503 is M206 X5 Y,
instead of the correct M206 X5 Y-5.
Try to send the M206 command to the terminal
should answer correctly M206 X5 Y-5 Z0
Then there is the problem with Marlin. It would be advisable to report this issue to the Marlin Forum.
https://github.com/MarlinFirmware/Marlin/issues

sarvenn, 2021-06-09:
I will take detailed pictures of the issue @radek8, afterwards my issue will be understood much better. After that, If we are sure that it is about Marlin side, I will raise a ticket there.
But as I up to remember, a similar issue has been occured in the past. I will try to find that PR for an example.

radek8, 2021-06-09:
ok

sarvenn, 2021-06-09:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1633
https://youtu.be/ZWL2C5YVrQY

I found it. It was you @radek8 :) My first problem is similar to this. I click on the value, I type -5 for Y value. Hit the tick. Turn to upper screen. X value is shown correctly but Y value turn back to 0. But When I come back and save the settings to eeprom and do a homing, I see X5 Y-5 Z0. So even it is shown as 0 for the Y value, it is saved correctly.

Second thing is that, when I send M503 or M206 it responds as M206 X5 Y. That's all. These are new issues.
Whenever I update my TFT35 fw, I also update my Marlin to latest bugfix2 fw. So I can not be sure if it is about TFT or Marlin side.

I'll add pictures tonight. May be I upload a video to youtube and share it as well.

radek8, 2021-06-09:
I checked the code on TFT and it looks fine.

The problem is Marlin's answer. It should look like this

[image]
[image]

sarvenn, 2021-06-09:
> I checked the code on TFT and it looks fine.
> 
> The problem is Marlin's answer to your printer. It should look like this
> 
> M503
> [image]
> [image]
> 
> [image]

Exactly. I may try to re-install both TFT and Marlin FWs to see if that solves the problem. Otherwise I will raise a ticket at Marlin repo.
Radek what is you Marlin FW version? Latest bugfix?

By the way I disable this because I use 0 value for my manual bed leveling and I set 0.2mm for MBL starting point. That's why I disable this. This can't be relevant to this issue, right?

 * MBL settings * Apply the "level_z_pos" configurable parameter value as the * starting Z height for each point during MBL process. * If not enabled, you can set the desired starting Z height * in Marlin fw (MANUAL_PROBE_START_Z in Configuration.h). */#define ENABLE_MBL_START_Z

radek8, 2021-06-09:
I'm using Marlin with a bug fix from 5/21/2021

https://github.com/MarlinFirmware/Marlin/tree/87a943756a36fe7f1e3422868dcfa35d5dd54518

sarvenn, 2021-06-09:
> I'm using Marlin with a bug fix from 5/21/2021
> 
> https://github.com/MarlinFirmware/Marlin/tree/87a943756a36fe7f1e3422868dcfa35d5dd54518

Would you please update your fw and try it again?

sarvenn, 2021-06-10:
[home offset issue.pdf](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6628508/home.offset.issue.pdf)

oldman4U, 2021-06-10:
What if you use the TFT's Terminal to set the offset instead of the TFT setting?

sarvenn, 2021-06-10:
> What if you use the TFT's Terminal to set the offset instead of the TFT setting?

I will try and respond.
I will sen m206 x5 y-5, then m500 and m503. Also I will check the home offset menu if it has changed or not.

radek8, 2021-06-10:
From your photos, it is clear that the problem is in the answer to the M503, which is non-standard.
Therefore, Y sets the value to 0. And the Z value is not displayed.
TFT analyzes the M503 command and sets the values to the TFT menus accordingly. If Y is no value, it sets zero.
The problem is in Marlin's response to command M503 in line M206.
Update to the latest Marlin and if the problem persists open the ticket to GitHub Marlin.
Thank you

sarvenn, 2021-06-10:
> From your photos, it is clear that the problem is in the answer to the M503, which is non-standard.
> Therefore, Y sets the value to 0. And the Z value is not displayed because M503 is missing in the response.
> TFT analyzes the M503 command and sets the values to the TFT menus accordingly. If Y is no value, it sets zero.
> The problem is in Marlin's response to command M503 in line M206.
> Update to the latest Marlin and if the problem persists open the ticket to GitHub Marlin.
> Thank you
> 
> [image]

That's the most reasonable explanation to the issue @radek8. I think you are right. The weird thing is that if I send M206 alone it responds correctly but not to M503.

I will close this ticket and raise at Marlin repo.
Thanks for your interest.

closed 2021-06-10 by sarvenn

sarvenn, 2021-06-10:
https://github.com/MarlinFirmware/Marlin/issues/22098

radek8, 2021-06-10:
I updated to the latest Marlin with bug fixes.
Same problem as yours

sarvenn, 2021-06-11:
Thank you @radek8 . It was an important info.