bigtreetech/BIGTREETECH-TouchScreenFirmware #1988: [Q] What is the best way to read a host variable into LCD and vice versa?

mink007, 2021-06-11:
1. 
Let us say I defined a global variable, A, in host side of the firmware that runs on LPC1769.
I want to read the value of that variable, A, in the LCD firmware.

2.
Let us say I defined a global variable, B, in LCD side of the firmware that runs on STM32.
I want to read the value of that variable, B, in the host firmware.

radek8, 2021-06-13:
Your question is too general. I don't know if anyone will answer you.

I'm not a programmer, but I'll tell you what I know
TFT and Marlin communicate with each other via RS232 via G-codes.
On the TFT side is the parseACK.c file. It analyzes the incoming communication and if it encounters a variable defined here, it stores it in a defined variable in the TFT.
Conversely, outgoing traffic is analyzed by the interfaceCmd.c file. If a defined variable occurs, it also saves it.

On the Marlin side, it will be done similarly. I don't know the names of the files, but a person who knows programming would certainly find them.

So if your variable is reported in the M503 command, or has its own G-cod that Marlin sends to the serial port, it should not be a problem to define it and save it in the TFT.

mink007, 2021-06-13:
Thank you for your response. I am aware of parseACK.c & interfaceCmd.c and was using M117 to send and receive information. In the terminal when i use this command, "M117", it works and does display the texts in the terminal which is response from the host (after i modified the M117 part of the code in the host side to add extra string signature).

When I use the M117 outside of the terminal then I am struggling to get the message from the host. Part of it is due to i am trying to understand the code itself. Therefore, was looking for if someone who knows the code better can offer their 2 cents.

I am trying to understand this piece of the code in interfaceCmd.c under "case 117" statement:
******************
strncpy(message, &infoCmd.queue[infoCmd.index_r].gcode[cmd_index + 4], CMD_MAX_CHAR);
******************
more specifically this part
************
&infoCmd.queue[infoCmd.index_r].gcode[cmd_index + 4]
************

I am also looking for a simpler form of communication between the touchscreen and the host. More along the lines of like
 Serial_Puts(SERIAL_PORT, "<string that I want to send>");
where I can use the value (string that I want to send) of the string in the host if I am using Serial_Puts in touchscreen.

mink007, 2021-06-20:
By using 
Serial_Puts(SERIAL_PORT, "M117 HELLO");

in the touchscreen side of the code, I am able to send the command to host and the host does respond with the HELLO back.
Need to capture this HELLO that is sent from host in the touchscreen side of the code.

mink007, 2021-06-20:
I am going to refer this response:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1050#issuecomment-686074269

and see if things work for me as I expect.

mink007, 2021-06-24:
> By using
> Serial_Puts(SERIAL_PORT, "M117 HELLO");
> 
> in the touchscreen side of the code, I am able to send the command to host and the host does respond with the HELLO back.
> Need to capture this HELLO that is sent from host in the touchscreen side of the code.

I was able to figure out that "toggleTool();" function is the one that updates. More specifically nextScreenUpdate(UPDATE_TOOL_TIME).

closed 2021-06-24 by mink007