bigtreetech/BIGTREETECH-TouchScreenFirmware #1995: TFT suddenly went black

Thijsvanzwam, 2021-06-13:
For some reason my TFT43 V3.0 just stopped working. No response or anything when touching the screen.
I reflashed the firmware that worked before and now it does nothing. Black screen, no light up button, no beeps. just a black screen and only a little red led on the backside that is on.
I have no way of reinstalling the firmware as it does nothing when inserting the SD card.
Anyone any idea what to do now?

### Steps to reproduce

1. turn on printer
2. wait...
3. nothing

**Expected behavior**
the screen to turn on

**Actual behavior**
just black, no way of reinstalling the firmware

### Hardware Variant
TFT43 V3.0
### TFT Firmware Version & Main Board Firmware details
latest

I tested the connection using the TFT35 V3.0 from another printer and that works fine.
[image]
[image]

kisslorand, 2021-06-13:
How is this firmware related? For hardware issues please contact your supplier or some HW related forums, channels.

Thijsvanzwam, 2021-06-13:
trying to figure out a way to reinstall the firmware...

kisslorand, 2021-06-13:
Ah! Put an empty "reset.txt" on your SD card's root folder. Literally this just happened to me 15 minutes ago (after flashing new FW -> black screen).

Thijsvanzwam, 2021-06-13:
no luck with the reset.txt... nothing happens.

Black screen happened after flashing firmware. At 99% it went black and the knob led turned off. Since then nothing works.

Thijsvanzwam, 2021-06-13:
maybe a strange question but could you send me your reset.txt file? I made my one on a Mac using text editor but I have had a situation in the past where that was the reason why something didn't work.

kisslorand, 2021-06-13:
Sure.
[reset.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6644776/reset.zip)

Thijsvanzwam, 2021-06-14:
thanks!
tried it but no reaction at all.
it's weird because all was working well. Flashing worked like always but this time it just went black at 99% and stopped completely. Don't think it is hardware related. Besides the reset.txt, is there anything else I could try to get the firmware flashed correctly?

kisslorand, 2021-06-14:
Perhaps another SD card.

Thijsvanzwam, 2021-06-14:
tried several. all reformatted.
Only thing I can think of is that formatting a SD on a Mac is different then on a windows but I never had any problems with that on BTT products before.

radek8, 2021-06-14:
Read the #1973

Thijsvanzwam, 2021-06-14:
ok, nailed it.
Tried the reset.txt file several times by pressing the reset button like 2000 times (on a very frustrated way), suddenly I got a message saying "outdated files"
Installing firmware went all normal after that. now all works. Don't know what triggered it but printer works again. Thanks all for your reply's!

closed 2021-06-14 by Thijsvanzwam