bigtreetech/BIGTREETECH-TouchScreenFirmware #1997: Resume after Pause not working under RRF

Wombat37, 2021-06-16:
**Description of Error**

After pressing the Pause button on the TFT touch screen, the active print is paused, the temperatures are held at their respective set points and the printhead is homed - all as expected.

However, when the Resume button is now pressed, I get the following message and a loud squawking noise from the buzzer. After a few seconds the message is cleared and nothing else happens. The same thing happens each time I press the Resume button.  

[image]

I opened DWC and clicked on Resume there and a M24 message was displayed and the printer immediately resumed printing.

In this instance, I was printing from the SD card on the mainboard (not the SD card on the TFT).

I repeated this exercise but this time using the TFT SD card to hold the print file. This time Resume worked as it should. 

This appears to be a disconnect between the TFT firmware and the RRF firmware and is related to which SD card is being used to hold the print file.

### Hardware Variant
Ender 3 Pro
BTT-E3-RRF v1.1 
BTT-TFT35-E3

### TFT Firmware Version & Main Board Firmware details
RepRap 3.2.2
BIGTREE_TFT35_V3.0_E3.27.x.BIN (downloaded today 06/15/2021)

digant73, 2021-06-16:
if you can monitor the communication with mainboard, verify if action::paused or action::pause is received by TFT when pressing on the pause button (to pause the print). In order to work, action:pause should be received (instead of action:paused). action:paused should be received in case of M600 or blocking pause (asking to purge more or continue to resume the print).

Wombat37, 2021-06-16:
I would be pleased to help with further testing. However, I don't fully understand what you are asking me to do.
I think you want to find out if the Pause process is still in progress or whether it has completed and is now ready to Resume. Could you give me some more specific details on how I can check this out? Pressing a button on the TFT and look to see if the TFT received it is not very clear.

Note that the TFT Pause button is replaced by the Resume button and DWC gives a Paused message and can be Resumed from there so these are some indicators that the system has been fully Paused.

Let me know what else you want me to do.

digant73, 2021-06-16:
With Marlin the pause and resume is handled with a dialog. It is asked to press on "Continue" button to resume the print. If this is not happening or it cannot be configured with RRF, the current logic must be differenciated in order to work also with RRF. A PR is required

Wombat37, 2021-06-16:
When the Pause button is pressed on my RRF system, no Continue button appears on the TFT touchscreen but rather the Pause button changes to a Resume button.

As I described in my original posting, the Resume button works OK when printing from the TFT SD card. No Continue button appears then, either and the Pause button changes to a Resume button, too.

Wombat37, 2021-06-16:
FYI, in the past, I've seen a Continue message on the TFT screen when I've put pause commands into the g-code print file. This is different - I'm pressing a Pause button on the run screen on the TFT display.

digant73, 2021-06-16:
ok. so a PR to fix the bug when printing from onborad is needed.
@bigtreetech, currently the pause gcode sent to RRF is the same sent to Marlin (so M25 P1). That gcode allows the use of a dialog to resume the print from onboard. Before providing a PR avoiding the problem on RRF (probably I should disable the check on pause already ongoing), it should be better to know (and eventually try to use) a gcode for RRF equivalent to M25 P1. If also with that gcode the bug is present I will push a PR removing the check on pause already ongoing in case of RRF is used.

jaysuk, 2021-06-16:
The object model should really be used to monitor the state of the printer.
That will report of its printing etc.
M25 is valid but the P1 sent afterwards is ignored.

digant73, 2021-06-16:
Is there an equivalent of M25 P1? P1 allows the interactive dialog

jaysuk, 2021-06-16:
Not that I know of
https://duet3d.dozuki.com/Wiki/Gcode#Section_M25_Pause_SD_print

kisslorand, 2021-06-16:
In Marlin M25 has no parameter. If ADVANCED_PAUSE is defined and M25 is requested than Marlin redirects it to M125 and passes P1 to it.

Wombat37, 2021-06-17:
> ok. so a PR to fix the bug when printing from onborad is needed.
> 
> @Wombat37 What was that pause gcode you used in the gcode file?
> 
> @bigtreetech, currently the pause gcode sent to RRF is the same sent to Marlin (so M25 P1). That gcode allows the use of a dialog to resume the print from onboard. Before providing a PR avoiding the problem on RRF (probably I should disable the check on pause already ongoing), it should be better to know (and eventually try to use) a gcode for RRF equivalent to M25 P1. If also with that gcode the bug is present I will push a PR removing the check on pause already ongoing in case of RRF is used.

As per your request, the code snippet below shows an example of the pause at layer code in a Cura-generated print file that gave the Continue dialog box when it paused. It looks like it's using M226 rather than M25 to initiate the pause. This worked on my RRF system.

G1 F900 X111.326 Y102.275 E0.01754
G0 F6000 X111.326 Y102.3
G1 F1500 E-6.5
;MESH:NONMESH
G0 F300 X111.326 Y102.3 Z0.4
G0 F6000 X107.569 Y102.564
G0 X106.501 Y102.835
;TIME_ELAPSED:44.957820
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 1
M83 ; switch to relative E values for any needed retraction
G1 F300 Z1.4 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S0 ; standby temperature
M226 ; Do the actual pause
M109 S245 ; resume temperature
G1 F300 Z1.4
G1 F9000 X106.501 Y102.835
G1 F300 Z0.4 ; move back down to resume height
G1 F1500 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E0.01754
;LAYER:1
M106 S42.5
;TYPE:WALL-INNER
;MESH:CableClipExtra.stl
G1 F1500 E6.5
G1 F900 X106.25 Y102.585 E0.01119
G1 X106.25 Y102.5 E0.00269
G1 X106.501 Y102.5 E0.00374

jaysuk, 2021-06-17:
M226 should be used within a gcode file to generate a pause but should not
be used as a general pause. See the duet docs
https://duet3d.dozuki.com/Wiki/Gcode#Section_M226_Synchronous_Pause
M25 is the correct gcode to use

Wombat37, 2021-06-17:
I'm not arguing :) 

I was asked to show what code was in the print file that worked.

jaysuk, 2021-06-17:
Yea, I know. Just wanted to make it clear to the BTT guys (it wasn't aimed
at you).

Wombat37, 2021-06-17:
OK - that makes sense.

digant73, 2021-06-18:
@jaysuk, @Wombat37 ok clear. I will fix the bug. At the moment simply avoid to pause a print

digant73, 2021-06-18:
@Wombat37 please try PR #2008 and let us know the result. It should fix your issue

Wombat37, 2021-06-18:
Will do, but it may be tomorrow (no time today).

Thanks for working on this.

closed 2021-06-19 by bigtreetech

Wombat37, 2021-06-20:
Have now checked - Resume now works well under RRF.

PR #2008 (2021-06-19): fixed bug #1997
The check on blocking pause (requesting user interaction to resume) is now provided only for Marlin (it is fully supported). That will avoid the impossibility to resume the print on all the other FWs for which the support is not provided (e.g. RRF) or not known.

fixes #1997.