bigtreetech/BIGTREETECH-TouchScreenFirmware #1999: TFT35 MZ V3.0 knob led

Dapox11, 2021-06-16:
Hello!

I want to make an encoder backlight on the board. (knob led) (Like TFT35 E3 V3.0)
Is it possible? Need to add missing LEDs and line driver (already done)
And modify the firmware. 
If you add the following lines to the pin_TFT35_V3_0.h file:

#define LED_COLOR_PIN PC7
#define WS2812_FAST_WRITE_HIGH() GPIOC->BSRRL = 1<<7
#define WS2812_FAST_WRITE_LOW()  GPIOC->BSRRH = 1<<7
#ifndef NEOPIXEL_PIXELS
  #define NEOPIXEL_PIXELS 4
#endif

then the display stops working.

What am I doing wrong? Entering the wrong pin?

bigtreetech, 2021-06-22:
You can update the firmware of TFT35 E3 V3.0 directly. TFT35 MZ V3.0 and TFT35 E3 V3.0 are the same in software.

oldman4U, 2021-06-23:
Any news @Dapox11?

Dapox11, 2021-06-23:
The firmware is installed (TFT35 E3 V3.0 ), tft display works fine, but knob backlight still doesn't work(

oldman4U, 2021-06-25:
I have a E3 version so I can not help. Sorry. 

What is the MZ version?

Dapox11, 2021-06-26:
It’s tft35 v3.0 for anycubic mega zero
https://www.amazon.com/BIGTREETECH-Graphic-Display-Dual-Mode-Printer/dp/B08L9FBPMS

oldman4U, 2021-06-27:
According to the pictures I found at the link, this TFT does not have LEDs like the E3 version. [image]

This is missing and I do not believe that they are somewhere else.

[image]

Maybe it would be best to contact a printer related user group.

Happy printing

Please help the community and close the ticket once you do not need it anymore. Thank you

Dapox11, 2021-06-28:
In the first post I indicated that I installed the missing items (LEDs and line driver). Maybe something else is missing.

oldman4U, 2021-06-28:
@radek8

Do you know if it is possible to add leds like this? As far as I know this code only works with BTT TFT43 and bigger.

radek8, 2021-06-28:
Can you take a detailed photo of the  TFT35 V3.0 MZ display  board from both sides?

oldman4U, 2021-06-28:
The link above provides a good overview of the hardware. As far as I know even the E3 variant does not provide external neopixel support.

radek8, 2021-06-28:
This is not the support of an external neopixel, but the illumination of the knob, which in the TFT35 E3 V3 variant is made by a neopixel

oldman4U, 2021-06-29:
This TFT has no on board led and the user installed external hardware for this and want to control it via pc7.

Dapox11, 2021-06-29:
Detailed photo of the TFT35 V3.0 MZ display board from both sides i will do it later, the printer is in operation now.

The following components are installed:
U7: 74LVC1G125 SOT23-5
LED4 LED6 LED8 LED10: WS2812-2020

[image]
[image]

Also installed short the red sides of switch S2:

[image]

Dapox11, 2021-06-29:
Owners of TFT35-E3 V3.0, please take a detailed close-up photo of the marked area? Thanks!

[image]

oldman4U, 2021-06-29:
[image]

[image]

Dapox11, 2021-06-30:
At the moment, the behavior is that one LED flashes green when turned on:

https://user-images.githubusercontent.com/86003215/123900268-95fc8280-d98a-11eb-991d-6dff93a06c3e.MOV

[image]

Dapox11, 2021-07-05:
Thank you all, I figured it out! The main error was in the wrong setting of the S2 jumper. Special thanks for the photo of the board. Problem solved, everything works!

[image]
[image]
[image]

closed 2021-07-05 by Dapox11