bigtreetech/BIGTREETECH-TouchScreenFirmware #2000: [BUG] (Part of the information of status screen stays, when entering U Disk to start a print. Screen capture not working!)

discip, 2021-06-16:
### Description

### Steps to reproduce

1. On main screen enter **`Print`** than **`U Disk`**.
2. Watch the information previously scrolling in the status section partially overlaying the navigation buttons.

**Expected behavior**
Screen gets completely redrawn every time a submenu is entered, except for e.g.: back button.

**Actual behavior**
Unwanted parts of the previous screen stay, if the **`U Disk`** menu is entered.

### Hardware Variant
**MKS_TFT28_V4_0**

### TFT Firmware Version & Main Board Firmware details
both compiled today / Marlin (is the bugfix branch)

### Additional Information
To document the behavior, explained above, I tried to use the screen capture option, by defining it in **`Configuration.h`**.
Unfortunately this did not work!
```
Scrennshot/U:_9.bmp failed!
```
Interestingly a folder named **`ScreenShot`** on the SD Card is created, but without any content.
On the U Disk there is nothing created.

thank you in advance

oldman4U, 2021-06-16:
I can not confirm the issue you report using BTT TFT35 E3 v3 and Marlin 2.0.9.

Screenshot is for TFT24 V1.1 (ILI9341) and TFT43/TFT50/TFT70 V3.0 (SSD1963) only.

oldman4U, 2021-06-17:
Before you see the content of the UDisk, you have to select which store you would like to use. Can you see the overlay there also? Has this been working fine with a previous version and if so, which version broke it?

Take a picture with your camera or even better a movie and upload it here.

guruathwal, 2021-06-17:
Please provide an image to inspect.

oldman4U, 2021-06-17:
Or a movie. 😁

discip, 2021-06-17:
Good afternoon,
here are some pictures:

[image]
[image]
[image]

Unfortunately I do not know since when this started.
thanks in advance

kind regards

PR #2003 (2021-06-18): Fix text overlay on icons in File menu Icon mode
- Fix text overlay on icons in file menu with Icon mode. (#2000)

Fixes: #2000
closed 2021-06-18 by bigtreetech