bigtreetech/BIGTREETECH-TouchScreenFirmware #2001: Purge more resets board (BIGTREE_TFT35_V3.0_E3.27.x.bin)

JosemiZubi, 2021-06-17:
When loading in new filament and a request is made to purge more the board resets.

This problem happens since I have updated to firmware BIGTREE_TFT35_V3.0_E3.27.x.bin.

Is it possible that this new firmware is the cause of this behavior?

Thanks in advance.

kisslorand, 2021-06-17:
Update your Marlin.

oldman4U, 2021-06-17:
If you have a STM32 based mainboard, this is related to the old Marlin firmware you use. 2.0.8.1 is the minimum Marlin version mentioned in config.ini and ReadMe and it would be important to read the information provided before opening a ticket. Like mentioned in the Read me before posting a New Issue! also on the front page of this repository.

Please close this ticket

JosemiZubi, 2021-06-17:
O.K.
Thank you for your answers. Checking yesterday on the web I drew that conclusion and now it is confirmed.
Thanks a lot!!!

closed 2021-06-17 by JosemiZubi