bigtreetech/BIGTREETECH-TouchScreenFirmware #2004: [BUG] (Filament Sensor don't work in touch mode, if it is connected to the base board)

Ari-SSO, 2021-06-17:
### Description

The runout filament sensor do not function correctly in touch mode, my environment:

- **BTT TFT35 v3.0**
- **BTT SKR 1.3**
- **Filamente sensor connected to SKR 1.3 board, PIN 1.29** 
-   **Firmware TFT35 v3.0: BIGTREE_TFT35_V3.0.27.x.bin**
- **Firmware SKR 1.3: Marlin Bugfix (Version 2.0.9, according to the configuration.h file)**

  I have tried various combinations of the Marlin configuration, and the M119 command always behaves correctly in the absence or presence of filament when i use a host software connected by USB cable to my PC.
  Testing from Repetier Host, it also behaves correctly as the configuration changes with the M412 command.
  I have tried all the possible combinations of the M412 command, along with the configuration of the TFT35 v3.0 screen, plus the different combinations of Marlin. And it has always worked fine from Repetier Host.
  I use the BTT TFT35 v3.0 in _Marlin Mode_ then it also runs perfectrly and detect correctly the filament, but If i use it in _Touch Mode_, I have always obtained the following, regardless of the marlin configuration, the M412 command, and the existence of filament in the sensor:

  Filament sensor **ON**: With or without filament, it always says that filament is missing and cannot continue printing.
  Filament Sensor **OFF**: It never detects whether or not it has filament (I guess this is correct).
  Filament Sensor **SMART**: It starts printing and after a few seconds, it says that there is no filament. When pressing "Resume" it continues but says again that there is no filament one or two minutes later (This regardless of whether or not there is a filament). And the same always repeats for ever.

The problems appears only when i use the tocuh mode in the display (The mode that normaly I use).

### Steps to reproduce

1. Configure filament sensor in marlin firmware
2. run M119 to verify thar the sensor detect correctly the ,presence or absence, of filament.
3. Set the TFT to touch mode
4. Set, in TFT, the runout filament sensor ON
5. Print something with and without filament in the the printer (Always display a message reports that there is not filamente)
6. Repeat 4 and 5, but with OFF and SMART option.

**Expected behavior**
If there is filamente then print normaly, else stop printing,  (This if you set the filamente sensor on, if off contiune printing without filament)

In touch mode, depending on runout filament sensor setting in TFT:
  Filament sensor **ON**: With or without filament, it always says that filament is missing and cannot continue printing.
    Filament Sensor **OFF**: It never detects whether or not it has filament (I guess this is correct).
    Filament Sensor **SMART**: It starts printing and after a few seconds, it says that there is no filament. When pressing "Resume" it continues but says again that there is no filament (This regardless of whether or not there is a filament). And the same thing always repeats.

### Hardware Variant
 **Firmware TFT35 v3.0: BIGTREE_TFT35_V3.0.27.x.bin**
### TFT Firmware Version & Main Board Firmware details
 **BTT TFT35 v3.0**
    **BTT SKR 1.3**
    **Filamente sensor connected to SKR 1.3 board, PIN 1.29**
    **Firmware TFT35 v3.0: BIGTREE_TFT35_V3.0.27.x.bin**
    **Firmware SKR 1.3: Marlin Bugfix (Version 2.0.9, according to the configuration.h file)**
[BIGTREETECH-TouchScreenFirmware-Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6672609/BIGTREETECH-TouchScreenFirmware-Configurations.zip)
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6672611/Marlin.zip)

### Additional Information

* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

kisslorand, 2021-06-17:
Put M75 in the start of your gCode file.
If I had a cent for each time I wrote this sentence...
Wouldn't it be easier to search around and check it it has been discussed or not?
The TFT has no way to know if filament sensor connected to the motherboard is triggered or not unless someone (in our case Marlin on the motherboard) tells him so. Marlin is not transmitting filament sensor info unless it is printing. So Marlin doesn't know that the TFT is actually printing, not just sending gCodes, unless someone (in our case the TFT) tells him so. 
M75 tells Marlin that there's a print ongoing and Marling knowing this will send info about the filament sensor connected to it.

I hope it helps...

guruathwal, 2021-06-18:
The filament sensor option in the TFT is only for connecting the sensor to the TFT itself. In your case, the sensor is connected to the mainboard so the filament sensor settings on the TFT are useless. you will have to control it through marlin mode or through gcodes.

oldman4U, 2021-06-18:
We should add a note to the filament sensor part of config.ini that the Settings here have only to be activated in case the sensor is connected to the TFT and that you have to add M75 to the start gCode in case it is connected to the mainboard. Who wants to add this?

Ari-SSO, 2021-06-18:
if you add M75 at start code, may be also add M77 at end code, for consistency.
If you are OK, then i will add it in the config.ini...

oldman4U, 2021-06-19:
Please do so. But please add a note and do not change the gCode itself.

Ari-SSO, 2021-06-19:
I which branch i must add this? in develop branch?

Thanks!

oldman4U, 2021-06-19:
You just make a PR. BTT merges it correctly.

Ari-SSO, 2021-06-19:
it is done, i hope it is ok.

Thank to all of you!

bigtreetech, 2021-06-22:
fixed in #2017, so close here

closed 2021-06-22 by bigtreetech

oldman4U, 2021-06-24:
Thank you. I will change the order of the text and the text itself a bit and hope this is ok for you. 

Happy printing 

oldman

Ari-SSO, 2021-06-24:
Yes, please!... I Know that my english is not the best of the world.

Thanks!

oldman4U, 2021-06-25:
No problem at all. It was more the order of the information and to make it shorter. Check my PR.

Happy printing