bigtreetech/BIGTREETECH-TouchScreenFirmware #2007: [BUG] Filament Sensor Problems

omexlu, 2021-06-18:
Hello,

I have a firmware for the TFT based on your development:
https://github.com/Dtcreation/Firmware-Molise-TFT

However, I have the problem with my "Artillery Genius" that the LED from the filament sensor is always green whether now filament is inside or not.

Is there a bug here?

Currently, the filament sensor does not work so.

PS. In the firmware of the TFT, the filament sensor is of course activated.

Thanks in advance.

kisslorand, 2021-06-18:
Where is your sensor connected to? Did it ever work? If yes, what has changed since?

omexlu, 2021-06-18:
Hello,
It is connected to the standard port (i.e. display), with the original firmware it worked, if there was no filament it was red otherwise green.

Since then, only Marlin and the TFT firmware has been updated (the one mentioned above).

oldman4U, 2021-06-18:
Why you use a 2 months old firmware instead of the one from here and why not opening a ticket there?

omexlu, 2021-06-18:
@Dtcreation 

Because the "developer" pulls his data from here and thus is based on here, I am looking for help here. He only compiles this and adapts it to his firmware.

oldman4U, 2021-06-18:
But he is 2 months behind.

So even there would be a problem in the firmware here and we fix it - you would not get it.

oldman4U, 2021-06-18:
So technically you ask for help for another software nobody uses here.

omexlu, 2021-06-18:
It is also based on your software so to speak a fork only with adaptations, so it is already the software that is used here :)

I assume that he will also soon update to the latest version, have him times pointed out.

Even if I want to use the software directly from here I do not know which and which is used for the Artillery Genius (stock screen).

oldman4U, 2021-06-18:
We don't know which adaptions he made. Maybe he made a mistake.

omexlu, 2021-06-18:
I don't know either, I'm a total newbie in this field:
https://github.com/Dtcreation/Firmware-Molise-TFT/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT%20STOCK

Let's see if he does something :)

kisslorand, 2021-06-18:
With this FW in this repository no one reported such issue.

radek8, 2021-06-18:
Use FW from this repository.
Look at the display circuit board, there could be a description of the MKS display version. If not specified, use the files on the MKS one by one, one will work. :-)

omexlu, 2021-06-18:
I think: MKS TFT28
An Mainboard is MKS GEN_L v1.0

radek8, 2021-06-18:
MKS TFT28 V3.0 or MKS TFT28 V4.0?

omexlu, 2021-06-18:
Don't know can't open at this point and find no Info on the web :/

radek8, 2021-06-18:
The @digant73  may advise which version of TFT Artillery  uses. He also has Artillery :-)

kisslorand, 2021-06-18:
It's tricky with Artillery. I have a ASWX1 and the display says it is v3.0.0 but I have to compile v4 for it., v3 doesn't work.

digant73, 2021-06-18:
it's a clone of MKSTFT28 v4.0.

omexlu, 2021-06-18:
OK so it is MKSTFT28 v4.0 😊
But now I so smart as before 😂

Do you guys with Artillery have also my behavior with the filament sensor, that very strange 😬

digant73, 2021-06-18:
with a stock ASX1 use the following settings for the filament sensor:
fil_runout:1
fil_runout_inverting:0
fil_noise_threshold:200 (safe value. That value depends a lot on the printer. On my printer it's OK to use even 50.

omexlu, 2021-06-18:
The problem is that the sensor does not run at all, as soon as I turn on the Genius the LED briefly turns red and then immediately jumps to green, whether now filament is in it or not does not matter :/

With the stock firmware is not, so the sensor does not seem to be broken.

Find that very strange.

digant73, 2021-06-18:
try my settings on the official BTT fw. If the problem is still there, try to invert the filament connector on the TFT (the 3 pins connector on B1)

omexlu, 2021-06-18:
The settings are already present except for the last (not relevant) one:
https://github.com/Dtcreation/Firmware-Molise-TFT/blob/master/Genius/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT%20STOCK/config.ini

What do you mean by invert over exactly? Where should I plug it in? I can not just turn the wrong way in then the power supply etc is not correct

digant73, 2021-06-18:
try to set fil_runout_inverting:1
Eventually provide a picture of the TFT board with the connector.

omexlu, 2021-06-18:
I don't know what the option has to do with how the filament sensor works?
# Invert the logic of the sensor

This only inverts the logic so it will not work again 😊

The pin is exactly where it is on every artillery, nothing has been changed to it, the problem only exists since the firmware updates, with stock it works without problems

digant73, 2021-06-18:
are you using the official BTT fw? if not, switch to that and use the attached config.ini file just for testing 

[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6677314/config.zip)

omexlu, 2021-06-18:
Die Werte, die in der veralteten INI vorhanden sind, sind genau die gleichen für meinen Drucker angepasst:
https://github.com/Dtcreation/Firmware-Molise-TFT/blob/master/Genius/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT%20STOCK/config.ini

Ich kann nicht einfach diese Config nehmen, weil die weder angepasst noch für meine aktuelle Firmware ist.

Das macht die Probleme doch nur schlimmer.

omexlu, 2021-06-18:
But maybe you have a link to all the needed files for the Artillery Genius then I'll try to adjust that and update to the latest version, maybe that will fix the problem but I doubt it :/

radek8, 2021-06-18:
The current FW is free to download here

omexlu, 2021-06-18:
So i need the following:
- MKS_TFT28_V4.0.27.x.bin
- config.ini (adapt to my printer)
- language_en.ini
- and where i find the MKS Folder with this files? (https://github.com/Dtcreation/Firmware-Molise-TFT/tree/master/Genius/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT%20STOCK/MKS)

digant73, 2021-06-18:
@omexlu unzip the following zip file on the root of the SD card and install the FW as usual.
Both, the bin, icons and config.ini file are installed.
After installation, verify if it works

[MKSTFT28_v4.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6677486/MKSTFT28_v4.zip)

omexlu, 2021-06-18:
@digant73 Thanks i give a try later, is something special do define in config.ini except the size of the printer?

digant73, 2021-06-18:
only the build volume "size_max" (it is currently set for the ASX1 size_max:X300 Y300 Z400)

omexlu, 2021-06-18:
There is no language file on it?

digant73, 2021-06-18:
default language is english. So no need to load again the english language

omexlu, 2021-06-18:
I just flashed the firmware of the TFT successfully but unfortunately the problem persists that the filament sensor always shows green no matter if there is filament in it or not.

Here I think there could be an error/bug somewhere.

kisslorand, 2021-06-18:
Let's do a test.
Remove the middle pin from the runout sensor and check if the LED reacts to filament presence. In not, the sensor is broken.
If yes... I don't know ...
I had hunch that it must be some pin configuration problem.
I just checked the pinout definition back from 2020, it is not changed, it's still on pin PB1 and it is still defined as an input pin, either pull-up or pull-down, depending on the parameter "logic invert". Other than the input pin state, nothing else can make a working sensor not to sense the filament.

omexlu, 2021-06-19:
Which connector is it? The problem is that it was working on stock firmware :/

kisslorand, 2021-06-19:
Sorry, I was talking about the connector on the filament sensor.
So, remove the pin that faces forward (faces towards the front of the printer) from the filament sensor's cable connector.

omexlu, 2021-06-19:
I don't know if this will help, it worked under the stock firmware and gets power, but it doesn't work properly. LED always remains green whether filament is inside or not.

There are only 2 possibilities end,  a bug in connection with the tft firmware with this sensor or he is completely broken 🙈

I think but rather a bug because it works in the stock firmware without problems.

kisslorand, 2021-06-19:
So, if you write back original firmware, it works again?
Can you measure the voltage on that pin? If yes, does it change between 3.3V and 0V if filament is inserted and removed?

omexlu, 2021-06-19:
Yes under the original firmware it works, no filament in = red, filament in = green.

That's why I find that so strange, I get from support a new sensor but that may take some time until this is here (from China).

Unfortunately, I have no meter to measure that 😬 but Current seems to be on it.

kisslorand, 2021-06-19:
@digant73 
Do you have your filament sensor connected to your TFT?

omexlu, 2021-06-19:
For me I connected it to the TFT (standard port as shipped), so I think it's a bug in the firmware.

Whether the official current version from here or the one from Dtcreation both do not work with the sensor. The original however already.

I have also heard that in new deliveries a different sensor is installed, the new light green, the old blue when filament is inside.

kisslorand, 2021-06-19:
We all get it, you think it is a FW bug, no need to repeat it in every post of yours.

omexlu, 2021-06-19:
Sorry that I repeat myself I just want to contribute to the solution of the problem, I can currently work without sensor, but it would be nicer with of course :)

oldman4U, 2021-06-19:
The color of the led is irrelevant. The smart sensor works for me and I will test a standard sensor later today.

omexlu, 2021-06-19:
Maybe different Modells that makes issues, don't know. I tried with on and smart on tft settings.

oldman4U, 2021-06-19:
I have BTT mainboard, TFT and smart sensor in Ender3.

kisslorand, 2021-06-19:
I would love to know if the TFT is who's is blocking the sensor's signal wire. Without measurements I do not know how else it is possible to find out. I checked inside out the filament runout signal pin's configuration, couldn't find anything wrong with it.

oldman4U, 2021-06-19:
This will finally end with a "I found the problem" - and it was not the firmware.

omexlu, 2021-06-19:
Let's wait and see what Oldman's tests reveal about him.

Can hardly imagine that my sensor is already broken :/

oldman4U, 2021-06-19:
It is not the firmware. I tested the firmware from today with a standard sensor and it works like expected and the blue led on my sensor changes on/off whenever filament is present or not.

@guruathwal Before your PR has been merged I tested your PR with the BTT smart sensor and now with a standard sensor. In both cases the new stuff works great.

With the new firmware it is possible to define standard sensor or smart sensor in config.ini and also if the sensor is default on or off. In the TFT settings you can then turn the sensor on/off at any time. I'm loving it - or so.

@omexlu Please buy a new sensor and close this ticket, thank you.

omexlu, 2021-06-19:
However, it is strange that it works without problems under the stock firmware.

The lever of sensor is also not blocked when I inserted/executed filament I clearly hear the click but at the LED does not change.

The manufacturer will send me a new sensor (may take some time) and will then test again and then report whether it works or not.

I thank you in any case now for your help.

omexlu, 2021-06-20:
In a Facebook group 2 others already have the same problem, the filament sensor stays green.... Our sensors are definitely not broken.

oldman4U, 2021-06-20:
The BTT TFT firmware definitely works with both kinds of sensors on my printer using BTT mainboard and BTT TFT.

omexlu, 2021-06-20:
Maybe, but I just want to point out that already 3 people with stock mainboard/filament sensor (Artillery Genius / maybe actual batch) have the problem with the sensor and it just does not work there, that can not be a coincidence :)

Artillery has no BTT TFT and BTT Mainboard, maybe there is the problem with contabily to.

oldman4U, 2021-06-20:
This repository is about the firmware which is made for the BTT TFT displays.

omexlu, 2021-06-20:
Maybe but usualy supports it, because many Artillery machines use this firmware (thingiverse projects and also other maintenaner). 

Also mentioned in the Readme that MKS is supported even if not officially. 

MKS TFT Warning: BTT does not officially provide MKS TFT hardware support, MKS TFT is maintained by open source contributors, and BTT does not bear any risk of MKS TFT hardware using this firmware

MKS_TFT28_V3.0 and V4.0

MKS_TFT32_V1.3 and V1.4

cedtr0nik, 2021-06-20:
Hello, i have the same issue. Before updating : stock firmware motherboard and screen, filament green led, no filament red led. After updating motherboard and screen, filament green led, no filament green led...

kisslorand, 2021-06-20:
Please, any of you having problems, do the measurement I proposed.
I have an Artillery, but I have the old sensor with the blue led. It works just fine.

[28 more replies]

PR #2018 (2021-06-22): Filament runout sensor fix for new Artillery Genius printers
### Requirements

Artillery Genius - new version with dual led on the filament runout sensor

### Description

This PR fixes the non working filament runout sensors on newer Artillery Genius printers if the sensor is connected to the TFT as it is from factory.

Also, a small, non apparent bug is fixed, the MKS TFT has already pull-up resistors on the PCB connected to PB0 and PB1 pins, they are defined now as floating (no internal pull-up/down) if used as input pins.

### Benefits

Fixes #2007

### Related Issues

On newer Artillery G [... cut]
closed 2021-06-22 by bigtreetech