bigtreetech/BIGTREETECH-TouchScreenFirmware #2010: [Q] (Your Question?)

LTSpenny, 2021-06-19:
Using a TFT 35 E3 V3.0 with latest build of TFT35_V3.0)E3.27.x.bin if I erase configurations for preheat that will be unused, preheat screen becomes unreadable.

**specify the device variant related to your question.**I am using TFT 35 E3 V3 connected to an SKR E3 Turbo on an Ender Max
[image]

**Additional context**
Add any other context or screenshots about the question here.

would like to know if this is a bug or if I am removing these presets incorrectly, should they just be commented in the config file?

thank you for your time.

radek8, 2021-06-19:
Don't delete something you don't know how it works.
Reload the correct config.ini file.

LTSpenny, 2021-06-19:
@radek8 ok I've done that, could you point me in the direction where I can learn how this works? Thanks again!

radek8, 2021-06-19:
I don't know why you are bothered by some items in the preheat menu.
If you are bothered by the item NYLON:
In the Config.ini file, delete the name NYLON (overwrite to none) and set the zero temperature.
but do not delete the line
for example
preheat_name6: NONE
preheat_temp6: T0 B0

LTSpenny, 2021-06-19:
@radek8 lol ill leave it alone. Im sorry im just fascinated by all of this and think its really cool to be able to customize items so much that im just trying to learn. Thank you for taking the time to share with me!

closed 2021-06-19 by LTSpenny