bigtreetech/BIGTREETECH-TouchScreenFirmware #2012: [BUG] Wrong Colors when changing Color for Knob

LTSpenny, 2021-06-19:
Description

Steps to reproduce

1.- Download latest build of TFT 35 E3 V3 = E3.27.x.bin
2.- Navigate to menu to change color of wheel.

Expected behavior
Expect Orange to be Orange, Yellow to be Yellow, White to be White

Actual behavior
Orange = Light Green
Yellow = Light Green
White = Very Light Blue

Hardware Variant

BIGTREE TFT35 V.3.0 and SKR E3 Turbo
TFT Firmware Version & Main Board Firmware details

BIGTREE_TFT35_V.3.0_E3.27.X.bin
Latest Marlin 2.0.x
[config.ini.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6681109/config.ini.zip)
[image]

Thank you for the help.

oldman4U, 2021-06-19:
And which TFT firmware version works for you?

radek8, 2021-06-19:
Yes, it is true that the yellow and orange backlight of the knob is not completely ideal and may differ slightly on different HW, Fortunately, the display function and print quality have no effect :-)

Can you try this?

[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6681291/BIGTREE_TFT35_V3.0_E3.27.x.zip)

LTSpenny, 2021-06-19:
@radek8 Hi thank you, the orange looks yellow and white still look green and blue, I don't mind it as I prefer red, I was just trying to contribute something I found and if I could figure out where to adjust it possibly share it, appreciate the help!

@oldman4U Hello, I just got the TFT 35, so I haven't had any experience with prior versions or if those had this same inconsistency, not sure if that answers your question.

radek8, 2021-06-19:
My edit

[image]

LTSpenny, 2021-06-19:
Thank you @radek8 ill tweak from here.

closed 2021-06-19 by LTSpenny