bigtreetech/BIGTREETECH-TouchScreenFirmware #2015: [BUG] From Print Menu Bed/T0 Heat => LAST Heat settings from status screen

northwestfrog, 2021-06-20:
### Description
While Printing, when selecting the Bed Heat or T0 menu button, instead of showing the proper Heat Settings, It shows the LAST  heat settings that was selected from the main menu.

### Steps to reproduce
1. On the status screen, select the Bed Heat menu button
2. Return to the status screen (return)
3. Start a print 
4. Back to the Print status menu, Select T0 Heat menu button

**Expected behavior**
The T0 Heat settings are showing and are being available for modification

**Actual behavior**
The Bed Heat settings are showing and are being available for modification

### Hardware Variant
TFT35-E3-V3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x.bin
RepRapFirmware for STM32F4 based Boards FIRMWARE_VERSION: 3.3_2 ELECTRONICS: STM32F4 FIRMWARE_DATE: 2021-06-19 17:14:27
### Additional Information
It's as if the method "heatSetCurrentIndex" was not called before going to menuHeat when on the print status menu.
To verify this, after having done the above, once the print is completed, I went back to the status screen, selected the T0 heat menu button... Started a new print, and when on the print status menu, selected the Bed Heat menu button and the T0 Heat settings showed up.

oldman4U, 2021-06-20:
This is so by design like the heat function.

@kisslorand can tell you why.

radek8, 2021-06-20:
It would be logical that the buttons for the nozzle and for the bed will be separate on the screen.
Unfortunately, they are merged into one button. So no matter which icon you click on, the same window will always open.
To maintain the same logic on all screens, only 8 buttons are available.
It was more important for Kssloránd to switch the percentage of the shift than to select the heating source :-(

northwestfrog, 2021-06-20:
If there is no intent to change the fact both buttons act as a single one, at least disable the bed heat button and make sure tbe extruder heat comes up when the extruder is selected.

Right now the "Randomness" of the functionality may provide more frustration than anything else.

Thanks.

radek8, 2021-06-20:
Yes, it was the intention of fact both buttons act as a single one,

northwestfrog, 2021-06-20:
@radek8 how were you able to read that both buttons were defined as one? I went through printingmenu.c a few times and was not able to read that. I can see that option 2 is currently NOP:

      case KEY_ICON_2:
        if (isPrinting())
        {
          //  will be implemented in a future PR
        }
        break;

radek8, 2021-06-20:
The action on button 2 has not yet been defined.
Waiting for PR merge https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2006

The button area is defined in the API \ menu.c file

northwestfrog, 2021-06-20:
Just realized that once on the heat modification menu there is a button to toggle de different devices (T0/BED)
I will close the issue.

PR #2026 (2021-06-28): Extended clickable icons in printing menu
### Requirements

BTT or MKS TFT

### Description

This PR extends the clickable icons function in the printing menu. Each upper status icon has it's own function, except the Layer icon, which has no function associated yet, albeit it is clickable.

### Benefits

Nozzle status click brings up the Heat menu with the current nozzle preselected, clicking the bed status icon brings up the Heat menu with the bed preselected . Hopefully this clears the confusion created by previous implementation where those two icons had the same functionality, before they  [... cut]
closed 2021-06-20 by northwestfrog