bigtreetech/BIGTREETECH-TouchScreenFirmware #2019: [BUG] MKS_28_V1_0 support missing (or dead)

pfn, 2021-06-21:
### Description

It seems like #1815 renamed MKS_28_V1_0 to MKS_TFT28_V4_0; however when I build against that configuration and apply the firmware along with all the associated required updates (font/bmp/config) after rebooting the screen it comes up blank.

Unfortunately, my last working version prior to this was back in March (b3698ba771b449d843d097ba38d09268a2fb0b78) -- I have not updated since then.

I tried the last build before MKS_28_V1_0 got renamed at 157b016b0f90fd2bab0d54ae218914e69d28986d, however it results in a screen that has snow (looks like resolution mismatch?) when applying MKSTFT28.bin

cc: @digant73 since it looks like his changes removed MKSTFT28.bin output

### Steps to reproduce

Use the MKS28 TFT from the artillery printers, use MKS_TFT28_V4_0 as the build target and apply the firmware, it does not load.

**Expected behavior**
MKSTFT28.bin should still work

**Actual behavior**
TFT is blank upon boot

### Hardware Variant

MKS 2.8" TFT from Artillery printer

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

pfn, 2021-06-21:
I was able to roll back to 285c73502ea86c8d42299682b758f49f6391e8c1 and have a working TFT, however it's about 2.5 months out of date (April 7th)

digant73, 2021-06-22:
you must rename MKS_TFT28_V4_0. bin to MKSTFT28.bin and also install the icons (MKS dir on the root directory) and config.ini.
Try also to put the "reset.txt" on the root dir on the sd card. Probably it will fix your problem

oldman4U, 2021-06-22:
I am sure this user has been reading the ReadMe on the front page, before he did the update - or at least before he took the time to start this ticket.....

pfn, 2021-06-22:
Yes, I renamed the bin output of MKS_TFT28_V4_0 to MKSTFT28.bin when trying the latest from Master along with all the requisite font/bmp/language/config as usual. 

After completing the update process, the tft screen goes blank, and remains blank with any further resets.

The screen is not functional but can apply the older version to revert. I'll give MKS_TFT28_V4_0 another try with reset.txt to see if it makes any difference.

pfn, 2021-06-22:
Looks like re-applying the changes with reset.txt got everything working again (although it looks like there's an order of operations bug where config.ini is also reset after the same update run) -- I had to apply config.ini a second time in order for that update to succeed.

Thanks for the reset.txt suggestion @digant73

closed 2021-06-22 by pfn