bigtreetech/BIGTREETECH-TouchScreenFirmware #2020: [Q] Display Firmware and screen for IDEX mode ?

Schtom, 2021-06-21:
Hello,

I would like to put a screen firmware for your TFT screen (I have a TFT50)
A complete firmware FOR the IDEX!

Currently you have IDEX motherboard, but no display firmware for IDEX

Can you help me create a screen firmware specially designed for the IDEX?

I would need to include three commands with three images when selecting the print:

1st: "Normal" printing with independent print heads
Command: M605 S1

2nd: "Duplication" printing with the two heads which prints the same identical part.
Command: M605 S2

3rd: "Mirror" printing with the two heads which print the same identical part, opposite to each other.
Order: M605 S2 and M605 S3

Can't wait to see if you have a solution to this problem :)
Regards,
Raphael

[image]

radek8, 2021-06-23:
What is your idea?

oldman4U, 2021-06-23:
Print modes for two independent print heads. Duplicate or two colors or mirror. Guess this is a lot of work like portrait mode and there has been a ticket related to this already which is also in the FR list.

Schtom, 2021-06-23:
The idea is to add the 3 proposals after the "print" logo:

Normal mode
Mode duplication
Fashion mirror

Pets27, 2021-06-27:
Hello,
this idea is great. Now I have to switch to marlin mode to run duplication mode (mirror mode). I would add to the preheating so that when selecting everything, all nozzles are heated.
Thank you very much Petr

radek8, 2021-06-27:
Why don't you use your own TFT preferences

Schtom, 2021-06-27:
I don't know how to add options to the screen, with a selection between these 3 modes.
I am using the marlin command "M605 ..."
But I would have liked with a selection by image, it was much cleaner and simpler.
This is what I would need help on, to design and set up the images on the screen

radek8, 2021-06-27:
The config.ini file
lines 680 to 709
You define your own commands, for example:
M605 S1
M605 S2
M605 S3

Schtom, 2021-06-27:
Good evening,

I modified my firmware and the "custom" image to make it as clean as possible.

I would have liked to have this menu when you click directly on the "print" logo on the home screen

Thanks for your advice =)

https://user-images.githubusercontent.com/4550320/123559647-9552c900-d79d-11eb-923d-d4351a834ccb.mp4

radek8, 2021-06-27:
Replace this file and recompile the FW.

[Print.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6722361/Print.zip)

radek8, 2021-06-27:
here are the changes

[image]

Schtom, 2021-06-28:
> 
> 
> Replace this file and recompile the FW.
> 
> [Print.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6722361/Print.zip)

I configured the file you shared with me and compiled.

Everything is working very well.

Thank you very much for your help, I would share my whole final project with the group =)

Still 2-3 details to settle in the marlin to get me purged in my tanks.
I still have trouble finding the right mechanics

Pets27, 2021-07-04:
Hello,
it is nice. I have another problem. Even if I have M605 S0 (M605 S2) entered in the custom code (config. Ini) and press the button, it does not respond.
This works correctly in Terminal. Don't know what it can be?
Peter

Pets27, 2021-07-04:
Hello,
it's nice and it works very well.
Thank you very much
  Peter

oldman4U, 2021-07-12:
Radek. Are you going to make a PR for this? Anything we should add to the documentation?

radek8, 2021-07-12:
No I won't do PR.
This is not a good solution.
The correct solution would be if an icon was displayed on the IDEX machine and the parameter (single, diplicate, mirror) was in the description of the icon.
Clicking on the icon would change the description of the icon and the print parameter.
It wouldn't be hard to program, but unfortunately I'm not a programmer. I might be able to do it, but I'll leave it to someone more experienced.
Unfortunately, most local programmers do not have IDEX :-(

closed 2021-09-18 by stale[bot]